roleoroleo/yi-hack-MStar #86: Password reset?

Pantho, 2020-02-07:
Hi,

So I have a 6FUS and I flashed this firmware then set a pswd.
Should this be FTP as well as HTTP etc?

Either way none of the services accept my password now. I've tried formatting the SD card and flashing the firmware again but still wants a password. Is there a way to reset/default this?

Pantho, 2020-02-07:
Oh damn, the reset pin on the back worked for factory firmware ... good, and odd :) but 

disregard i guess.

Although I am still struggling to find default FTP/FTPD passwords.

Pantho, 2020-02-07:
After many reset/repairs etc I can't get anything to work - are there some default passwords for ip:8080 I can try?

roleoroleo, 2020-02-08:
Default password for ftp is blank.

Pantho, 2020-02-08:
Ah.

I changed the Web panel password, it won't accept it and now I can't get in
:(

Anyway to reset it, tried reflashing.

roleoroleo, 2020-02-08:
If reflash doesn't work, reflash with an older version and then again with the last version.

Pantho, 2020-02-09:
I've a 6RFUS and tried all the way back too yi-hack-6FUS-9FUS-BFUS 0.2.0

I can still connect to WIFI but if i flash and to go http://ip:8080 i get a
password prompt screen.

No way to reset w/e memory segment this password is stored in?

roleoroleo, 2020-02-09:
The password is stored in /home/yi-hack/etc/system.conf
Edit it with putty or using ftp.

Pantho, 2020-02-09:
Ah, nice got back in using yi_home_1080p_6FUS_450_0.2.0 :)

Thanks.

Pantho, 2020-02-09:
I am still having issues connecting to FTP tho Roleo.

I can access the web console in :8080 but any connections to :21 are
instantly refused.
If there is no default user/pass then how is one meant to connect?

Response: 331 Specify password
Command: PASS *********************
Response: 530 Login failed
Error: Critical error: Could not connect to server

roleoroleo, 2020-02-10:
What do you mean "camera aspect"?
Now that it works you could try the latest version.
I'm using motioneye software, I dont'use a dedicated hardware. motioneye doesn't support onvif but reads rtsp streams.

nrrdzilla, 2020-04-10:
Hi all.

I had the same problem as Pantho but I was using a password manager, and just doing a copy&paste into the username and password fields. So it's not like I was typing it wrong...

I guess my question is what limitations are there for the password? I'm wondering if the ASCII-salad produced by my password manager caused my problem.

TIA,
M

roleoroleo, 2020-04-10:
The password affects only httpd and video services.
You can enter via ssh and change the password, editing the file /home/yi-hack/etc/system.conf

nrrdzilla, 2020-04-10:
Hi,

First -- I'd like to say thanks for the work you put into this. I really do
appreciate what you've provided for users of the Yi devices :)

OK -- this gets me in to change the password, but I still need to come back
to the original question:
are there any limitations to the password?

I'm at a loss as to why I was able to set the password using copy and paste
from my password manager (without warnings or error message) but then when
I rebooted the camera to effect the changes, the same password did not
work. I'm pretty sure my manager is generating 16 character length, with
upper and lower case letters, numbers and symbols, with no constraints on
the symbols. I'm thinking this last lack of constraints *might* be the
issue. (?)

roleoroleo, 2020-04-10:
The urlencode/decode function is very huge.
And all processes start within a script where there are many variable elaborations.
So it's possible that something is wrong and maybe some chars are not correctly evaluated.
Avoid to use classic linux shell chars.

roleoroleo, 2020-04-10:
And check if the file /home/yi-hack/etc/system.conf contains the string  you inserted.

nrrdzilla, 2020-04-22:
I ended up doing the successive downgrade-upgrade firmware thing to wipe and reinstall. I used a password that has no classic linux shell characters and everything seems happy. That seems to be have been the source of grief.

Thanks a bunch!
-n

closed 2020-05-07 by roleoroleo