roleoroleo/yi-hack-MStar #88: ONVIF

manuelepi, 2020-02-18:
Dear All,
flashed my Yi 1080p camera but cannot add to my NVR (Hikvision) not to any other ONVIF client (like onvifier or Tinycam). Only able to see stream rstp via VLC...

Would you be so kind to clarify parameters?

Thank You
M

myromeo, 2020-02-18:
Connects OK for me on Hikvision (Annke) NVR. How are you trying to set up? No user/password is needed but my NVR requires something for them, I use admin/password and it connects fine. Make sure to use ONVIF not IPCAM and protocol is fine Auto (TCP/UDP)

manuelepi, 2020-02-21:
Thanks for your reply. Flashed twice. Unable to connect by the way. Sadly going to return to Amazon.

Zippeyrude, 2020-02-28:
I'm unable to connect to hikvision nvr also, only rtsp via vlc etc.

Perhaps can you share config on hikvision please so I can see if I'm doing something different or wrong on the configuration?

Thank you 🙂

myromeo, 2020-02-28:
I’ve connected my NVR direct to the camera not via VLC. I simply went to camera management, added network camera with IP, protocol ONVIF, port 80 TCP. I have added a username and password on the cameras as NVR doesn’t accept blank user/pass fields.

mojo965, 2020-02-29:
Hi. I have the same issue. But only with my Synology Surveillance Station. I can see YI home 1080p camera only as rtsp stream (TCP port 554) - working perfect, but not as ONVIF. 
ONVIF tested at IPcam Viever (android) - working (TCP port 80).  So I think the problem is rather Synology or hikvision recorder.

myromeo, 2020-02-29:
Try changing port to 81

mojo965, 2020-03-01:
@myromeo  - I have tested other ports (81 too). Synology does not see YI camera at ONVIF at all. Auto searching and manual (ip, config etc) does not work.
As I mentioned works great as rtsp stream. Synology recognized it without problems.
I  have actual version of Surveillance Station. It works with other cameras af ONVIF mode.

gjstroom, 2020-04-03:
I flashed a 9UFS with the y203c_0.2.9 firmware, that working fine.
I like to use the camera with a Windows 10 build 19041 machine which should be able to add camera's by the ONVIF protocol as stated here: https://www.onmsft.com/news/latest-windows-insider-builds-add-support-for-onvif-compliant-network-cameras.
Can anyone confirm this should work using the yi-hack firmware ?

maureis, 2020-04-03:
> Can anyone confirm this should work using the yi-hack firmware ?

my Dome 1080p cannot be found in "Add device" windows utility

fosterym, 2020-04-09:
I have the same problem - with Synology does not see YI camera at ONVIF at all.
rtsp working fine, but without audio:-(
Yi 1080 Dome - BFUS
Yi 1080 Home - 9FUS

guyyar89, 2020-08-20:
manuelepi do you have answer how to fix it ?
i have same problem with onvif on nvr hikvision

marcor2196, 2020-11-20:
Hi, same problem with Synology Surveillance Station, rstp works fine but Onvif not.
Any news?

roleoroleo, 2020-11-21:
https://github.com/roleoroleo/yi-hack-MStar/issues/277#issuecomment-731021534

marcor2196, 2020-11-21:
> Try this solution:
> [#277 (comment)](https://github.com/roleoroleo/yi-hack-MStar/issues/277#issuecomment-731021534)

Hi, the bynary file that you linked it's a configuration file for Qnap SS?
I tried to import it to Synology SS but it's not accepted, i can attach here my configuration export for the camera and Synology SS.
In the file ".ExpCam" of Synology SS you can see the configuration of the rstp and onvif both.
(i tried to modify this file because i noticed that in the onvif configuration there was empty information such "media_port", "live_media_port", "path" and "live_path" that in the rtsp configuration are present instead, you can find the original file as Synology SS exported in the same folder, the name it's "original").
For onvif i see that there is another dedicated file in the folder ".Exp@SSCap".

Thank you so much!

[SSCamExport_CAM.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5577759/SSCamExport_CAM.zip)
[config.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5577760/config.zip)

roleoroleo, 2020-11-21:
The file I posted is a replacement binary for onvif daemon inside the cam.
You should try to overwrite the original file using a ftp client.

marcor2196, 2020-11-21:
I tried but it didn't solve, now on Synology SS i can't add it because the connection test result in error (a generic useless error).
Before instead the connection test end succesful but after the camera was indicated as disconnected and showed nothing.

roleoroleo, 2020-11-21:
Synology is a tough opponent.

marcor2196, 2020-11-21:
Can i do something to help? Provide other informations or logs?

roleoroleo, 2020-11-21:
I need a tcpdump of the traffic between the cam and the synology during the negotation procedure.

marcor2196, 2020-11-22:
Seems that the camera don't respond on the Onvif port (80)...
See the packets exchange attached.

Before the binary file replacement i'm sure that answered because on Synology SS the connection test result ok and i could add the camera on Home Assistant with the Onvif integration.

I tried to replace again the binary file with the original old one that i saved before (in attachments) but i didn't resolve.
The Onvif port still not responding, there is a way to restore the original firmware files?
Or what advise me to do?

Thank you for the help!

[packets.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5579380/packets.zip)
[onvif_srvd.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5579386/onvif_srvd.zip)

roleoroleo, 2020-11-22:
If you downgrade to the previous version, you can then reinstall the current release.

But it's a weird problem. Did you check the file permission after overwriting?
This is a correct sequence:

1. Login with ssh.
2. run `killall onvif_srvd`
3. Rename old onvif_srvd to onvif_srvd.bak
4. Copy with ftp or scp the file new onvif_srvd.
5. Set permission: `chmod 0755 /home/yi-hack/bin/onvif_srvd`
6. Reboot.

marcor2196, 2020-11-22:
Surely I had something wrong the first time, because now i've reinstalled the firmware and i've also retry to replace the binary file and now the Onvif port respond.
On Synology SS i see that more information it's shown during the configuration process, but unfortunately the problem persists, i can't see the video feed, the camera resuts "disconnected".

I attached the tcpdump requested, thank you!

[tcpdump.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5579711/tcpdump.zip)

roleoroleo, 2020-12-19:
The log shows that Synology sends CreateProfile and DeleteProfile requests.
The cam doesn't support dynamic profile creation.
If you want to try a beta version I could implement a fake response to these requests.

marcor2196, 2020-12-20:
Hi, yes I can try the beta version, thank you so much!

roleoroleo, 2020-12-20:
Here it is:
[onvif_srvd.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5720862/onvif_srvd.gz)

unnameable73, 2020-12-27:
Hi, sorry for the intrusion.
I made some modification to the onvif_srvd to make it work with my Synology Sourveliance station.
The dynamic profile creation it's used only to check the media soap stack.
The station uses only the factory defined profiles.
In this version, I solved also the PTZ functions recognition.

Here you have the binary
[onvif_srvd.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5745195/onvif_srvd.gz)

and the core source code
[onvif_srvd-src-folder.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5745196/onvif_srvd-src-folder.zip)

roleoroleo, 2020-12-28:
Thank you for your time.
I will check your changes.

roleoroleo, 2020-12-28:
@unnameable73 
Why did you update service versions?

unnameable73, 2020-12-28:
No, sorry.....

unnameable73, 2020-12-28:
> @unnameable73
> Why did you update service versions?

No, sorry....

unnameable73, 2020-12-28:
> Thank you for your time.
> I will check your changes.

Maybe for you, it's impossible to test the service with a Synology Sourveliance station. I make the modification for me and gladly I give those notifications to anyone who needs them.

roleoroleo, 2020-12-28:
I imported your changes into my sources with some minor refactoring.
Could you check if it works on the Synology Station?
[onvif_srvd.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5748992/onvif_srvd.gz)

unnameable73, 2020-12-28:
> I imported your changes into my sources with some minor refactoring.
> Could you check if it works on the Synology Station?
> [onvif_srvd.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5748992/onvif_srvd.gz)

OK! Your version working on my Synology Station.
As always I added a new camera. In this way I checked all the functions avoiding cached settings in the Sourveliance station.

Regarding the "service versions" it's not mandatory the updated values.. I just checked with yous original values.

In a SOAP service I always prefer to use the correct endpoint as in your updadated version for a better service dispatch.
> http://onvif_srv:80/onvif/device_service
http://onvif_srv:80/onvif/media_service
http://onvif_srv:80/onvif/ptz_service

roleoroleo, 2020-12-29:
> Regarding the "service versions" it's not mandatory the updated values.. I just checked with yous original values both in GetCapabilities and GetServices.
> [ServiceDevice.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5749066/ServiceDevice.zip)

I took the opportunity to update everything to 20.12

> In a SOAP service I always prefer to use the correct endpoint as in your updadated version for a better service dispatch.
> 
> > http://onvif_srv:80/onvif/device_service
> > http://onvif_srv:80/onvif/media_service
> > http://onvif_srv:80/onvif/ptz_service

Ok.

Thank you.

roleoroleo, 2021-02-11:
Released: https://github.com/roleoroleo/yi-hack-MStar/commit/b644f1427e6e190e4be22bb1895be4e372f0edd7

closed 2021-02-11 by roleoroleo