roleoroleo/yi-hack-MStar #90: Update via ww (like TheCrypt0's)

majkers, 2020-02-23:
Hello, I have my camera mounted high on the wall and it is really hard for me to take it off to update firmware. Is it possible to add update option via web interface just like in TheCrypt0's version? Thanks

roleoroleo, 2020-02-23:
Connect with ftp
Copy the upgrade files to /tmp/sd
Reboot the cam

majkers, 2020-02-23:
Thanks a lot!

GuitarSkater, 2020-02-23:
I do the FTP method every time. I am not sure if its a fluke, but now i have the tendency to reformat the SD card after i flash the update. (it could be just me, but I feel like it runs better and doesnt have to think about the firmware files in the event the camera restarts)

roleoroleo, 2020-02-23:
You can use the same SD without reformatting.
When a file is used to upgrade the firmware, after a succesfully upgrade, the bootloader stores the hash of the file.
So, if you don't delete the file, is not a problem: the bootloader checks the hash and doesn't start the upgrade procedure.

steef84, 2020-02-23:
This kind of update via ftp never worked for me. Tried update one 4FUs y203c from 0.2.6 to 0.2.7 just now and yellow light blinks for numerous seconds, then goes steady for eternal. Its been steady for 10 minutes now and cam doesn't come up. ftp'ed both files to sd card, and in web interface pushed reboot button.

roleoroleo, 2020-02-24:
So how do you upgrade?

steef84, 2020-02-24:
After roughly 24 hours the led still is solid orange. When rebooting via cutting off power the cam powers on instantly with solid orange led. Powered off the cam again, I removed the sd, transferred the files again and then inserted the sd card. Then power on again, update started again but now completes. Will try another cam with cloud features off, as I enabled them for now. Will report back

GuitarSkater, 2020-02-24:
> After roughly 24 hours the led still is solid orange. When rebooting via cutting off power the cam powers on instantly with solid orange led. Powered off the cam again, I removed the sd, transferred the files again and then inserted the sd card. Then power on again, update started again but now completes. Will try another cam with cloud features off, as I enabled them for now. Will report back.
> NOPE: same as above. Yellow light stays on. How you guys rebooting? Via webif?

Have you tried the older firmware? Also, what size SD card are you using?

steef84, 2020-02-25:
Using 64gb cards. 
Upgrade always working when removing sd card and transfer them via computer, but never worked for me when using ftp. 
Having 6 camera’s y203c 4FUS, 6FUS and 9FUS with this ‘issue’. Like to update the cams without the need of removing sd cards.

roleoroleo, 2020-02-25:
What ftp client do you use?

steef84, 2020-02-27:
Filezilla on MacOS.

roleoroleo, 2020-02-27:
Tried it now.
I copied the new fw in /tmp/sd using Filezilla on Windows (home_y203c and sys_y203c).
I rebooted the cam with the web interface.
After 2 minutes I have the new firmware.
I can't understand the problem...

ffix, 2020-02-27:
I tried to update 0.2.6 to 0.2.7 on 6FUS camera a few days ago. I copied home_y203c sys_y203c to /tmp/sd via ssh and rebooted cam using web UI multiple times. Nothing happened. Firmware stays 0.2.6.

roleoroleo, 2020-02-27:
But if you reformat the SD, it works...
Correct?

steef84, 2020-02-29:
Formatted the sd cards for now, and updated all cams to 0.2.7 via sd card. Now waiting for next release, maybe it has to do with formatting...

steef84, 2020-03-09:
Tried y203c cam today to update from 0.2.7 to 0.2.8.
First formatted 32gb sd card and installed 0.2.7 on cam around 9 days ago. 
Today overwrote the two files via FileZilla with the 0.2.8 files. Rebooted cam via webinferface and cams starts to upgrade. Led is now still solid for around 10 minutes. Seems not to work for me -.- 
I guess it has to do with bigger sd cards as I can't make this to work for neither of my 6 camera's.

majkers, 2020-03-10:
I tried it with FTP and it works. Thanks!

closed 2020-03-10 by majkers

roleoroleo, 2020-03-10:
@steef84 
I don't know why...

divemasterjm, 2020-03-30:
I try ftp to sd with filezilla but when reboot nothing happens still on 0.2.8

roleoroleo, 2020-03-30:
Try with another sd.