roleoroleo/yi-hack-MStar #91: 9FUS Y31 firmware 8.2.0

scorpio862, 2020-02-24:
I just bought a YI Home1080p 9FUS Y31 with firmware 8.2.0. Is there any chance of getting this hack to work with this new firmware?

roleoroleo, 2020-02-24:
To check if the hack works, I need a dump of your firmware.
Please read the wiki:
https://github.com/roleoroleo/yi-hack-MStar/wiki/Dump-your-backup-firmware

ruppert92, 2020-02-26:
Is there anyway to rollback firmware? I also have version 8.2, but only after accidentally updating in the app

roleoroleo, 2020-02-26:
Do you remember the exact string of the old version?
Like 4.5.0.0C_201910080934

ruppert92, 2020-02-26:
Unfortunately no I never saw the original number

roleoroleo, 2020-02-26:
No way.

micfogas, 2020-02-27:
I have this cam too. 8.2.0.0 firmware. I didn't think to hack it when I first set it up. I did a firmware update on it pretty much instantly. the previous firmware version was also 8.2.0.0 but an earlier date. I lost my USB serial cable so I'm gonna grab another off Amazon and dump the firmware for you. nothing happens so far with 2 SD cards I've tried. I even tried getting into the manual update mode to cap the firmware file but it's stored in the root-only data folder on my phone. hate to reroot just for that but I can easily. tried packet capping using laptop as hotspot for the cam, just to get the firmware file names at least. I'm guessing they renamed them maybe for the latest. I'm gonna keep working at it. I'll update you.

WattsC-90, 2020-02-29:
going to be following this thread keenly.. I have also run into this issue it seems! 8.2.0.0 firmware on a 9FUS 1080p home cam. how dare they change things! :p

starbuck93, 2020-02-29:
What should the log output look like? @roleoroleo Here's my first attempt at getting the logs.
[image]
[image]

roleoroleo, 2020-02-29:
It seems very very different from the platform I know.
Please try to repeat with SD inserted.

starbuck93, 2020-02-29:
printenv outputs:
https://hastebin.com/newejicuce.makefile <-- before I inserted an SD card
https://hastebin.com/lutedolugo.makefile <-- after I inserted an SD card

full raw log output with SD card --> https://hastebin.com/ifikotuxod.sql

WattsC-90, 2020-02-29:
@starbuck93 line 171, `---update file(home_y20gam) Not exist---` ...
@roleoroleo does this mean wrong file name with new version?

roleoroleo, 2020-02-29:
Unfortunately it means unsupported platform.

starbuck93, 2020-02-29:
Is there any other info I can grab off of the device for you?

starbuck93, 2020-02-29:
@WattsC-90 yeah I didn't have the update files on the SD card to attempt to flash it. I tried pretty much all of the firmware hacks before I ended up soldering these tiny wires to the circuit board to get this data.

roleoroleo, 2020-03-01:
It's a new platform.
Incompatible with v4 and Mstar.
I think we need a new hack.

roleoroleo, 2020-03-01:
@starbuck93 
Are you able to make a dump as explained in my wiki?
If yes, we need to modify a bit the script and the flash addresses.

starbuck93, 2020-03-01:
@roleoroleo I should be able to. I attempted to yesterday but only had a few minutes available and didn't know how to figure out the length of the parameters of the script.

WattsC-90, 2020-03-01:
@roleoroleo how easy is it to create a new hack?

roleoroleo, 2020-03-01:
@starbuck93 
The log doesn't report the mtd partition.
We have only this:
`partitions=boot@mtdblock1:rootfs@mtdblock2:home@mtdblock3:env@mtdblock4:mfg@mtdblock5:conf@mtdblock6:UDISK@mtdblock7`
Probably we need to dump the whole flash... very long operation.

@WattsC-90 
It depends on how many differences there are. And I don't know if I can help because I don't have this model.

starbuck93, 2020-03-01:
@roleoroleo how would I go about that? Do you think using the python script to grab every byteword from 0 to 64Mb or however large? I found this command `md` and it outputs this: https://hastebin.com/xagilehete.makefile

roleoroleo, 2020-03-01:
Yes, the python script runs the md command...
The flash is a en25qh128, so 16 MB.

starbuck93, 2020-03-01:
I ran `md 0 8000000` before I saw your previous message. It just stopped so let me show you what it outputted. https://drive.google.com/open?id=1-r8_MsrR2rx3vHyKjEDHSGUTghIczIjM

WattsC-90, 2020-03-06:
Any luck on this one? unfortunately I am entirely unable to help piece things together for you or get anything together :(

alpacalife, 2020-03-10:
Just bought one of these cameras! I never upgraded the firmware and out of the box it was version 8.2.0.0A_201912121707

I can attempt to get a dump if @starbuck93 doesn't have any luck. I need to review the wiki a bit.

roleoroleo, 2020-03-10:
You have to change the script.
The following line:
`if line == 'MStar # ':`
must be changed according to your u-boot prompt.
Pay attention to the space after the '#'.

rclsilver, 2020-03-11:
Hello, i've this version too (4.5.0.0C_201910080934), how can I help you?

I tried to put file on SD but camera boots without flash... I can, if required, dump my rom

roleoroleo, 2020-03-11:
4.5.0.0C is probably already supported.
See y203c files.

rclsilver, 2020-03-11:
Hum, you're right... I'll try this evening... thanks!

rogerb831, 2020-03-16:
I just bought a BFUS camera with version 8.2.0.0A_201912121707. It also wants me to do an upgrade to 8.2.0.0A_201912270941

rogerb831, 2020-03-18:
I just ordered a couple more for sacrificial use and a USB/TTL adapter. I'll try to dump if nobody beats me to it.

rogerb831, 2020-03-19:
Ok, got connected and grabbed a boot log. https://www.dropbox.com/s/pbvjdh3omhnocwk/bootlog.txt?dl=0. It's nearly identical to Starbuck's.  I tried to follow the directions to dump the flash, but theres no indicator of where the partitions are. Then I attempted to just dump 0 to 16MB using the python script, but the platform doesn't seem to include the "sf" command. Any tips? Maybe I'm missing something?

roleoroleo, 2020-03-19:
Yes.
I think it's the same situation as @starbuck93.
Did you stop correctly the boot at the bootloader?

rogerb831, 2020-03-19:
Sorry, I;m new to embedded and didn't see that in the wiki page. I was able to locate it. FYI, for this platform the string to watch for is 
`"Hit any key to stop autoboot: "`

sf probe gives the following
`=> sf probe
sf: bus-0 cs-0 spedd-50000000Hz mode-3000
SF: Detected en25qh128 with page size 256 Bytes, erase size 64 KiB, total 16 MiB`

So I can use the python script by updating the "MStar # " string and giving it the hex range 0x00 to 0xf42400 to  just grab the whole 16MB?

roleoroleo, 2020-03-19:
You need to dump the whole flash:
```
sf probe
sf read 0x21000000 0x0 0x1000000
py dump.py --timeout 1 --step 64 comX 115200 0x21000000 0xfffff0
```
A lot of patience.

alpacalife, 2020-03-19:
Just to give an update. I got mine connected over serial. I am able to connect and work with it.

I saw the same issue @starbuck93 saw, not mtd partitions defined. Don't know what to dump.

I got to the point of wanting to dump the entire 16MB flash, but I got an error or something when using sf command to dump to memory... I then gave up and went to bed. 

@roleoroleo I can test those commands tonight and post results unless I have any errors. those look like what I recall trying to do!

Nanook, 2020-03-19:
I have the same Cam. New to this stuff. Happy to crack it open to confirm anything. Came with 8.2.0.0A_201912270941 installed.

rogerb831, 2020-03-20:
Thanks roleoroleo, that helped. The dump is running now. This is going to take some time. Slight modification to your commands for the others who are playing along.

`sf probe
sf read 0x21000000 0x0 0x1000000
py dump.py --timeout 1 --step 64 comX 115200 0x21000000 0x22000000`

And here is the modified script that works for this platform

`#!/usr/bin/env python3
__author__ = 'Ion Agorria'
import argparse
from datetime import datetime
from time import time
from serial import Serial

def visible(text):
    text.replace("\n", "\\n")
    text.replace("\r", "\\r")
    return text

def parse_line(line, size):
    # Break down each part
    if len(line) != size:
        raise Exception("Line size %s doesn't match %s: %s" %
                        (len(line), size, visible(line)))
    i = 0
    addr = line[i:i + 8]
    i += 8
    assert line[i:i + 2] == ": "
    i += 2
    hex_data = line[i:i + 8]
    i += 8
    assert line[i:i + 1] == " "
    i += 1
    hex_data += line[i:i + 8]
    i += 8
    assert line[i:i + 1] == " "
    i += 1
    hex_data += line[i:i + 8]
    i += 8
    assert line[i:i + 1] == " "
    i += 1
    hex_data += line[i:i + 8]
    i += 8
    assert line[i:i + 4] == "    "
    i += 4
    text = line[i:i + 16]
    i += 16
    assert line[i:i + 2] == "\r\n"

    # Convert hex string into actual data
    data = []
    for i in range(0, len(hex_data), 2):
        piece = hex_data[i:i + 2]
        data.append(int(piece, 16))

    # For redundancy check if data matches text, but only for printable characters (non dot)
#    for i, piece in enumerate(data):
#        if text[i] != ".":
#            assert chr(piece) == text[i]

    return int(addr, 16), data, text

def write(serial, opts, data):
    if opts.debug:
        print("Debug write: " + visible(data))
    data = bytes(data, "ascii")
    serial.write(data)

def dump(serial, opts, log):
[... 192 lines cut ...]
        "%Y-%m-%dT%H%M%S") + "_" + hex(opts.start) + "_" + hex(opts.end)

    # Prepare to dump
    serial = Serial(port=opts.port, baudrate=opts.baud, timeout=opts.timeout)
    log = open(name + ".log", "w")
    try:
        data = dump(serial, opts, log)
    except Exception as e:
        raise e
    finally:
        serial.close()
        log.close()

    # Write to file
    file = open(name + ".img", "wb")
    try:
        for line in data:
            file.write(line)
    except Exception as e:
        raise e
    finally:
        file.close()

main()
`

rogerb831, 2020-03-20:
I just realized I was getting all zeroes. Turns out the sf probe failed. I'm a bit stuck on this one. I tried using different start addresses but get the same error.

```
=> sf read 0x21000000 0x0 0x1000000
device 0 whole chip
rx wait_dma_recv_over fail
SF: 16777216 bytes @ 0x0 time:5005 ms Read: ERROR -1
```

rogerb831, 2020-03-20:
ok, did some more exploration and found that the uboot command "fdt print /" lists out the device tree. In there I found the following.
```

                partitions {
                        device_type = "partitions";
                        boot {
                                device_type = "boot";
                                offset = <0x00000020>;
                                size = <0x00001200>;
                        };
                        rootfs {
                                device_type = "rootfs";
                                offset = <0x00001220>;
                                size = <0x00000c00>;
                        };
                        home {
                                device_type = "home";
                                offset = <0x00001e20>;
                                size = <0x00005c00>;
                        };
                        env {
                                device_type = "env";
                                offset = <0x00007a20>;
                                size = <0x00000100>;
                        };
                        mfg {
                                device_type = "mfg";
                                offset = <0x00007b20>;
                                size = <0x00000080>;
                        };
                        conf {
                                device_type = "conf";
                                offset = <0x00007ba0>;
                                size = <0x00000080>;
                        };
                        UDISK {
                                device_type = "UDISK";
                                offset = <0x00007c20>;
                                size = <0x00000000>;
                        };
                };

```

roleoroleo, 2020-03-20:
Try to use sunxi_flash command.
You can have an example looking at the printenv output.

I think this is the help:
```
Usage:
sunxi_flash read command parmeters :
parmeters 0 : addr to load(hex only)
parmeters 1 : the name of the part to be load
[parmeters 2] : the number of bytes to be load(hex only)
if [parmeters 2] not exist, the number of bytes to be load is the size of the part indecated on partemeter 1
```

rogerb831, 2020-03-21:
Ok, I made some progress on this. sunxi_flash seems to be the way to do it. It wants you to give it a RAM address to load into to and a partition name to read from the flash

`sunxi_flash read 45000000 home`

You can optionally give it a length as well, so I told it to start at "boot" and gave it as long of a length as I could without hitting the end of the flash:

`sunxi_flash read 45000000 boot f80000`

From there I can dump it to an sd card by doing the follow (nice to not have to copy it over the serial connection).

Load the card driver
`sunxi_card0_probe`

Verify you can access the card filesystem
`fatls mmc 0`

Write the image from RAM to the card
`fatwrite mmc 0 45000000 image.bin f80000`

After that I was able to identify the components in the image with binwalk. Extract with dd, and the appropriate subsequent tools (jefferson for JFFS2, unsquash for squash, gzip for kernel).

Now to figure out how to add to it and build a new image.

roleoroleo, 2020-03-21:
Good!
Normally the fatwrite command is not available.
Yes, probably you can boot directly from SD.
Bu don't worry, if u-boot is ok you can rewrite the flash as you want.
If you share your dump I could help to create a new hack with the first tools: busybox, telnet, ssh...

rogerb831, 2020-03-21:
Sure, I'd love to get your help with it. It's going to be some work though. It looks like it may be a completely different partition/filesystem structure. Maybe a different SDK. I dunno, I'm a newb.

```
$ binwalk image.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             uImage header, header size: 64 bytes, header CRC: 0x2057BD0B, created: 2019-12-13 10:11:19, image size: 2080440 bytes, Data Address: 0x40008000, Entry Point: 0x40008000, data CRC: 0xD20F4B45, OS: Linux, CPU: ARM, image type: OS Kernel Image, compression type: none, image name: "ARM OpenWrt Linux-4.9.118"
2268          0x8DC           device tree image (dtb)
10584         0x2958          device tree image (dtb)
14020         0x36C4          device tree image (dtb)
18136         0x46D8          gzip compressed data, maximum compression, from Unix, last modified: 1970-01-01 00:00:00 (null date)
2359296       0x240000        Squashfs filesystem, little endian, version 4.0, compression:gzip, size: 1427049 bytes, 219 inodes, blocksize: 262144 bytes, created: 2019-12-27 01:42:11
3932160       0x3C0000        JFFS2 filesystem, little endian
```

Anyway, not sure I want to post the image up, who knows whats in it. I know my wifi info is for sure but I'm not sure how to strip it out. Email me and I'll send it to you privately. Here a burner email you can find me at fe27eahy53ac@opayq.com.

rogerb831, 2020-03-22:
Thanks Roleo, I just sent you the details. The burner email is going down. Let me know if it looks like something you can work with or if you think it's a completely new hack effort.

roleoroleo, 2020-03-22:
Ok, I'm studying the image.
As you can read from binwalk, the image starts with a kernel image in u-boot format.
The first 5 lines that binwalk shows are inside this kernel image.
There is 3 other jffs2 but are empty (only header).
At 0x00f60000 you can find the private info, like serial number, wifi, etc...

Good news, the rootfs contains telnetd.

roleoroleo, 2020-03-22:
Other questions:

1. The image is less then 16MB. Why do you dump f80000 and not 1000000?
2. u-boot is missing. Is there in the missing part of the flash? Is there an additional internal flash?

roleoroleo, 2020-03-22:
And after telnetd, you should test dropbear:
https://drive.google.com/open?id=1Y_JKx5qf2BPpVVz4P-AI0tbQ-KTLNEbE

rogerb831, 2020-03-22:
Nice, I'll give it a shot! 

The reason the image is cut off at the start is because I can't seem to pass a starting RAM address to sunxi_flash, it seems to require a label. So I had to tell it to start at "boot". However that is mapped to a memory address, it's probably after uboot.

I have to use "sunxi_flash read 45000000 **boot** f80000".

I tried "sunxi_flash read 45000000 0x0 f80000", and similar variations, but it just doesn't work.

If you see any indicated or what I could pass instead, or how I can create a new label, I can make another dump.

roleoroleo, 2020-03-22:
If you run "sunxi_flash 45000000 boot 1000000"?
It works?

rogerb831, 2020-03-22:
I tried. It gives me a message indicating it passed the end of the FLASH.

On the positive side it looks like it did the update, and the home partition appears normal.
```
---tmp/sd/home_y20gam exist, update begin---
1+0 records in
1+0 records out
22 bytes (22B) copied, 0.001261 seconds, 17.0KB/s
```
Full log at https://www.dropbox.com/s/zcwc2krqq2ftvoh/bootlog2.txt?dl=0

I didn't actually see telnetd running. What changes did you make so I can verify?

roleoroleo, 2020-03-22:
The update appears to have failed because it does not recognize the version.
```
---tmp/sd/home_y20gam exist, update begin---
1+0 records in
1+0 records out
22 bytes (22B) copied, 0.001261 seconds, 17.0KB/s
check version: newver=¦
¦¦¦¦+¦n, curver=8.2.0.0A_201912270941
sh:
¦¦¦¦+¦n: unknown operand
---same version ? update fail---
```

rogerb831, 2020-03-22:
Seems the file in /home/homver needs to be incremented?

roleoroleo, 2020-03-23:
Try to remove the "m" from the name of the file: home_y20ga
We try to upgrade during u-boot and not after linux loads.

roleoroleo, 2020-03-23:
Other possibility.
Did you see these lines in init.sh?
```
if [ -f "/tmp/sd/Factory/factory_test.sh" ]; then
    /tmp/sd/Factory/config.sh
    exit
fi
```

rogerb831, 2020-03-23:
home_y20ga seems to have had no effect.

That init file is interesting. That seems like a wide open hook to execute any think on the SD card. Worst cast scenario one could load binary compatible files on the sd card an execute them through that script!

Also, I think we were on the right track earlier:
```
    echo "---/home/home_y20gam exist, update begin---"
        dd if=/home/home_y20gam of=/tmp/newver bs=22 count=1
        newver=$(cat /tmp/newver)
```

I think home wasn't updating because it's looking to pull the homever file from a specific block of the home_y20gam image. If it's not built the exact some way as the factory it's probable that the file won't be at that block..

roleoroleo, 2020-03-23:
In the MStar platform there are 2 methods to load firmwares.

1. A file without "m" (like home_y203c) containing a jffs2 image
2. A file with "m" (like home_y203cm) containing a 7zipped file with a password and a rsa signature

The first one (the method I use) is loaded by u-boot without checks and is useful for recovery processes.
The second one (used by online update) is loaded by a linux script in the final step of the boot.
The content of this archive overwrites the active file system.

The upgrade we tried didn't work because the file is not signed and doesn't match the yi format.
You can verify this procedure and the format in the file /home/base/tools/extpkg.sh
Unforunately we don't have the private key to sign this file.

rogerb831, 2020-03-23:
Perhaps is possible to use /tmp/sd/Factory/factory_test.sh to defeat the key check?

1. Load SD card with a /tmp/sd/Factory/factory_test.sh that overwrites /home/base/tools/extpkg.sh
2. Clear SD card and load with home_y203cm

roleoroleo, 2020-03-23:
Yes, I think it's possible to use factory_test.sh

About the binaries I'm trying some tests.
Helloworld works, complex programs give me a segfault.
I have to investigate.

rogerb831, 2020-03-24:
We are making progress guys. We've got a hack to root the device without having to use a serial adapter. Roleo is figuring out how to port the app stack over. Fingers crossed.

WattsC-90, 2020-03-24:
@rogerb831 thats brilliant! I cant believe the trouble it is! Wish they would make it just *a little* easier haha!

well done!

[53 more replies]

closed 2020-04-19 by roleoroleo