roleoroleo/yi-hack-MStar #93: Unable to record without cloud on SD

mihir8786, 2020-02-26:
I have the BFUS camera. 

Whenever I turn off the cloud (and make sure "recording without cloud" is on), all recording stops on the SD card. I have tried it with both "Save video when a motion is detected" on or off.

If I turn on the cloud, recording works on the card. Interestingly, if I leave leave the cloud on but block the camera from internet access at my router, SD card recording stops as well.

I have noticed it with previous versions as well but my current version is 0.2.7.

cccebek, 2020-02-28:
I Have the same problem anybody help?

hagenuck1, 2020-03-02:
Recording isn’t working for me, too. In 0.2.6 it worked for the first hours, but then stopped (probably because of no free space). Deleted the files and installed 0.2.7 y203, now it’s not recording anymore.

roleoroleo, 2020-03-02:
My y203c records correctly.
What are your settings?

cccebek, 2020-03-02:
IT happened when turn off wifi in my case :) i think wifi must be turned on ? When wifi turned on everything is ok

mihir8786, 2020-03-02:
Here's my system.conf:
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=no
RTSP=yes
RTSP_HIGH=yes
ONVIF=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
FREE_SPACE=5

And here's camera.conf
SWITCH_ON=yes
SAVE_VIDEO_ON_MOTION=no
SENSITIVITY=low
LED=no
ROTATE=yes
IR=yes

It feels like it's intermittent. It didn't record anything between Feb 23 and Feb 29. All of a sudden, it started recording on Feb 29 until March 1 morning and then stopped. Not sure if it's SD card space issue? I have the free space at 5%.

roleoroleo, 2020-03-02:
Tested now with your config and it works correctly.
Try to use a bigger SD card or to increase free space limit.

roleoroleo, 2020-03-02:
> 
> 
> IT happened when turn off wifi in my case :) i think wifi must be turned on ? When wifi turned on everything is ok

Unforunately I can't control the recording algorithm.
If there is a condition for which the cam doesn't record when wifi is off, I cannot change it.

mihir8786, 2020-03-03:
Thank you for testing. I think I found the issue.

I reset my cameras and match the settings on Yi app with the hack webpage such as motion detection, recording with motion, along with lights, rotation etc. I also formatted the SD card through the app. Once all settings were matched and to my preference, I disabled cloud through the hack page and the cameras are recording as expected. I can now toggle between motion only and continous recording through the hack page.

Hopefully this helps others. Thank you again for all your amazing work!

hagenuck1, 2020-03-03:
So, I’ve also activated cloud and given the camera internet access. 
Without resetting the camera I activated recording without motion on the Yi app. Also I Formates the sd Card there (6.4 GB was free). Now it started recording. 

After deactivating the cloud again and restarting the camera using hack it didn’t record anymore. In back the „record on Motion only“ was active. Set it to „always record“ and it still didn’t record. 

When I deleted the tmp.mp4.tmp file in the record folder by ftp the camera started recording again. The record folder was completely empty before activating cloud access. 

Thanks for your great work :)

closed 2020-08-20 by roleoroleo