roleoroleo/yi-hack-MStar #95: Swap File on SD

naeloob, 2020-02-27:
Hi!!
First of all, congrats for such amazing project, it brings a lot of features to our cameras.

Do you think is there any chance to get working an swap file on the sd??
There's no too much room, neither on ram or storage.
The storage can be supplied by the sd, and the memory using a swap file over there, or a partition.

What do you think?

Regards

roleoroleo, 2020-02-28:
I think is not a good idea to create a swap space on SD.
The SD card is terribly slow and not suggested for this purpose.
The frequent write operations can wear out flash cells quickly and destroy the memory.
Finally, as you already noted, the compiled busybox doesn't contain the swap programs. And the space available in the root partition is too little to add other components.

naeloob, 2020-02-28:
You are right on the SD comments.
But you don't need to use root partition in order to add commands or programs, you can use it from SD.
Can you point me to the crosscompiler to try to compile a more complet busybox? It's not only useful just for swap.

roleoroleo, 2020-02-28:
Here the cross-compiler:
https://dl.openipc.org/SDK/MStar/MStar-MSC3XX-SDK-toolchain.tgz
You could use my sources and edit directly the busybox .config to create another busybox binary,

I can't point you to any instruction but ask me and I will try to answer.
For example: the recording procedure is a feature of the original firmware.
All the control features (that I know) are exposed in the app and in the configuration web page:

- Recording without Cloud
- Save video when a motion is detected
- Detection sensitivity

naeloob, 2020-02-28:
Thank you very much for the info.
If used the docker enviromnent and i've succesfully compile busybox with swap support...
...now what don't have support is the kernel... :(
Compile, and test, the kernel are bigger words!!! Maybe i will try :)

About the files on /tmp/sd/record : 
 - I cannot control when it must record, or not. I tested with motion detector, without... i cannot trigger the motion detector, or very few times. And with any configuration it records all the time. How do you have it configured and whats the behavior.
 - Is there any way to make the files longer, not just 1 minute  (about 11Mb)
 - Is there any way to control the quality or codec parameters of the MP4 files

roleoroleo, 2020-02-29:
Build the kernel is a problem.
No sources and no /proc/config...

I don't know if it's useful but you can send a motion start event using ipc (also stop events).
I didn't change the motion behavior, I only replicated the controls you have on the app.
I think there is now way to change the file size, video duration or quality (for my knowledge).

naeloob, 2020-02-29:
The sources of the kernel are with the SDK.

Thank you for the info i will make some test.

roleoroleo, 2020-02-29:
Yes but the SDK is generic for the platform.

naeloob, 2020-02-29:
I think i could work...
If i compile the kernel, could you help me to boot it?

roleoroleo, 2020-02-29:
Yes, I will investigate the mtd tomorrow.

naeloob, 2020-02-29:
ok, thank you.
I'll report on progress.

roleoroleo, 2020-03-01:
On my cam the kernel is /dev/mtd1
It's a normal uboot format.
I suggest you to make a backup of the flash if you can.

naeloob, 2020-03-01:
Looking at a boot log we have : 

`+++++++++++++++++++   check one.bin    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one.bin 
    one.bin Not exist(fatsize err) 

+++++++++++++++++++   check one_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 one_y25 
    one_y25 Not exist(fatsize err) 

+++++++++++++++++++   check uboot_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 uboot_y25 
    uboot_y25 Not exist(fatsize err) 

+++++++++++++++++++   check kernel_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 kernel_y25 
    kernel_y25 Not exist(fatsize err) 

+++++++++++++++++++   check sys_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 sys_y25 
    sys_y25 Not exist(fatsize err) 

+++++++++++++++++++   check home_y25    +++++++++++++++++++ 
------>setenv filesize 0 
------>fatsize mmc 0 home_y25 
    home_y25 Not exist(fatsize err) 
`
Is this the way your firmware install works? If uboot finds these files it will flash?
Do you have any boot log from a 9FUS 4.5.0 (y203c)?
Do you have any boot log that finds any of these files? to see what it makes with them. 
Meanwhile i will dig on the uboot srcs.

Regards

roleoroleo, 2020-03-01:
Yes, this is the way my fw install.
At the moment I use only sys and home but you could try kernel also.
If uboot is ok you can reflash your original fw (if you have a backup of the kernel).

naeloob, 2020-03-01:
Thank you!! 

I've make a backup of the original kernel with : 

dd of="/tmp/sd/kernel.bin" if=/dev/mtd/mtd1 bs=16k conv=sync

So i think we have all to test kernel safety.
Just we need to compile some of them :)
I hope to have enought time tomorrow to try it.

Regards

naeloob, 2020-03-02:
Just a little update : I have a new Kernel Image. 
Tomorrow i will test it.

naeloob, 2020-03-03:
Well... It really flash the kernel just having a file called kernel_y203c on the / of the sd...
Now let's see if it can recover a briked cam :)
.
.
.
Recovered!!

I think i will need to open the cam and wire it to PC to see what's going on.
But it will have to wait more time...

roleoroleo, 2020-10-23:
Closed due to inactivity.

closed 2020-10-23 by roleoroleo

upais, 2023-12-25:
@roleoroleo, @naeloob 
Hi!
Do you have any SDK left? The link above shows 404.
I also want to add swapon/mkswap.

roleoroleo, 2023-12-28:
https://drive.google.com/file/d/1nEPkKyf7Xd8vsDFt1E1HGsxW4zS75d4t/view?usp=drive_link
Let me know when you complete the download.

upais, 2023-12-28:
Done. Thank you.

roleoroleo, 2023-12-28:
About swap, MStar doesn't have kernel support.
It's not possible to add it.