roleoroleo/yi-hack-MStar #98: Restart problem / Onvif question

Chapapa1, 2020-03-02:
Hey guys, 

thanks for this package @roleoroleo. it worked like a charm to flash my 9bus and BFUS cameras!
However i do have some questions...

1. restart: 

it seems that after every reboot it asks for a wifi connection. as soon as i show the camera the qr code it connects to my wifi and i can load the yi hack page. is there a way for the camera to connect directly to the wifi after a reboot? i couldnt find anything in here...

2. onvif

iam trying to integrate this camera into tinycam pro.. you can use custom xmls to add new cameras. i cant find the correct path to pan or tilt the camera. it expect sth like this: 

"<request name="PtzMoveRelLeft">/cgi-bin/action.cgi?cmd=motor_left</request>"

can you give me a hint on how to make that work?

it works with the app onvifier, however it automatically grabbed the ptz information and i wasnt able to find out where and how..

thanks in advance!

roleoroleo, 2020-03-02:
1. Try to search in closed issues. I remember something similar.
2. Why you don't configure the camera as onvif: the ptz will work automatically.
Otherwise you can use `http://192.168.1.242:8080/cgi-bin/ptz.sh?dir=left`

hagenuck1, 2020-03-02:
Wifi issue: 
You need to use the „Yi Home“ app to connect and NOT the „Xiaomi Home“ App. Had the same issue first. 

When connecting using the Yi home app the app continues it’s configuration and then is available in the app. If not it’s not completely configured.

Chapapa1, 2020-03-02:
Thanks for the replies:
@hagenuck1: i am using the yi home app. no luck :(
@roleoroleo:   your suggestion no.2 works perfectly :) - the url suggestion that is... 

when i want to use the onvif template that is already present in the tinycam pro cmaera list, it does not work. it shows me that i have ptz support, that i am running the yi hack etc., but it doesnt show me any pictures/streams. i think it has something to do with the /ch0_0.h264 part and the port being before the slash.. 

here is my custom_vendors.xml (info about [here](https://tinycammonitor.com/custom.html)):

<?xml version="1.0" encoding="UTF-8"?>
<vendors>
    <vendor name="MyYi">
        <model name="MyYi Advanced PTZ" defaultPort="8080" defaultRtspPort="554">
            <request name="Snap">/cgi-bin/snapshot.sh?res=low</request>
            <request name="Push">/cgi/video.mjpg</request>
            <request name="RTSP">/ch0_0.h264</request>
            <request name="PtzMoveRelLeft">/cgi-bin/ptz.sh?dir=left</request>
            <request name="PtzMoveRelRight">/cgi-bin/ptz.sh?dir=right</request>
            <request name="PtzMoveRelUp">/cgi-bin/ptz.sh?dir=up</request>
            <request name="PtzMoveRelDown">/cgi-bin/ptz.sh?dir=down</request>
        </model>
    </vendor>
</vendors>

roleoroleo, 2020-03-03:
Is cloud enabled?

Chapapa1, 2020-03-03:
On one it is disabled on the other it is enabled. However, the cameras are blocked from accessing the internet

roleoroleo, 2020-03-03:
You have to complete the pairing process before to enalbe the hack.
Enable the cloud, disable all the hack services (except httpd) and retry to pair with the app.

Chapapa1, 2020-03-03:
Ah OK.. I will try that.. Do they need an internet connection?

roleoroleo, 2020-03-03:
Yes. The cam must be connected to the Yi server for the pairing process.

closed 2020-04-16 by roleoroleo