roleoroleo/yi-hack-MStar #101: Turning off motion detection is impossible

Ne00n, 2020-03-04:
Hey,

I recently flashed the newest Firmware on my YYS2016.
Besides the already reported hostname canno't be changed bug.

I found 2 more.
When I uncheck the checkbox for "Save video when a motion is detected", it still saves Videos on the SD Card file when motion is detected.

Also when I try to block the saving's of files, by set "SD free space (%)" to 99% it still saves Video's to the SD Card.

Any idea on this?

roleoroleo, 2020-03-04:
This is the correct behavior.
If enabled, video will be saved only when a motion is detected.
If disabled, video will be always saved.

Ne00n, 2020-03-04:
Could you add an option to disable it completely?
Since I just need a RSTP stream and nothing else.

Killing the process seems to disable it but I would need to do that manually at each reboot.

roleoroleo, 2020-03-04:
Disable "Recording without cloud".

Ne00n, 2020-03-04:
Oh yea I see, thanks

closed 2020-03-04 by Ne00n