roleoroleo/yi-hack-MStar #103: Updated to 0.2.8 but in the terminal appears 0.2.3 Version

sonnar1800, 2020-03-08:
I have updated the camera to version 0.2.8 but in terminal it appears yi-hack for MStar platform - v. 0.2.3

I need to do a factory reset and then reinstall the 0.2.8 firmware version?

sonnar1800, 2020-03-08:
after having installed the version I have been downloaded alone to the source version 0.2.3

sonnar1800, 2020-03-08:
Nor does the Motion Event work for me in version 0.2.3 or 0.2.7, not in 0.2.8, why can it be?

roleoroleo, 2020-03-09:
Which version do you see in the status page of the web interface?

sonnar1800, 2020-03-09:
after updating, version 0.2.8 but then I don't know why I go back to 0.2.3

Now I have updated again and it already appears well on the web and in the terminal.

but the Motion event option still doesn't work

roleoroleo, 2020-03-10:
What do you mean with "doesn't work"?

roleoroleo, 2020-07-17:
Closing without response from issue filer for >14 days.

closed 2020-07-17 by roleoroleo