roleoroleo/yi-hack-MStar #109: nas recording

divemasterjm, 2020-03-12:
hi is there any way/option to record on NAS / Directory? would be very useful to have an option where to decide where we want to record.

roleoroleo, 2020-03-12:
The normal way would be nfs but it is not supported.
You can only export the saved files after creation.

divemasterjm, 2020-03-12:
Anyway i dont have videos recorded on se card when motion is detected, can you please help me?

roleoroleo, 2020-03-13:
Did you configure the cam correctly?
Cloud feature is enabled or disabled?

divemasterjm, 2020-03-13:
Cloud is diaabled, sd is the same I insert with hack, didn’t remove. Do you mean about configure in yi app?
By the way is the hack working removing ads after?

JuanMa

roleoroleo, 2020-03-14:
Did you enable "Recording without Cloud"?

divemasterjm, 2020-03-14:
Yes

Saludos
Juan Manuel

roleoroleo, 2020-03-15:
If you enable Cloud feature, does it work?

divemasterjm, 2020-03-15:
Nop, also yi app tells me sd card needs to be formatted on both cameras with hack, I didn’t try it cause i dont know if hack works without sd.

Saludos
Juan Manuel

roleoroleo, 2020-03-15:
The hack will works even without sd card.
Please, keep cloud enabled and try to configure the app until sd recording works.
After that, disable cloud feature.

closed 2020-08-20 by roleoroleo