roleoroleo/yi-hack-MStar #110: Custom firmware refusing to flash.

xfceKris, 2020-03-14:
I feel like I'm doing something wrong, but I'm following the directions to the letter. I've downloaded and extracted the latest .tgz for my 6fus cameras, format the 8gb sd card using the yi app (as per a recommendation from another issue here). Every time I reboot, I get a normal boot up procedure. I've checked tons of open and closed issues to see if anyone else had the same issue, and found nothing of note except for a mention of firmware version being 4.6. Currently, my cameras are running the stock firmware version 2.1.0.0E. I've seen nothing here mentioning this firmware version, and I checked the yi website for updated firmware, and see that this is the latest version. I feel like I'm simply missing something, but I know I've read the documentation thoroughly.

roleoroleo, 2020-03-14:
I think your cam is not MStar based.
Check if you can use https://github.com/TheCrypt0/yi-hack-v4

xfceKris, 2020-03-15:
That's a no-go. I've tried V4, V6, and MStar. None of them flash. They all have the exact same behavior. That is, a normal boot up and connection to wifi.

rogerb831, 2020-03-15:
I'm having the same experience on a BFUS I just bought yesterday.

rogerb831, 2020-03-16:
Actually my issue seems to be related to issue #91

roleoroleo, 2020-03-16:
If it's the same issue, your cam is not MStar based.
So this hack is not compatible.
You can check it connecting a serial adapter and dumping the log.

xfceKris, 2020-03-16:
If it's not MStar based, then one of the other hacks should work either V4 or V6, and neither one of those works either. Could it be that the manufacturer is simply mix and matching hardware and software as a way to block these hacks?

How would I go about connecting a serial adapter to dump logs? And what would another solution be if I can't get that done? simply try different hacks until one works (or bricks the camera)?

roleoroleo, 2020-03-16:
See the wiki for the dump:
https://github.com/roleoroleo/yi-hack-MStar/wiki/Dump-your-backup-firmware

Matchlighter, 2020-03-17:
I'm running into this with a 9FUS as well. I cracked it open and it is indeed not an MStar chip - it doesn't have a discernible manufacturer, but it has a logo and `QG2101A`. I'll see about trying to find the UART lines and getting some logs.

[image]

roleoroleo, 2020-03-17:
Use TX and RX.
And see this issue:
https://github.com/roleoroleo/yi-hack-MStar/issues/91

roleoroleo, 2020-03-22:
@Matchlighter 
Could you take a picture of the other side?

Matchlighter, 2020-03-23:
[image]

xfceKris, 2020-04-04:
Any update as to weather or not this will be updated to work with the 6fus with stock firmware 2.1.0.0E?

roleoroleo, 2020-04-04:
Please connect a serial adapter and dump the log.
I will check which platform is.

roleoroleo, 2020-04-20:
See here:
https://github.com/roleoroleo/yi-hack-Allwinner/

CaptianNemo, 2020-04-29:
If you re-read the OP, you'll see he has a 6FUS and not the 9FUS that Matchlighter has. You will also see his camera reports running 2.1.0.0E and not 8.2.0.0A which is Matchlighter's version.
I also have a 6FUS running 2.1.0.0E and have taken it apart, it is not the same as Matchlighter's. It does have an MStar chip. I belive this is a valid issue and should be re-opened.
[image]
[image]
[image]
[image]

roleoroleo, 2020-04-29:
I need a log of the boot to study this model.
At the moment is not supported.

CaptianNemo, 2020-04-30:
I'll see if I can get those logs for you. Might be about a week though, I ordered a flux pen for the tiny solder job, and amazon says it'll take about a week to get here.

CaptianNemo, 2020-05-07:
[Yii_log_boot.log](https://github.com/roleoroleo/yi-hack-MStar/files/4596131/Yii_log_boot.log)

roleoroleo, 2020-05-08:
It seems a MStar based cam with a different wifi chip.
The model prefix is y23.
I could try to add this model to the hack but I need the dump of the flash.

CaptianNemo, 2020-05-08:
Dumping the flash is covered in https://github.com/roleoroleo/yi-hack-MStar/wiki/Dump-your-backup-firmware under "dump sys_XXX" and "dump home_XXX", correct?

roleoroleo, 2020-05-08:
Yes.

CaptianNemo, 2020-05-13:
[new_6fus_dump.zip](https://github.com/roleoroleo/yi-hack-MStar/files/4625162/new_6fus_dump.zip)

roleoroleo, 2020-05-14:
Try this beta:
https://drive.google.com/open?id=1yb22W8clLFR8sDAjiwyDctTM4uZSzV83

CaptianNemo, 2020-05-14:
Flashed successfully and camera's webpage shows up. Is there anything you want me to test while I still have wires soldered to it? I really appreciate you taking time to add this model.

roleoroleo, 2020-05-15:
Please, try the main feature:
1. RTSP
2. Snapshot
3. ONVIF
4. Mqtt

CaptianNemo, 2020-05-17:
RTSP - Working normally.
Snapshot - Working normally.
ONVIF - I think it's working, but the app I used (lingodigit) is having problems. 
Mqtt - I don't think I can test this, it seems to require a raspberry pi. The only raspberry pi I own is a zero (regular, not zero-w), which the home assistant website says is not recommended.

roleoroleo, 2020-05-18:
You can test mqtt with a generic broker, home assistant is not necessary.
But if you can test it, it doesn't matter.
I will add y23 support in the next release.

closed 2020-05-18 by roleoroleo