roleoroleo/yi-hack-MStar #112: update firmware through webbrowser

MoiZie, 2020-03-16:
Would it be possible to somehow update the firmware through the webbrowser? 
I have already tested scp-ing the latest firmware files to /tmp/sd and rebooting; which is working without problems. It would be a nice feature to have. 

The latest release file can be downloaded by using:
`cd /tmp/sd | curl -s https://api.github.com/repos/roleoroleo/yi-hack-MStar/releases/latest | grep '"tag_name":' | sed -E 's/.*"([^"]+)".*/\1/' | xargs -I {} curl -sOL "https://github.com/roleoroleo/yi-hack-MStar/releases/download/"{}"/y203c_"{}'.tgz' 
`

If you could implement a check on the main statuspage:
'firmware version 0.2.8'
and then a line with
latest available version  'latest i.e. 0.2.9'

the number could be found by using only

`curl -s https://api.github.com/repos/roleoroleo/yi-hack-MStar/releases/latest | grep '"tag_name":' | sed -E 's/.*"([^"]+)".*/\1/'`

And could then be checked against the currently installed version

Obviously you would have to build with curl, but then I reference this issue: https://github.com/roleoroleo/yi-hack-MStar/issues/77 in which it appears you already have a working version.

roleoroleo, 2020-03-16:
I'll take a look.
There is an additional problem: I should save the configuration and restore it after the update.
Is not so simple.

MoiZie, 2020-03-16:
You could just integrate the command as far as I can see. just 'cp' the .conf files to /tmp/sd, which is read only for the system after reset/upgrade. Then, upon boot, check if /tmp/sd/conf exists and if /etc/conf exist, then 'cp' the conf files back if they do not exist?

roleoroleo, 2020-03-17:
I'm working on this issue.
Wait for the next release.

majkers, 2020-03-18:
One more question. Would it be possible to store and restore configuration automatically during update via web? Now I have to manually export my config before update and later import it.

roleoroleo, 2020-03-18:
Yes, it's possible.

steef84, 2020-04-06:
Oke, your new version 0.3.0 updated for 4 cams from 0.2.9 Y203C fw 4.5.0 without a hitch via webinterface. However the fifth Y203C fw 4.5.0 cam updated (I think), but the webinterface doesn't respond anymore. YI cam is visible via YI cloud though and working, so I think somewhere the settings restoring corrupted. Also no ssh / ftp / telnet working, as I disabled them in the interface 0.2.9. Any things I can try for now, otherwise I just downgrade and update again for this cam?

roleoroleo, 2020-04-06:
Downgrade and upgrade again.

MoiZie, 2020-04-21:
Upgrade from 0 29 to 0.30 and now 0 31 is without problem, only the hostname is not saved on firmware upgrade. Could you fix this as well?

roleoroleo, 2020-04-21:
I will fix it in the next release.

MoiZie, 2020-05-12:
Can you add the saving of hostname on upgrade? It appears this has not been fixed yet.

roleoroleo, 2020-05-12:
This thing will drive me crazy.
I think I tried to fix it at least 3 times.

arthur0412, 2020-06-05:
Hostname on upgrade from 0.3.3 to 0.3.4 ist still not saved.

roleoroleo, 2020-06-05:
If i'm not wrong the problem is solved in 0.3.4.
The procedure will work when you will install 0.3.5.
Or, if you want to check it now, edit the version file with vi and reapply the same upgrade.

arthur0412, 2020-06-05:
ok, thank you for the info. I test this with next release ;o)

phrogg, 2020-09-02:
for me the new update worked without problems.

closed 2020-09-03 by roleoroleo