roleoroleo/yi-hack-MStar #115: I've bricked my 9FUSY214TP

tdiz, 2020-03-18:
Hello
I applied FW 0.2.8 release update for my 9FUSY214TP. After that camera had additional options and work well (thx to author).
I tried to implement "copy to Yandex.disk" script follow by [4pda manual](https://4pda.ru/forum/index.php?s=&showtopic=814875&view=findpost&p=89857248) but something was wrong. I think after this command `chmod +x /etc/init.d/S90yandexcp` and rebooting. Now it doesn't respond at all only yellow light is always on.
Unfortunate I didn't make a dump.
I have only original FW from seller (attached [home_y23m.zip](https://github.com/roleoroleo/yi-hack-MStar/files/4351420/home_y23m.zip) ) which was sent me separately. I have TTL and CH341A EEPROM programmer.

My question: Is it possible to repair it in this case by TTL or not?

roleoroleo, 2020-03-18:
Try to downgrade to 0.2.7 and then again upgrade to 0 2.8.

tdiz, 2020-03-19:
Yes! It works now! THX! What is the trick? What did I wrong?

roleoroleo, 2020-03-20:
I don't know what you did.
But you can't reload the fw if the versions are the same.
The bootloader check the hash of the file that you want to upgrade.
So you need to use a different file.

tdiz, 2020-03-20:
thx for explanation

closed 2020-03-20 by tdiz