roleoroleo/yi-hack-MStar #116: Y25 becomes bricks after updating official firmware.

imakou, 2020-03-19:
Hi Guys,

I don't know where I can get help.
I hope you don't mind that I am asking the questions here.

So, yesterday, my two Yi home camera3 did back to back after updating official firmware.
The worth thing is that the reset function doesn't work.
The camera just restarts itself with "click" sound.
I cannot rest and repair my cameras.

Does anyone know how to fix the camera in this situation?
Thanks.

roleoroleo, 2020-03-19:
Please, give me the version number you installed.

imakou, 2020-03-19:
> Please, give me the version number you installed.

I'm not sure the version I installed.
Because I lost the paired cameras in my app since this failing.
But, I check the website, it seems that the latest version for Y25 is `4.2.0.0H_201909041620`.

Thanks.

roleoroleo, 2020-03-20:
You could try to reload the firmware.
Download the file, copy it on a SD card, insert in the cam and start.
Try with 4.2.0H.
If the firmware upgrade doesn't start try with the 4.2.0F.

http://download.eu.xiaoyi.com/yifirmware/smarthomecam/familymonitor-y25/4.2.0.0F_201905241332home_y25m
http://download.eu.xiaoyi.com/yifirmware/smarthomecam/familymonitor-y25/4.2.0.0H_201909041620home_y25m

Please, let me know when you "copy & paste" this message because I will remove the links.

imakou, 2020-03-20:
> You could try to reload the firmware.
> Download the file, copy it on a SD card, insert in the cam and start.
> Try with 4.2.0H.
> If the firmware upgrade doesn't start try with the 4.2.0F. [...]

Copied.
Will try when I get my sd card. It's on the way to my home.
Thanks.

imakou, 2020-03-20:
Update:
I just went to a local store to buy a 64GB Class10 microSDXC.
And I format it as fat32 and put the files you recommend into the sd card.
BUT, it seems nothing happen....

roleoroleo, 2020-03-20:
Mmmmh
Probably an older sd it's better.
Do you have something smaller?

imakou, 2020-03-20:
No. I don't.
Should I buy another 16GB sd card?
The 16GB sd card's price is the same as 32GB, lol.
BTW, I can see the camera wrote `log` folder to my SD card.

imakou, 2020-03-20:
I think that I may need to return these two cameras.
That is very frustrating because I really need cameras to monitor my baby.
Starting to order Yi Camara 1080P version, not Camera 3.
It's very hard to get information about Camera 3 from Google.

roleoroleo, 2020-03-20:
The cam sees the sd card when the fw is loaded.
But the upgrade procedure is started from the bootloader and you don't know if it sees the card.
Try to reformat the card from a linux machine.

imakou, 2020-03-20:
I re-formated my card by MacOs.
The card shows its format is `DOS_FAT_32`.
Is it expected? Thanks.

roleoroleo, 2020-03-20:
I think it's correct.
Copy the file on the card.
Rename it home_y25.
Insert the sd card.
Reboot.

imakou, 2020-03-20:
Hi,
I'm thinking of buying another 16GB SDXC.
But, I don't know if we can fix my problem by re-loaded firmware?
Is my case fixed by re-loaded firmware..?

Thanks.

imakou, 2020-03-20:
Roleo,
I very appreciate for the help.
It seems that something is changed.
The camera doesn't make the "Click" sound after I renamed the file name.
But, the camera doesn't reboot....

roleoroleo, 2020-03-21:
Ok, try this:
https://drive.google.com/open?id=1GC9Aw1SvOpTi6fma52O5Ub2hjAGhh-Pa

Copy the file on the card.
Don't rename it.
Insert the sd card.
Reboot.

imakou, 2020-03-21:
Roleo,
The file WOKRS!
My cameras are ALIVE!!!!!!!!!
THANKS VERY MUCH!!

But, what happens? How can this file work?

roleoroleo, 2020-03-21:
Because this file completely rewrite your partition and renew all the content.

imakou, 2020-03-22:
Roleo,
How do you know that I should use that file to rewrite my camera?
 I could not find a file which fits my camera, but yours works.

roleoroleo, 2020-03-22:
This file is a dump of a real camera (same model as yours).

imakou, 2020-03-23:
> This file is a dump of a real camera (same model as yours).

I see.
Thanks for your help.
BTW, I couldn't get the firmware from YI official website which doesn't make sense :(

roleoroleo, 2020-03-23:
You can download a firmware from Yi site but it's only a partial update.
It isn't useful for recovery purposes.

imakou, 2020-03-24:
Yi tech just announces the mistake they made and provides a solution.
It's the same as you provided.

[Link](https://forum.yitechnology.com/t/yi-home-camera-3-firmware-revision/223?utm_source=NA+Audience&utm_campaign=84d7c09af3-EMAIL_CAMPAIGN_2020_03_23_03_29&utm_medium=email&utm_term=0_fa305f78ea-84d7c09af3-95687410&mc_cid=84d7c09af3&mc_eid=69ec5d6c13)

closed 2020-03-24 by imakou

roleoroleo, 2020-03-24:
No, is not the same but probably it works.