roleoroleo/yi-hack-MStar #125: Feature request: mqtt video list before motion stops

agsola, 2020-03-27:
Sometimes we want to just get the first video or videos available ASAP once motion start, and don't wait until motion stops.

Could it be possible to send a video list message once the first clip/video is available when motion starts? 
Otherwise we have to wait until motion stops to see the last video file listing,

roleoroleo, 2020-03-28:
I miss the trigger to start the action.
I have a trigger when the motion starts (but the file doesn't exist yet) and a trigger when the motion stops.
I could send a message when the motion starts but you have to poll the link until the file is available.

agsola, 2020-03-28:
I will try to upload another solution this weekend.

agsola, 2020-04-05:
I had the solution. But unfortunately only inside the cam.
I've just updated and lost it, as it was stored in the script folder.

Is there any way to recover files from previous build?

roleoroleo, 2020-04-05:
No, sorry.
The partition is overwritten.

agsola, 2020-04-09:
I attach kind of a dirty solution. It is a bash cgi script that retrieves either the name, the URL or the video itself. You can choose the oldness or to wait for a new one.

# Use. Available variables:
# "oldness" variable. Defines which video to retrieve. 
# - "0" or "1" means latest already avaiable.
# - greater than "0" specifies the oldness, so "3" means the third oldest video. Only looks into the current hour folder. So if value is greater than the videos in the latest hour folder, the last one will be sent.
# - negative value (e.g. "-1") means to wait for the next available video. Only works if a video is already being recorded, max. time to wait 80 seconds.
# "type" variable. Defines what to retrieve.
# - "1" (default). Gets relative name in the format of DIR/VIDEO.mp4
# - "2". Gets full URL video.
# - "3". Gets the video itself as video/mp4 inline.
# - "4". Gets the video itself as video/mp4 attachment.
# Examples of use:
# http://IP:PORT/cgi-bin/getlastrecordedvideo.sh?oldness=-1&type=4 -- Wait for a new video and sends it as an attachment.
# http://IP:PORT/cgi-bin/getlastrecordedvideo.sh?oldness=-1&type=4 -- Wait for a new video and shows it inline in the browser.
# http://IP:PORT/cgi-bin/getlastrecordedvideo.sh?&type=1 -- Send the relative route of the last available video.
# http://IP:PORT/cgi-bin/getlastrecordedvideo.sh?&oldness=2&type=2 -- Send the URL of the second to last available video.

I hope you can find it useful and add to the main branch. 
Code could be better, but I had to program twice and was a bit "bored" the second time.

[getlastrecordedvideo.sh.txt](https://github.com/roleoroleo/yi-hack-MStar/files/4457739/getlastrecordedvideo.sh.txt)

roleoroleo, 2020-04-10:
Thank you.
I will try it tomorrow.

roleoroleo, 2020-04-18:
Hi, I found the time to try your script and I added it to the main.
Thank you for your contrib.

closed 2020-04-19 by roleoroleo