roleoroleo/yi-hack-MStar #127: High load average and corrupted snaphost

rafal98, 2020-03-29:
Hi,

My main usage is to drive this camera using Domoticz, so only focus on the snaphot feature, but image are heavily corrupted with random glitch. I suspect that the high load average may cause this behavior.

At the moment I just use the snapshot HTTP request from Firefox (not Domoticz)

Here my config:

Firmware Version | 0.2.9
Base Version | 4.2.0.0H_201909041620
Model Suffix | y25
Serial Number | 9FUSY25G0HXX71191010
Local Time | Sun Mar 29 15:14:55 UTC 2020
Uptime | 1d 5h 32m 59s
Load Average | **1.56 1.44 1.42**

Disable Cloud: On
Recording without Cloud: Off
RTSP: Off
MQTT: not configured
Switch on/off the camera: Off
Save video when a motion is detected: Off

[image]

Here a "top" from ssh:
```
CPU: 21.9% usr 14.8% sys  0.0% nic 63.0% idle  0.0% io  0.0% irq  0.2% sirq
Load average: 1.79 0.82 0.33 1/86 1739
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
  187     1 root     S    49436 80.8   0 28.8 ./rmm
  178     1 root     S     4036  6.6   0  0.4 ./dispatch
    7     2 root     RW       0  0.0   0  0.4 [rcu_preempt]
 1677  1620 root     R     2064  3.3   0  0.2 top
 1617   890 root     S     2012  3.2   0  0.2 dropbear -R
    3     2 root     SW       0  0.0   0  0.2 [ksoftirqd/0]
  489     1 root     S     2256  3.6   0  0.0 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_sup
  916     1 root     S     2204  3.6   0  0.0 ntpd -p pool.ntp.org
 1620  1617 root     S     2064  3.3   0  0.0 -sh
  868     1 root     S     2060  3.3   0  0.0 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
  559     1 root     S     2060  3.3   0  0.0 udhcpc -i wlan0 -b -s /home/app/script/default.script
  890     1 root     S     1728  2.8   0  0.0 dropbear -R
  879     1 root     S     1660  2.7   0  0.0 pure-ftpd (SERVER)
  177     1 root     S     1508  2.4   0  0.0 ./log_server
   39     1 root     S      408  0.6   0  0.0 /bin/ueventd
    1     0 root     S      384  0.6   0  0.0 /init
```

Can I miss something ? Wrong parameters ? What is rmm process ?

roleoroleo, 2020-03-29:
Your config is ok.
Sometimes this problem occurs but it should be infrequent.
I don't know how to solve it.
Probably we should rewrite the snapshot algorithm.
rmm is the main yi process that read the stream from the kernel. The cpu use is normal.

roleoroleo, 2020-03-29:
Does Domoticz use the http snapshot feature?
Or extract image from rtsp stream?
Do you have the same problem if you call the url from a browser?

rafal98, 2020-03-29:
Thank you for your fast answer

Snapshot attached was captured by http://192.168.2.47:8080/?page=snapshot
But most of my tests were done on the direct url http://192.168.2.47:8080/cgi-bin/snapshot.sh
And in my config, the corrupted image or color issues are very common, but only on 1080p. If lowres, it's ok

FYI Domoticz does not support RTSP, but "simple" snapshot. It's perfect for my usage.

Question: can I remove the SD card after the flash ? Or the SD card is mandatory now ? Perhaps my SD card is bad, or slow ?

An congrats for this useful hack

roleoroleo, 2020-03-29:
Try this:
https://drive.google.com/open?id=1Vwn4Bb-BGKm1WWi1cYPD0hqlxb3ZZlXx

You can remove the SD, is not mandatory.

rafal98, 2020-03-30:
It seems to works better but not perfect. I generate a dozen of snapshots and 2 where truncated (the bottom of the picture was missing). Not the same visual corruption.

roleoroleo, 2020-03-30:
Could you post a sample?

rafal98, 2020-03-30:
Of course. The most important is to have a reliable snapshot, but it is possible to select the jpg compression ? Because I think it's a bit too aggressive : a 1080p snapshot that use only 400KB, even without issue the result is a bit fuzy

[image]

Note: the corrupted picture does not show the same result after been imported on github. Here a screenshot of my viewer (fastone image viewer)

[image]

File size on SD show that the corrupted snapshot has an abnormal size:
-rwx------    1 root     0           411444 Mar 30 09:07 test1.jpg
-rwx------    1 root     0           409611 Mar 30 09:07 test2.jpg
-rwx------    1 root     0           385505 Mar 30 09:08 test3.jpg
-rwx------    1 root     0           397648 Mar 30 09:58 test4.jpg
-rwx------    1 root     0           **229376** Mar 30 09:58 test5.jpg
-rwx------    1 root     0           392642 Mar 30 09:59 test6.jpg
-rwx------    1 root     0           322767 Mar 30 18:42 test7.jpg
-rwx------    1 root     0           313405 Mar 30 18:42 test8.jpg

pReya, 2020-03-31:
Same problem here. Images created either via API call or the Web UI have artefacts or are truncated most of the time, just like the examples by @rafal98. Even when the image is created without errors, the image quality is much worse than when using the original Yi Home App.
[image]
Left side is Web UI. Right side is Yi Home App.

roleoroleo, 2020-03-31:
Please try this:
https://drive.google.com/open?id=1spgOt1E2S7Vdp4W8KX7oNEMqZwIbGU5p
Added a integrity check and increased jpeg quality from 90 to 100.

pReya, 2020-03-31:
@roleoroleo Not sure what to do with this file. It's a binary, right? Do I just put it on the SD card? How do I call it then? Or is this supposed to be run locally on my machine?

roleoroleo, 2020-03-31:
Copy it to /home/yi-hack/bin/ and overwrite the current version.
You can use ftp protocol.

rafal98, 2020-03-31:
@pReya download the imggrabber file (it's a binary) and use scp to copy to the camera. If you have an SD card, you can copy on it directly or use scp to put in /tmp/sd

@roleoroleo, I just do some runs, but my camera is located outside and it's the night, so result is not terrible, but there is no corruption at the moment :-) File sizes are now about 570/600KB

pReya, 2020-03-31:
I tried with the new imggrabber bin and it works a lot better. No large artefacts anymore. But there's still a consistent "artefact line" in the very bottom of the image. You can also see this in the left side of the captured image above. There is a row of 3-4 weird pixel at the bottom. This does not happen in the Yi Home app. Anyway, thanks for the new bin. It's already great improvement.

roleoroleo, 2020-04-01:
All the shots?

roleoroleo, 2020-04-01:
Solved the problem of last 8 lines.
The original images is 1920x1088. There was an error during cropping operation to 1920x1080.
https://drive.google.com/open?id=1H_Lx8Zqrlm_65_ccSJYSjdxmEJF30BHu

pReya, 2020-04-02:
Thanks so much @roleoroleo. This fixed the artefact lines for me. Works like a charm now. I guess this issue should be closed, then.

roleoroleo, 2020-04-02:
Yes, in the next release.

rafal98, 2020-04-02:
Hi, thx a lot @roleoroleo. You can close this issue

Just an idea to put in the TODO list: it will be useful to add an option to put the date+time in a corner of the snapshot

roleoroleo, 2020-04-02:
Do you know a specific software?

rafal98, 2020-04-02:
I was talking about option/feature like this old camera:
[image]

I just find some tools like "convert" from ImageMagick, but it's probably a too big software to embed

pReya, 2020-04-02:
@roleoroleo I just discovered that the RTSP Stream is still in 1920x1088. It doesn't show any artefacts, but I think it might still be good for compatibility reasons to crop it to 1920x1080, maybe.
[image]

roleoroleo, 2020-04-02:
@rafal98 Too big and too many dependencies (libjpeg, freetype, fontconfig...).
@pReya We need a lot of hw resources. I think the original format in the yi app is also 1920x1088.

roleoroleo, 2020-04-10:
I found a simple source for watermarks:
https://github.com/lindenis-org/lindenis-v536-package/tree/master/allwinner/camerademo/src
The main page says "opensource" so I will use it.
It's simple and use low memory. I tested it and it works like a charm.
Next release...

rafal98, 2020-04-10:
Ho ! Very nice :-)

closed 2020-05-07 by roleoroleo