roleoroleo/yi-hack-MStar #128: The Yi Home app stops working with the cam

rabbitsf, 2020-03-30:
I have the Yi Dome 1080p. It's been working fine for around an hour after flashing with the h201c (0.2.9). But the iOS Yi Home app suddenly stops working when trying to access the cam. I tried resetting the cam a couple of times and adding the cam back but it's not successful. It always stuck after telling me wifi is connected. 
I can still access the cam from port 8080 from the browser. It's just the app that does not work. Any suggestion?

roleoroleo, 2020-03-30:
Enable cloud and retry.

rabbitsf, 2020-03-30:
> Enable cloud and retry.

Yes, enable cloud fixed it. Thanks very much, roleoroleo!

closed 2020-04-05 by roleoroleo

linsong837, 2020-04-11:
Isn't cloud enabled by default?
I have the same issue but enable cloud didn't fix it.

roleoroleo, 2020-04-11:
Cloud is enable by default.

rabbitsfschool, 2020-04-11:
> Isn't cloud enabled by default?
> I have the same issue but enable cloud didn't fix it.

I suggest you redo everything once again. 
Disable cloud, reboot the dome, access the dome from port 8080, enable the cloud, reboot the dome, add the dome to the app.