roleoroleo/yi-hack-MStar #131: Bug: No audio in RTSP (with or without "disable cloud")

agsola, 2020-03-31:
I've tried using the "Disable cloud" option.

But when I connect to RTSP using either VLC directly or through HA (using ffmpeg conversion) there is no audio available. I cannot access the audio in any way, just the video without audio. 
I've tried enabling the cloud again, and I can access the audio, but only through the official app, not through RTSP.

So, I can't access unless the audio through RTSP, just through the official app. 
I'm in 0.2.9 with 9FUS.

I don't know if this is a known issue, but I haven't found it in "issues", just "only audio" related issues.
And the original yi-hack seems to support audio, right? https://github.com/TheCrypt0/yi-hack-v4/

agsola, 2020-03-31:
I answer myself.
Audio is not available in RTSP, and yi-hack-v4 from TheCrypt0 is closed-source.

However, I've found a previous yi-hack with RTSP and just audio options: https://github.com/fritz-smh/yi-hack

Maybe that could be a start to find how to enable audio.

agsola, 2020-03-31:
Nevermind...

I've checked https://github.com/fritz-smh/yi-hack and it used a RTSP server that the manufactured provided in previous firmwares and then removed. But it is just a binary..., so no luck there...

roleoroleo, 2020-03-31:
https://github.com/roleoroleo/yi-hack-MStar/issues/32
https://github.com/roleoroleo/yi-hack-MStar/issues/21

The problem is where to get audio from the kernel, not how to stream it.
I tried with alsa but no luck.

agsola, 2020-03-31:
You should already know this, but mp4record seems to be the one recording the 60 seconds file. 
There is some info here: https://github.com/shadow-1/yi-hack-v3/issues/207

I wonder if it would be possible to reverse engineering the binary to check how the audio is got through the kernel, though it doesn't seem to be easy...

roleoroleo, 2020-03-31:
I already tried.
Too complicated for me.

agsola, 2020-04-01:
What did you find when using ALSA?
I'm trying to find more information, and I've found that the is a /dev/snd/ device folder, which seems to be audio related.
/dev/snd/pcmC0D0c seems to be the recording device and pcmC0D0p the playback (useful to send audio messages?).

I've also found some audio information in here: /proc/asound regarding ALSA. Is that from you?

And some links with information related to that:
https://bootlin.com/doc/legacy/audio/embedded_linux_audio.pdf
http://www.embeddedlinux.org.cn/essentiallinuxdevicedrivers/final/ch13lev1sec2.html

roleoroleo, 2020-04-01:
I compiled tinyalsa and tried tinycap and tinyplay beacuse you can find the library in the original firmware.
The alsa devices are busy during normal operation. You need to kill rmm to free the access to the devices.
When rmm is killed tinycap works and captures correctly from internal microphone.
Tinyplay doesn't work.

agsola, 2020-04-01:
Could make sense. rmm seems to be the main yi process that read the stream from the kernel. It is the one raising the messages when motion start, that launches the records... It keeps all resources busy.
According to @msmarks in here (https://github.com/fritz-smh/yi-hack/issues/58):
p2p_tnp is process handle data communicate with Mi home App
rmm is process to handle camera hardware device.
And that can be used to send sounds to the camera.
Furthermore, there is there some information about how to debug these files using IDA.

However, there should be a way to get data from rmm, since using the official app we can listen the sounds from the camera and record at the same time. The key should be the communication with rmm or derivates, right? Rather than trying to get it from the kernel directly, right?
Trying to "imitate" p2p_tnp that seems to get data from "rmm".

I guess I'm just saying obvious things you already knew :D

Regarding "rmm", I've found very little information. I don't know how much you know about the process.
Trying to invoque I get this error, even if I kill it before:
```
insmod: can't insert '/home/app/localko/ssp_ms41909.ko': No such file or directory
[MI_SYS_Init]MI version:MVX2####g5ddb6feMI##########[BR:I3/sz1109]#XVM
[MI_SYS_Init]MI log level:2
[MI_SYS_Init]MI log module:0xffffffff
[MI_SYS_Init]MI ISP status_check:FALSE
<49840 ms>	OMX[893]	OMX log level:1
<49841 ms>	OMX[893]	OMX log modularfilter :0
<49842 ms>	OMX[893]	enable memory leak check!
<49851 ms>	OMX[893]	Cannot open /lib/libOMX_ACODEC.so, error = /lib/libOMX_ACODEC.so: cannot open shared object file: No such file or directory
<49852 ms>	OMX[893]	Cannot open /lib/libOMX_ALSA.so, error = /lib/libOMX_ALSA.so: cannot open shared object file: No such file or directory
<49853 ms>	OMX[893]	Cannot open /lib/libOMX_AVQE_A.so, error = /lib/libOMX_AVQE_A.so: cannot open shared object file: No such file or directory
<49855 ms>	OMX[893]	Cannot open /lib/libOMX_ASPL.so, error = /lib/libOMX_ASPL.so: cannot open shared object file: No such file or directory
<49856 ms>	OMX[893]	Cannot open /lib/libOMX_VMFE.so, error = /lib/libOMX_VMFE.so: cannot open shared object file: No such file or directory
<49857 ms>	OMX[893]	Cannot open /lib/libOMX_VVHE.so, error = /lib/libOMX_VVHE.so: cannot open shared object file: No such file or directory
<49858 ms>	OMX[893]	Cannot open /lib/libOMX_VSPL.so, error = /lib/libOMX_VSPL.so: cannot open shared object file: No such file or directory
<49859 ms>	OMX[893]	Cannot open /lib/libOMX_VSRC.so, error = /lib/libOMX_VSRC.so: cannot open shared object file: No such file or directory
<49859 ms>	OMX[893]	Cannot open /lib/libOMX_IJPE.so, error = /lib/libOMX_IJPE.so: cannot open shared object file: No such file or directory
mi-info:	ISP_Wraper_init
mi-info:	bShrinkMem:1
mi-info:	_MI_OSD_Init_Prop gop_use:0
mi-info:	OmxComponentCreate: name = OMX.mstar.video_src.camera.vjpg
<49880 ms>	OMX[893]	Component not found with current ST static component loader.
mi-error:	 OmxComponentCreate:615 OMX_Wraper_GetHandle error:ErrorComponentNotFound
mi-error:	[_MI_VI_Init:1089] OmxComponentCreate err=ErrorComponentNotFound
mi-info:	_MI_VI_Deinit Exit
mi-error:	 _MI_VI_Init: init failed
MI_SYS_Init error = -1
rmm_init_isp error
hue, rmm main(), 7 
bitmapformat =a
w:1, h:1, bitCount:24 offset:54
infoHead.ciCompress:0
[MI_OSD_CreateUserWidget 1065] (3)para too big! [0 0]
mi-error:	_MI_VENC_GetFrameTimeOut 945: state:0
bitmapformat =a
w:1, h:1, bitCount:24 offset:54
infoHead.ciCompress:0
[MI_OSD_CreateUserWidget 1065] (0)para too big! [0 0]
mi-error:	_MI_VENC_GetFrameTimeOut 945: state:0

***** SMART BABYCRYING - V20160905.1 - *****
system reboot for get venc_0 stream failed!

```

However, in here:
https://github.com/fritz-smh/yi-hack/blob/master/sd/test/equip_test.sh
It seems that some parameters can be send to play sounds. But I couldn't make it work.

PieVo, 2020-04-07:
Please check https://github.com/PieVo/yi-hack-MStar

Not yet complete and producing valid firmware, the library is still in the wrong place:
https://github.com/PieVo/yi-hack-MStar/issues/2

Any ideas on how to fix that one?

roleoroleo, 2020-04-08:
I'm trying to apply your patch but I'm having some problems.
I have compiled the sources correctly and the programs are running.
But I cannot read the stream.
Do you have any suggestions?

PieVo, 2020-04-08:
So, if you replaced the rRtspserver and libtinyalsa.so in /home/ms the only
other change it the one in system.sh.

The “open” call on the pipe causes rmm to block, therefore ideally it would
start when the rtsp server starts. Unfortunately the rtsp server does not
start when there is no video. Sort of deadlock. To prevent this, I dump
some data from the pipe to kickstart rmm.

Actual reading of audio only happens when a client connects.

Hope this helps otherwise we need some more debugging.

roleoroleo, 2020-04-08:
I reinstalled from scratch and now it works.
Good job.
I noticed the issue you already posted https://github.com/PieVo/yi-hack-MStar/issues/3

PieVo, 2020-04-08:
> I noticed the issue you already posted PieVo#3

Good to hear! First thing I want to fix is the audio quality.. I bothers me
that the Yi app has better audio with less noise.

In general the Yi cameras have very good/low noise audio with good
sensitivity (compared to other chinese cameras).

roleoroleo, 2020-04-19:
News about the issue https://github.com/PieVo/yi-hack-MStar/issues/3 ?

PieVo, 2020-04-19:
Slow progress.. I was digging into the noise issue #https://github.com/PieVo/yi-hack-MStar/issues/2  and there seems to be some buffering issues in combination with the live555 code.

Eventually I expect this loss of audio is also buffer related. Once I have proper buffering and logging issue #2 should also fixable.

ryanamaral, 2020-05-05:
> News about the issue [PieVo#3](https://github.com/PieVo/yi-hack-MStar/issues/3) ?

@PieVo already fixed it!  💪

PieVo, 2020-05-05:
I’m still working on the noise issue too, with some filtering. If that’s
fixed the implementation is finished in my opinion.

roleoroleo, 2020-05-05:
I ported your changes in alsa lib (for the allwinner version) and it seems to work.

roleoroleo, 2020-05-08:
Do you agree if I add your changes into my tree?

PieVo, 2020-05-08:
Yes, no problem. The noise can be fixed at a later stage...

Will you do manual merging or shall I make a pull request?

PieVo, 2020-05-08:
https://github.com/roleoroleo/yi-hack-MStar/pull/171

roleoroleo, 2020-05-08:
I'm working on your patch with some minor changes to adapt to my last changes.
Which files should i use?
Audio subsession class under development or patched WAVAudioFileSource class?

PieVo, 2020-05-08:
The WAVAudioFileSource is the one to use.. it appeared I did not need the
subsession and only the presentation time needed a fix.

It was good practice and I did not want to throw away the subsession. So I
put that in development.

mimes123, 2020-05-09:
Hi i just tried installed the firmware some days ago. and saw this channel when searching for live video with audio in HomeKit. If i install your new firmware on my came can i then get the audio on the live video stream ?

roleoroleo, 2020-05-09:
You have to wait for release 0.3.2.

mimes123, 2020-05-09:
okay great i will wait thanks for the answer :)

mimes123, 2020-05-09:
i can now see it is released and i tried it out. but when i open the rtsp link in VLC there is no audio what am i missing ?

roleoroleo, 2020-05-09:
Did you enable it in configuration?

mimes123, 2020-05-09:
yes i choose high in rtsp audio

StreneSaco, 2020-05-10:
Hi roleoroleo, Thank you so much for this Yi-Hack, works fine. I can't find the rtsp audio option you mentioned. It doesnt appear on the web interface of the cam. 
Rest of the RTSP is working fine, thanks.

mimes123, 2020-05-10:
@PieVo what is the rtsp link for video with audio ?

roleoroleo, 2020-05-10:
The same url.
rtsp://IP/ch0_0.h264

jd1900, 2020-05-10:
Same here.

I gave to the low stream but I find no audio yet :(
And..., low stream is now in the same port as high stream. Right?

Finally..., is there a way to only stream audio?

agsola, 2020-05-10:
It works in HR for me, but not in LR.

mimes123, 2020-05-10:
@agsola which rtsp do you use for low res. the link i got does not work in vlc. but the high res work but with no audio

mimes123, 2020-05-10:
if you choose mediainformation in vlc do you then have 2 streams one with video and one with audio?

mimes123, 2020-05-10:
@agsola thanks for the hint! it works now. need to have both settings at high otherwise the audio is too poor

mimes123, 2020-05-10:
My next question. when i use the vlc the audio and picture is fine. but when i open the camera in HomeKit the sound lacks a lot, do you know how to change the settings in the homebridge to get a good quality of the sound ?

roleoroleo, 2020-05-10:
No sorry, I don't use homebridge.

PieVo, 2020-05-13:
Audio quality improvements are work in progress!
https://github.com/PieVo/yi-hack-MStar/issues/2

PieVo, 2020-05-13:
Any preferences where to integrate this? It is using the MI libraries from the camera FW itself (and prints out all sorts of nasty banners (NEON) etc).

We have 3 options:
- tinyalsa 
- separate process and create an additional pipe
- rtspserver as a filter class

Last one has my preference, but it creates a new instance for each connection.

roleoroleo, 2020-05-13:
Do you think we have hw resources?

PieVo, 2020-05-13:
Plenty.. Below the "audio" process is a program that denoises a raw file. I added 32ms sleep between each 512-sample frame to simulate the actual stream.

CPU: 33.4% usr 14.3% sys  0.0% nic 49.7% idle  0.0% io  0.0% irq  2.4% sirq
Load average: 2.48 2.54 2.38 1/105 1519
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
  150     1 root     S    47040 76.9   0 34.7 ./rmm
  528     1 root     S    19012 31.1   0  4.2 ./p2p_tnp
 1442  1381 root     S     3220  5.2   0  1.5 ./audio

torcuat, 2020-05-30:
Is there a way to increase de volume of the audio? I hardly hear it with volume to the max in my smartphone.

PieVo, 2020-05-30:
Of course a bit of gain could be added somewhere in the code... How do you
experience the audio with the original app?

What kind of camera model do you have? I don’t experience that with a y203.
When in the room with the camera, talking normally, the playback with VLC
is also normal.

If you want to hear all the background sounds and turn up the volume, you
(now still) get a lot of noise from the bad electronics design. In the
 original app this is filtered away in the camera firmware.

torcuat, 2020-05-30:
The audio is also weak in the original app. A normal conversation in the room is barely audible.

I have a y25, concretely a 9FUSY256JI.

closed 2020-11-21 by roleoroleo