roleoroleo/yi-hack-MStar #133: Issue trying to build the project

fjcabello, 2020-04-01:
Hi,
I have followed all the steps but it's failing when I try to compile.

I'm using docker image https://hub.docker.com/r/borodiliz/yi-hack and I have followed al the steps.

I'm using master branch.

` ./scripts/compile.sh

------------------------------------------------------------------------
 YI-HACK - SRC COMPILER
------------------------------------------------------------------------

MOD_DIR:        /yi-hack-src/yi-hack-MStar/src/busybox
MOD_NAME:       busybox
MOD_INIT:       init.busybox
MOD_COMPILE:    compile.busybox
MOD_INSTALL:    install.busybox
Initializing busybox...

HEAD is now at b84194b13 Bump version to 1.29.3
Compiling busybox...

  HOSTCC  scripts/basic/split-include
scripts/basic/split-include.c: In function ‘main’:
scripts/basic/split-include.c:134:6: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
  134 |      fgets(old_line, buffer_size, fp_target);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
fixdep: /usr/include/i386-linux-gn: No such file or directory
make[2]: *** [scripts/Makefile.host:104: scripts/basic/split-include] Error 2
make[1]: *** [/yi-hack-src/yi-hack-MStar/src/busybox/busybox/Makefile:357: scripts_basic] Error 2
make: *** [Makefile:522: include/autoconf.h] Error 2
`

Any ideas?

Thanks,

Paco

roleoroleo, 2020-04-01:
What toolchain are you using?

fjcabello, 2020-04-01:
Hi,
I'm using the configured toolchain in docker image ( https://hub.docker.com/r/borodiliz/yi-hack).

gcc version 4.8.3 20140401 (prerelease) (crosstool-NG linaro-1.13.1-4.8-2014.04 - Linaro GCC 4.8-2014.04)

Thanks,

Paco

roleoroleo, 2020-04-02:
Sorry but i don't know this error because I don't use docker.
Try to ask to the author: @borodiliz

fjcabello, 2020-04-02:
Ok , thanks.
I'll try to setup the environment in a linux machine. Do you have any guideline for the setup?
I have downloaded MStar MSC3XX SDK.zip. 
I would like to have a setup close as possible to one that it's working.

Thanks,

Paco

roleoroleo, 2020-04-02:
Copy the toolchain in /opt/yi/
gcc, g++ and the other binaries must be in /opt/yi/arm-linux-gnueabihf-4.8.3-201404/bin
Install all needed packages: autoconf automake libtool flex bison build-essential ...
Run the scripts.
If the build exit with errors, probably you miss a package.
Check the log, install the missing package and rerun the script.

closed 2020-05-07 by roleoroleo