roleoroleo/yi-hack-MStar #134: Switch Video On/Off bug in 0.2.9

fir3burn, 2020-04-02:
Hi,
I've installed 0.2.9 firmware in yicam 6FUS (y203) and the video switch on/off doesn't work.
I've tried the command "ipcmd -t off" in the shell and it doesn't work.

Thanks 4 your great work!!

roleoroleo, 2020-04-02:
I checked it now with a sniffer and when I run "ipc_cmd -t off" the ipc message starts correctly.
02 00 00 00 08 00 00 00 75 00 01 00 00 00 00 00 00

fir3burn, 2020-04-03:
Yes, the command is executed, but I can get the video of the camera after the command is executed.

roleoroleo, 2020-04-03:
Video off where?
RTSP? App?

fir3burn, 2020-04-04:
This is my config
`/home/yi-hack/etc # cat camera.conf
SAVE_VIDEO_ON_MOTION=no
SENSITIVITY=low
LED=no
ROTATE=no
IR=yes
SWITCH_ON=yes
/home/yi-hack/etc # cat system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=no
MQTT=no
RTSP=yes
RTSP_HIGH=yes
ONVIF=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
RTSP_STREAM=high
ONVIF_PROFILE=high
RTSP1_PORT=8554
FREE_SPACE=0
`
My Cams has cloud disabled, and i'm trying to stop video when I am at home, and start the video when I'm out. I'm getting the video from Hassio with RTSP.
In v. 0.2.2, I was sshing to the cam and killed the rtsp process when I wanted video off, and started again when I wanted video on.
In this version, there is a watchdog that start the process when it is killed. That's the reason that i'm trying with the video switch on/off.

Thanks for time.

roleoroleo, 2020-04-04:
The video on/off option doesn' work in this manner.
It's a clone of the on/off command available in the original Yi app.
When you switch off, only the Yi services stop.
The rtsp service continues to work normally.
If you want to stop it, you have to kill the watchdog wd_rtsp.sh and then rtsp service.

fir3burn, 2020-04-05:
Ok.
Then with cloud disabled, this switch don't have any function.
I have changed to kill the processes again, and it works.
Thanks

roleoroleo, 2020-04-05:
With cloud disabled the cam receive the off command and switch off the video to the app.
But in fact nothing happens.

majherek, 2020-04-10:
@fir3burn 
Can you tell me, how you send kill command  to camera? I want to do same things... When the alarm in house will be turned on, the camera will start rstp, and when alarm is off, to kill (stop) video processing.

fir3burn, 2020-04-11:
To start the video stream
`/usr/bin/ssh root@IP -o StrictHostKeyChecking=no -i /ssl/id_rsa 'export LD_LIBRARY_PATH=/usr/local/lib:/usr/lib:/home/lib:/home/hisiko/hisilib:/home/app/locallib:/home/yi-hack/lib:/tmp/sd/yi-hack/lib; export PATH=/usr/bin:/usr/sbin:/bin:/sbin:/home/base/tools:/home/app/localbin:/home/base:/home/yi-hack/bin:/home/yi-hack/sbin:/tmp/sd/yi-hack/bin:/tmp/sd/yi-hack/sbin; PID=``ps auxw | grep wd_rtsp.sh | grep -v grep``; if [ "$PID" = "" ]; then (``/home/yi-hack/script/wd_rtsp.sh`` &) else  echo "is running"; fi'`

To stop the video stream
`ssh root@IP -o StrictHostKeyChecking=no -i /ssl/id_rsa 'PID=``ps auxw | grep rRTSPServer | grep -v grep``; if [ "$PID" = "" ]; then echo "no running";else killall wd_rtsp.sh; killall rRTSPServer; fi'`

To check status
`ssh root@IP -o StrictHostKeyChecking=no -i /ssl/id_rsa 'PID=``ps auxw | grep wd_rtsp.sh | grep -v grep``; if [ "$PID" = "" ]; then echo "off";else echo "on";fi'`

You must change the destination IP, and the private/public key.
Private key in the scripts, and public in camera.
`mkdir /home/yi-hack/.ssh`
`chmod 700 /home/yi-hack/.ssh`
`chmod 600 /home/yi-hack/.ssh/authorized_keys`
Note: `` it's only 1, but it's a special symbol in the post and i have to duplicate it.

closed 2020-04-20 by roleoroleo