roleoroleo/yi-hack-MStar #135: App-less Operation

ArminHobbyProjects, 2020-04-03:
Hi,
first of, congrats to your effort!

I'am trying to use the cam without the app. A few questions:
-Despite using the hack, I need to initially configure the cam using the app, correct?
-Once the camera is configured, is it possible to use the camera without internet access?
So, after a reboot or power cycle, does the camera (dispatch process) try to connect a chinese server,
or is the authentication/authorization stored locally on the camera? 

What I achieved with yout help so far, based on a "virgin" camera:
put a modified factory_test script on a sd card,
included a wifi configuration,
started telnetd via newly built busybox on the sd card.

I can log in with telnet, but how do I get the stream started?
I can envision two routes:
- tweak the dispatch/rmm/oss/... processes
or
- try to create a new service using the SDK (which is currently way passt my abilities...)

Any input is greatly appreciated!

Armin

roleoroleo, 2020-04-03:
> -Despite using the hack, I need to initially configure the cam using the app, correct?

Yes

> -Once the camera is configured, is it possible to use the camera without internet access?

I don't know, some users have reported problems in this mode.
After a power cycle the camera connects to the server only if you don't disable cloud function from the web interface. But the authentication is always stored locally

Which model do you have?

ArminHobbyProjects, 2020-04-03:
Is a 9FUSY25.
Do you know how/where the authorization is stored locally?
I will probably need to get a spare android phone to register my unit to get ahead. The SDK is awfully complex...

roleoroleo, 2020-04-03:
Are stored in flash in a specific area.
And copied in a file in /tmp

paulioliolio, 2020-04-09:
> > -Once the camera is configured, is it possible to use the camera without internet access?
> 
> I don't know, some users have reported problems in this mode.

I have noticed that the camera stops doing alarm recording if I disable internet access for it on my router.

roleoroleo, 2020-04-10:
Already other users have reported this problem.
Unfortunately there is nothing I can do about it.

closed 2020-05-07 by roleoroleo