roleoroleo/yi-hack-MStar #136: Support for 4BUS firmware?

ChrisWeiss, 2020-04-03:
I have several Yi 1080p cameras with serial numbers that start with 4BUSY. I've not seen any sign of custom firmware for this series, is that something that's in the works? Am I mis-identifying the series?

roleoroleo, 2020-04-04:
Are you able to do a dump of the log boot?
See the wiki.

ChrisWeiss, 2020-04-05:
I should be able to do that in the next day or two. I have the technical knowledge, but haven't tried dumping a ROM (now's as good a time as any to learn).

ChrisWeiss, 2020-04-06:
So I tried dumping the firmware, I managed to figure out the wiring and get a serial console, but I can't seem to interrupt the boot process. There's a particular line in the output log that I'm wary of:
`### UBOOT:test pin high! ###`
Is there some test point I need to pull low to allow interruption of boot?

I've attached the full log and also a picture of my board with wiring. Sadly, whatever those labelled pins are, I couldn't get any data from them.
[YiCam_4BUSY_Firmware_Dump.txt](https://github.com/roleoroleo/yi-hack-MStar/files/4435457/YiCam_4BUSY_Firmware_Dump.txt)
[image]

roleoroleo, 2020-04-06:
Your cam is based on Hisilicon 3518 chip.
Probably you can try yi-hack-v3 or v4.
My fw doesn't support this platform.
Sorry.

ChrisWeiss, 2020-04-06:
Ahh. Thank you for that feedback. I have several cameras and was able to successfully use your firmware on some of them. If nothing else, I learned how to monitor the serial console on a consumer hardware object. Thanks!!

closed 2020-04-06 by ChrisWeiss