roleoroleo/yi-hack-MStar #138: latest still image

divemasterjm, 2020-04-06:
Hi there is a time lap between still frame and last frame, how we can get the current frame on notification? my homekit notifications gets always non current live frame.

roleoroleo, 2020-04-06:
Sorry, I don't understand.

divemasterjm, 2020-04-06:
the snapshots are not in realtime.

roleoroleo, 2020-04-06:
Compared to the rtsp stream?

divemasterjm, 2020-04-06:
yes, when i receive notification with snapshot or look for snapshot on homekit app always is not real time snapshot

divemasterjm, 2020-04-06:
here is my config
"videoConfig": {
                        "source": "-rtsp_transport tcp -i rtsp://192.168.8.221:554/ch0_0.h264",
                        "stillImageSource": "-rtsp_transport tcp -i rtsp://192.168.8.221:554/ch0_0.h264 -vframes 1 -r 1",
                        "maxStreams": 5,
                        "maxWidth": 1280,
                        "maxHeight": 720,
                        "maxFPS": 15,
                        "debug": false,
                        "audio": false

roleoroleo, 2020-04-06:
Try to use (if it's possible with homekit) the http protocol.
http://192.168.8.221:8080/cgi-bin/snapshot.sh
The image retrieved with ffmpeg could be very slow.

divemasterjm, 2020-04-06:
do you know how to say to ffmpeg to retreive http?

roleoroleo, 2020-04-06:
No, sorry.

closed 2020-04-08 by divemasterjm