roleoroleo/yi-hack-MStar #139: 9FUS Y25 / Y203C

paulioliolio, 2020-04-06:
Yi 1080p Home v3 on stock 4.2.0.0H_201909041620  firmware.
I have tried the Y25 and the Y203C firmware here and the Home 1080P firmware from yihack v4 each on 2 separate 16GB SD cards, but none will flash, the device just starts as normal. 

Looks like an MStar device to me!

[image]

[image]

[image]

roleoroleo, 2020-04-06:
Your fw is y25.
Try again with a different sd card.

paulioliolio, 2020-04-06:
Thanks, re-formatted both cards on my Linux machine (formatted under windows previously) and tried again with Y25 files (both in root) but still not flashing new firmware.

paulioliolio, 2020-04-06:
Third time lucky! Found another card and this one works perfectly. Thanks!

paulioliolio, 2020-04-07:
Took me a while to discover that FTP uses same credentials as telnet. And I can't get the camera working on my QNAP (no alarm recording!) but I suspect that's because it's not the best NVR. Thanks for the great work.

closed 2020-04-07 by paulioliolio