roleoroleo/yi-hack-MStar #141: Activate Crond

miclass, 2020-04-09:
Hi,

I would like to activate crond to set some automatic task, for example a command like the following to take a snapshop each X minutes and save it on the SD:

`imggrabber > /tmp/sd/$(date +"%Y-%m-%d_%H.%M.%S").jpg`

Which is the best way to add this in crontab?

Thank you.

roleoroleo, 2020-04-09:
See at the end of the system.sh script: https://github.com/roleoroleo/yi-hack-MStar/blob/master/src/static/static/home/yi-hack/script/system.sh

miclass, 2020-04-12:
Thank you for the hint, I did as suggested and it worked.

I have another question: if I edit the system.sh file locally on my pc and I upload it using FTP then the cam does not work anymore. It boots and connect to wifi (I can ping it) but I cannot access it through ssh/ftp anymore and through the web only the header of the page gets loaded. The only solution is to flash again with a different version of the firmare. I managed to edit the file directly on the cam using vi, in this way it works with my changes.

Do you have any idea why it behaves in this way?

Thanks!

roleoroleo, 2020-05-07:
2 common problems:
1. You don't set the correct permissions (the file must be executable).
2. You are using CR-LF from a Windows system.

miclass, 2020-05-07:
Thank you for the feedback, I solved it in the meatime, it was problem number 2 :-)

closed 2020-05-07 by roleoroleo

comanche3, 2021-01-10:
i tried
* * * * * imggrabber > /tmp/sd/record/123.jpg

on yi hack configuration crontab, but nothing seems to happen. Does anyone know where i made a mistake ?

miclass, 2021-01-10:
> 
> 
> i tried
> 
>     * * * * * imggrabber > /tmp/sd/record/123.jpg
> 
> 
> on yi hack configuration crontab, but nothing seems to happen. Does anyone know where i made a mistake ?

I activate mine with these lines of code inside system.sh (one image each 15 minutes):

`mkdir -p /var/spool/cron/crontabs/
echo "*/15 * * * * sh /home/yi-hack/script/snapshot.sh" > /var/spool/cron/crontabs/root
/usr/sbin/crond -c /var/spool/cron/crontabs/`

snapshot.sh is a shell script with the following content:
`#!/bin/sh
TIMESTAMP=`date +"%Y-%m-%d_%H.%M.%S"`
imggrabber > /tmp/sd/"$TIMESTAMP".jpg`

Ciao.