roleoroleo/yi-hack-MStar #142: Alexa support

fjcabello, 2020-04-11:
Hi,
I guess Alexa support is not working with the firmware. Do you have an idea how to enable it?
Also, I was wondering how you managed to maintain compatibility with yihome app. Did you implement the same API? Do you have a daemon to connect with yihome servers? Any documentation about how this is implemented?

Thanks

roleoroleo, 2020-04-11:
All my feature are added.
If you set cloud enable, the yi processes are running.
You cannot have alexa with cloud disabled.

fjcabello, 2020-04-12:
Thanks. It should be a different issue not related to yi-hack.
One question, in this firmware, are you including the original yihome applications? I mean, in order to keep compatibility with the app.

Thanks

roleoroleo, 2020-04-12:
Th original app are untouched.
Are stopped only if cloud option is disabled.

closed 2020-04-29 by roleoroleo