roleoroleo/yi-hack-MStar #143: 6FCN model - Can't flash

ShikyC, 2020-04-12:
According to https://github.com/roleoroleo/yi-hack-MStar/issues/33 the 6FCN model is supported however I've tested all of the 3 pairs of firmware and none of them could be flashed. I saw another user in that thread sharing a similar problem as I do.

I formatted the SD card, put the 2 files in the root folder and power on the camera, the camera just booted as normal without flashing the firmware, repeating "waiting to connect." This is a brand new camera bought from China and I'm unable to connect it to the app to see any further information due to the region lock.

roleoroleo, 2020-04-12:
Unfortunately 6FCN is not enough to say that is supported.
The cam must be based on MStar chipset.
Try to post some pictures of the internal board or try to get a log of the boot process.

ShikyC, 2020-04-12:
Thanks @roleoroleo for the quick response! I tore it down and took a picture of its motherboard. Unlike any other Yi cams (as I can see on the internet) this is a rectangular motherboard, fixed on the base of the camera and partially occluded by the motor. As far as I can see it uses the same MSC313E chip.

By the way, the firmware for this model listed on the official website is named `4.0.0.0F_201909251646home_y30m` so I suppose the naming of the firmware might also be an issue. It's downloaded here: [http://download.xiaoyi.com/firmware/smarthomecam/4.0.0.0F_201909251646home_y30m](http://download.xiaoyi.com/firmware/smarthomecam/4.0.0.0F_201909251646home_y30m)

Here is the picture:

[image]

linsong837, 2020-04-13:
I have 6FCN too.
I can flash it. I can connect it to wifi. I can view the video stream through VLC.
I just can't bind it with Yi Home App.
Searched around, somebody said had to be Chinese version Yi Home. So I downloaded right version and tried again. Still no luck.

@ShikyC When you heard waiting to connect, did you follow the instruction in Yi Home App? Did you generate the QR code to let camera scan? Were you able to connect to WiFi?

roleoroleo, 2020-04-13:
@ShikyC 
It's based on MStar but I have never seen a cam like this.
At the moment it's not supported.
Are you able to dump a bootlog? See the wiki for details.

ShikyC, 2020-04-13:
@linsong837 If you can flash the firmware I think all you need is to configure ProxyChain-ng to bypass the region lock. I'm wondering how did you flash the firmware and connect it to wifi _without_ binding the Yi app? Which firmware did you use?

On my end when I hear "waiting to connect" I just follow the steps in the app (scanning code, configuring WiFi, etc.) and finally the camera says "this camera can only be used in China" and nothing more happens.

ShikyC, 2020-04-13:
> @ShikyC
> It's based on MStar but I have never seen a cam like this.
> At the moment it's not supported.
> Are you able to dump a bootlog? See the wiki for details.

@roleoroleo If I were able to find the RX/TX pins on the board... I'll give it a try.

linsong837, 2020-04-13:
@ShikyC Bind the app is the step after connect to WiFi.
When you heard "this camera can only be used in China", that meant you had already connected to WiFi, otherwise, it wouldn't be able to know where you are.
What you should do is find the IP address and access the web management console through IP address.
There are many ways to figure out the IP address of your device.
You can run some IP scan tools on your phone or PC, like Fing.
It will scan local subnet and list all devices. 
You will find the IP address by the device with name of "Shenzhen Tong Bo Wei Technology".
Or, you can go to your router directly, check the DHCP client list. You will find the new added device and its IP.

Once you grab the IP address, you can access the web interface through link http://ip:8080
In the manage console, you can go to Configurations tab, turn on Proxychains-ng.
Reboot your device, then you won't hear the warning of "this camera can only be used in China".

You can even view the video in VLC by the link rtsp://Your_Device_IP:554/ch0_0.h264

The only problem I had is no app binding. It is always failed no mater English version app or Chinese version app.

ShikyC, 2020-04-14:
@linsong837 thanks for the detailed steps. I tried accessing the web interface at my camera's IP:8080 and the rtsp streaming but none of them work so I think the firmware had never been successfully flashed. I bought it last October so it might have a newer version of firmware blocking it from being flashed with unverified firmware...

Is there any other way to bypass the region lock without flashing the firmware? For instance, using the proxy on the router instead of on the camera? @roleoroleo

linsong837, 2020-04-15:
@ShikyC Make sure your IP is correct.
Can you ping that IP?
Can you ssh to it?

f0854, 2020-06-03:
good morning everyone, I have a yi Dome X 2019 cam, BFUS model and 4.0.0.0F_201909251646home_y30m firmware, the same as ShikyC, and I also cannot flush the cam. Obviously I tried with all the firmware and all available versions.

roleoroleo, 2020-06-03:
y30 is not supported at the moment.

f0854, 2020-06-03:
> y30 is not supported at the moment.

Thanks for the quick response and for the great work you do, I look forward to it.

roleoroleo, 2020-06-04:
If you want to help me to add the support for this model, please make a dump of the flash as explained in the wiki.

f0854, 2020-06-05:
> 
> If you want to help me to add the support for this model, please make a dump of the flash as explained in the wiki.

@roleoroleo Good morning and thanks again for the work you are doing for all of us. I would love to help you but I have read the wiki and unfortunately I do not have the manual skills and familiarity necessary to produce the dump that interests us. I will try to talk about it with my friends but I see it difficult

closed 2020-08-20 by roleoroleo