roleoroleo/yi-hack-MStar #144: Recordings stopped saving to SD until reformat

arpanett, 2020-04-13:
Hi, firstly thank you for the awesome firmware! I've been using MStar firmware on my Yi 1080 BFUS using y203 0.2.7, and it's been working perfectly for 2 weeks with no issues. I did run into a weird issue the other day when I turned it on and noticed that the RTSP feed is working as normal, but I had no recordings being saved to the SD card, and my motion event list was empty. I had it set to record 24/7 previously, with all cloud features and motion detection disabled.

To fix it I had to reformat the SD card and put the firmware onto it again. Now the SD card is working again as normal with recordings showing up in the web UI. I was just wondering if anyone would know the cause of this could be? 

_Additional note: I did try a few other things to resolve this issue before trying the SD card reformat. But none of the following were able to fix it:_
- turning settings on/off
- resetting the device
- updating firmware from 0.2.7 to 0.3.0

roleoroleo, 2020-04-13:
Maybe a FAT32 issue on SD?
Did you try to check the filesystem?

mihir8786, 2020-04-14:
I also have a similar issue on a BFUS camera with 0.3.0 (although it was always there so the version doesn't matter). Basically, SD card would record for a while and then stop working. I have space set at 10%  but the recording can stop any time with more than 80% space on the card. To restart recording I have to allow cloud access, go to Yi app, and format the card. Once I disable cloud again, it would continue to work but would stop eventually.

Interestingly, I have another camera (same model, same version, same SD card type, same size etc) and it records flawlessly so don't think it's a firmware issue. Might be the SD card or the actual camera.

arpanett, 2020-04-17:
> 
> 
> Maybe a FAT32 issue on SD?
> Did you try to check the filesystem?

Hi I just did chkdsk on the SD card and it found no errors, and file system is listed as being FAT32. 

I also just noticed today that since I last reset to fix the recordings it has happened again, about 2 days after last. But the RTSP + web interface has continued working like it did last time. 

I wonder if SD card size could be the issue, as I'm using an 8GB card on it currently. Also I have noticed when it has stopped recording it happened when it was last unplugged from power, then the next day when I've turned it on it won't record again (until I reset it as mentioned). Although I have unplugged it probably about 20+ times since then with no issue - so not certain it can be the cause of this.
> I also have a similar issue on a BFUS camera with 0.3.0 (although it was always there so the version doesn't matter). Basically, SD card would record for a while and then stop working. I have space [...]

Hi I also tried changing space set values, as I originally had it set to 1% then to 10% but it didn't seem to remedy the recording issue. I was also thinking the Yi app sync may be a fix for it, that was the last thing I was going to try before I did the reformat. I will give it a go later tonight since it happened again, and will report back.

roleoroleo, 2020-04-18:
Could you try another SD card?

roleoroleo, 2020-07-17:
Closing without response from issue filer for >14 days.

closed 2020-07-17 by roleoroleo