roleoroleo/yi-hack-MStar #145: can't flash BFUS model

lumpfisch, 2020-04-13:
I have a BFUS model and tried to flash the camera but that didn't work.
I put both files from the y203c_0.3.0.tgz file on the sd card. After booting the camera it connects to the WIFI.
I tried already a reset. The camera created two directories on the sd card (log and record), so in my opinion the sd card is working properly.

roleoroleo, 2020-04-13:
If the fw is 8.2.0 the cam isn't MStar based.
This hack is not compatible.

lumpfisch, 2020-04-13:
Is Allwinner-Yi-Hack applicable?

roleoroleo, 2020-04-13:
Allwinner would be applicable but doesn't exist yet.

Mattsouthgate, 2020-04-19:
I have a BFUS model, (BFUSY212MR) bought from Amazon in a two-pack - arrived today, running firmware version 4.5.0.0C_201910080934.  I have followed the instructions using an 8GB microSD card with home_y203c and sys_y203c files and the camera does not go through the update firmware, flashing yellow led, process.   What further information do you need?

roleoroleo, 2020-04-20:
Change SD and try again.

Mattsouthgate, 2020-04-20:
I tried another Samsung card 16GB this time not 8GB and it worked as expected.  Orange led flashing for about 40 seconds.  I setup the camera using the app and have now accessed the web interface at port 8080.  Thank you.  However, I am going to transfer the card into the other camera and see if the update process work with that too.  I am unsure why the previous card should prevent the flash ocurring.  I will let you know how I get on with the other camera.  

Is there anything I could preserve for you or others to indentify how the update problem manifested?

roleoroleo, 2020-04-20:
Other users have reported the same problem but a reason has not been identified.

Mattsouthgate, 2020-04-20:
The update of both cameras was successful with the second card: a Samsung 16GB basic affair.  I have attached images of the 'problem' Samsung card which I have had for a long time and used without problem - the camera even created the default directories on it when initialised.

Attached are images of the Samsung microSD 8GB class 6 card: front and back, and the contents of the card with the contents of the record directory cleared of videos due to size.  Hopefully, this can help someone.

Lastly, thanks again.

[SD Contents.zip](https://github.com/roleoroleo/yi-hack-MStar/files/4502960/SD.Contents.zip)

[image]
[image]

closed 2020-05-07 by roleoroleo