roleoroleo/yi-hack-MStar #147: Adding remote control

metalnodeug, 2020-04-13:
Hi 

I don't know if you can help but this is my issue. I have a YI DOME 1080P model h20 and I use in the past the Yi-Hack original from TheCrypt0 but unfortunately the project is dead since 1 year.

I use my cam with disable cloud and I showing the stream with my Synology NAS but when my connection restart or I reboot the cam, she reset the position. 

I will know if it's possible to add to my cam, remote control ? And if it's possible, how ?

I have test to reset my camera for updating, upgrading and restore the firmware but nothing. Doesn't work and I don't know why ?!

Thanks a lot for your help.

roleoroleo, 2020-04-15:
Do you mean ptz support?

metalnodeug, 2020-04-15:
Yes I thing

roleoroleo, 2020-04-15:
Probably Hisilicon cams use the same method as MStar.
On my fw, you can use ipc_cmd program to send ptz commands.
You can move left, right, up, down and go to a saved preset.
You could try to cross compile the sources and check if it works.

metalnodeug, 2020-04-15:
I can't upgrade or dowgrade the firmware. The process doesn't work.
I use an SD Card, upload the 2 files at root path, disconnect the cam, insert sd card, power on the cam and the cam boot normally :/ she will not launch the process. My process are faulty ? Have you an idea ?

roleoroleo, 2020-04-16:
If you are using TheCrypt0 fw you can't use MStar firmware.
Different platform.

closed 2020-04-20 by roleoroleo