roleoroleo/yi-hack-MStar #148: IP address not correctly reported

RomRider, 2020-04-15:
Hey @roleoroleo,

Super great job here! Love it :)

One thing that I've found is that `onvif_srvd` seems to not find the ip address on the interface (during SOAP answers). It always reports `127.0.0.1`. I'm running `Yi 1080p Dome BFUS` with the latest version `0.3.0` (was already the case with previous versions)

It seems that this function always reports `127.0.0.1`:
https://github.com/KoynovStas/onvif_srvd/blob/ed7e51be6968d52808e2d2040fb586ddfbbafef2/src/ServiceContext.cpp#L29-L48

This fails with home-assistant for example as it uses the IP address reported in the answers (which is expected) and tries to connect to `127.0.0.1`

RomRider, 2020-04-15:
Note: I've hardcoded the IP address in the function and recompiled the binary and it's now working as expected. So either the parameter passed to the function is not correct or something's wrong inside this function :)

roleoroleo, 2020-04-15:
Very strange.
I use onvif with ha and I have never had this problem.
Please post your ifconfig.

RomRider, 2020-04-16:
Sure, here you go:
```
/home/yi-hack # ifconfig -a
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr <REDACTED>
          inet addr:<REDACTED>  Bcast:<REDACTED>  Mask:<REDACTED>
          UP BROADCAST RUNNING MULTICAST  MTU:1200  Metric:1
          RX packets:8097966 errors:0 dropped:37357 overruns:0 frame:0
          TX packets:17131175 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:857521912 (817.7 MiB)  TX bytes:152766893 (145.6 MiB)
```

roleoroleo, 2020-04-17:
Could you explain me which messages report the wrong address?
A screenshot of onvif device manager or a network packet capture...

Skelly82, 2020-06-27:
Hi, I have same issue with Home Assistant. Camera works fine on Onvif Device Manager, but when I try to add Onvif integration to HA this error occurs:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/aiohttp/connector.py", line 936, in _wrap_create_connection
    return await self._loop.create_connection(*args, **kwargs)  # type: ignore  # noqa
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 962, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 949, in create_connection
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.7/asyncio/selector_events.py", line 473, in sock_connect
    return await fut
  File "/usr/local/lib/python3.7/asyncio/selector_events.py", line 503, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('127.0.0.1', 80)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_protocol.py", line 418, in start
    resp = await task
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_app.py", line 458, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_middlewares.py", line 119, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/real_ip.py", line 39, in real_ip_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 73, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 127, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 129, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 145, in post
    return await super().post(request, flow_id)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 60, in wrapper
    result = await method(view, request, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 106, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 153, in async_configure
    result = await self._async_handle_step(flow, cur_step["step_id"], user_input)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 201, in _async_handle_step
    result: Dict = await getattr(flow, method)(user_input)
  File "/usr/src/homeassistant/homeassistant/components/onvif/config_flow.py", line 170, in async_step_auth
    return await self.async_step_profiles()
  File "/usr/src/homeassistant/homeassistant/components/onvif/config_flow.py", line 242, in async_step_profiles
    profiles = await media_service.GetProfiles()
  File "/usr/local/lib/python3.7/site-packages/zeep/asyncio/bindings.py", line 13, in send
    options["address"], envelope, http_headers
  File "/usr/local/lib/python3.7/site-packages/zeep/asyncio/transport.py", line 107, in post_xml
    response = await self.post(address, message, headers)
  File "/usr/local/lib/python3.7/site-packages/zeep/asyncio/transport.py", line 95, in post
    proxy=self.proxy,
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client.py", line 483, in _request
    timeout=real_timeout
  File "/usr/local/lib/python3.7/site-packages/aiohttp/connector.py", line 523, in connect
    proto = await self._create_connection(req, traces, timeout)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/connector.py", line 859, in _create_connection
    req, traces, timeout)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/connector.py", line 1004, in _create_direct_connection
    raise last_exc
  File "/usr/local/lib/python3.7/site-packages/aiohttp/connector.py", line 986, in _create_direct_connection
    req=req, client_error=client_error)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/connector.py", line 943, in _wrap_create_connection
    raise client_error(req.connection_key, exc) from exc
aiohttp.client_exceptions.ClientConnectorError: Cannot connect to host 127.0.0.1:80 ssl:None [Connect call failed ('127.0.0.1', 80)]
```
Other Onvif-enabled cameras work so I assume HA integration is fine. Could be the issue that the camera is on different subnet to HA?

roleoroleo, 2020-06-27:
You are right.
If you take a look here:
https://github.com/KoynovStas/onvif_srvd/blob/master/src/ServiceContext.cpp
@ line 48

When the connection is not in the same subnet, the function returns localhost.

RomRider, 2020-06-27:
It shouldn't be a problem to not have the camera on the same subnet. It's a pretty common setup where the controller is wired in one vlan and one subnet while the camera is wireless on another vlan and subnet.

roleoroleo, 2020-06-27:
Try this:
[onvif_srvd.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4840637/onvif_srvd.tar.gz)

Skelly82, 2020-06-27:
Great job @roleoroleo , works like charm. I also think that different subnets for different devices is common scenario.

roleoroleo, 2020-06-27:
I will update the onvif server in the next release.

Skelly82, 2020-06-27:
OK, thank you. Also great job with this firmware, I'm happy user of 3 cameras with your firmware on them.

jonferreira, 2020-07-08:
I've recently flashed my BFUS YI Home camera and so far seems to work great but I'm experiencing some issues with ONVIF

While RTSP works fine when I try to add the camera as ONVIF to the Synology Surveilance Station it just fails. A network Scan doesn't seem to find it either.

If I try with Tiny Cam Pro app the camera is added successfully but when trying to live view I get a "Hostname should be specified" error. Camera is in the same subnet as my phone/pc/synology but the "hostname should be specified" message makes me wonder if it's somehow related?

roleoroleo, 2020-07-09:
Try with ONVIF Device Manager and check if onvif works.
There is a know compatibility problema with Synology.
Configure the cam as a standard rtsp.

jonferreira, 2020-07-09:
I did.

A network scan only finds my Foscam Cameras :(

jonferreira, 2020-07-09:
This is all that was found, all Foscam cameras.

YI Home is 192.168.1.20

[image]

roleoroleo, 2020-07-09:
I added the WS-Discovery daemon 10 days ago.
It's not yet present in the current released version.
But if you add the cam manually with the ip address, it should work.

jonferreira, 2020-07-09:
Indeed it does.

Strange both TinyCam Pro and Synology doesn't seem to like it though. TinyCam Pro mentions Profile S - not sure if that helps?

I'll try to compare their output with my Foscam cameras and see if something pops up.

roleoroleo, 2020-07-10:
onvif is a hard world.

closed 2020-08-20 by roleoroleo