roleoroleo/yi-hack-MStar #149: Frequent connection drops

fabiosci, 2020-04-15:
Hi,
I own two 9FUS with stock firmare 4.2.0*. Inside the camera is written "Y203C_MB_RT2.0 2019/008/08". I successfully flashed both with y25 firmware and everything went fine and the cameras seem to work as expected. At the moment one camera has been flashed with the 0.2.9 version and the other one with the 0.3.0.
After a while (hours or days, randomly) the cameras disconnect from wifi network (they appear not connected in my router page) and the only way to recover them is to unplug the power supply. This behaviour is not synchronized between the two cameras but occurs with both firmware versions. The 0.2.9 seems to be more stable than the 0.3.0.
Another info: I turn off wifi network during the night but in the morning the cameras usually connect normally.

roleoroleo, 2020-04-15:
It could be a memory and/or cpu usage problem. Try to disable unnecessary services.

fabiosci, 2020-04-15:
I think unnecessary services are already disabled. This was and still is my configuration:
- Disable cloud, RTSP, FTP, NTPD and HTTPD: on. I've just disabled FTP because at the moment I don't record on sd card (not inserted in the cam)
- Recording without cloud, ONVIF, SSH, Legacy FTP, Telnet,: off
- RTSP stream: high
- ONVIF profile: high
- MQTT: on
- Camera setting page: everything on, Detection sensitivity low
Do you think it could be useful to reduce RTSP stream and ONVIF profile to low?
Can it be useful to monitor the free memory from the main page?

Another detail (maybe important): I can randomly hear a sound from the cameras as if were reboot (like a click), but they continue working after this "click". Never understood meaning and reason of this random sound.

Thanks for all your work!

roleoroleo, 2020-04-16:
Did you noticed the same behavior with the original fw?
Because my fw doesn't apply change to the wifi part.

fabiosci, 2020-04-16:
I don't know because I flashed with your firmware 5 min after unboxing and didn't backup the original firmware (at my own risk, I know).
The strange thing is that both cameras act the same way...

roleoroleo, 2020-04-16:
A serial log would be useful.
Are you able to do it?

fabiosci, 2020-04-16:
do you mean I have to follow the first part of this guide: https://github.com/roleoroleo/yi-hack-MStar/wiki/Dump-your-backup-firmware ?
If not, could you please point me in the right direction?
thanks

roleoroleo, 2020-04-17:
Yes.

fabiosci, 2020-04-19:
here the putty 
[putty.log](https://github.com/roleoroleo/yi-hack-MStar/files/4499193/putty.log)
log

roleoroleo, 2020-04-19:
The logs are ok.
Is it possible to record the log across a disconnection?
Could you leave the log enabled for a long time?

fabiosci, 2020-04-19:
sure!
I'm already logging and I'll stop it the next time it disconnect.
In the meanwhile I took a picture because log in this moment is continually changing and this is a difference with the one attached to the previous post
[image]

fabiosci, 2020-04-22:
Just an update: after 3 days my cam is still online (unfortunately, I'd say...) and the log still keep changing as shown in the previous picture. On the other cam I applied the new firmware release (0.3.1). I'm monitoring the updated cam to see if anything changes

fabiosci, 2020-04-30:
Another update: I noticed that the camera with v0.3.1 (not under log recording) worked fine for some days until I blocked its traffic from firewall. after more or less 24h it stopped working: it disconnected from wifi network and blue led started blinking

fabiosci, 2020-05-04:
it finally stopped working under monitoring. I found the led blinking, the cam disconnected from wifi and here there is the serial log:
[putty.zip](https://github.com/roleoroleo/yi-hack-MStar/files/4573822/putty.zip)
[blinking led.zip](https://github.com/roleoroleo/yi-hack-MStar/files/4573839/blinking.led.zip)

hope this can help in finding the issue
thank you

roleoroleo, 2020-05-04:
Unfortunately the log doesn't show any problem.

fabiosci, 2020-05-05:
ok, thank you. I'll try by disabling one by one all features and check if one of them can cause the crash

fabiosci, 2020-06-05:
Just an update: the cam seems to stop working when it can't communicate with the cloud. I tried the following scenarios:

SCENARIO 1
Option "disable cloud" ON --> the cam stops working after a while (hours or days) (tested with firmware 0.3.3)

SCENARIO 2
Option "disable cloud" OFF AND cam with access to the internet --> the cam is still working after several days (tested with firmware 0.3.3)

SCENARIO 3
Option "disable cloud" OFF AND cam traffic blocked by router --> the cam stops working after a while (hours or days) (tested with firmware 0.3.1, I'm going to test also with 0.3.3)

roleoroleo, 2020-06-08:
I have the same behavior regarding scenarios 2 and 3 but not for scenario 1.
My default config is "Disable cloud" ON and my cam works correctly.
Are you blocking the traffic from your router in this scenario?

fabiosci, 2020-06-08:
Sorry but I don't remember. I thought it doesn't affect the behaviour and I didn't take note of this.
In the next days I'll try the following scenarios and report the results:
SCENARIO 1.1: Option "disable cloud" ON AND traffic blocked from the router
SCENARIO 1.2: Option "disable cloud" ON AND traffic NOT blocked from the router

fabiosci, 2020-06-11:
ok, i've got news regarding the
**SCENARIO 1.1: Option "disable cloud" ON AND traffic blocked from the router**
the cam stopped working (as expected) just after less than 2 days and in the router's configuration appears not connected. I'm talking about the cam with firmware 0.3.3

regarding the SCENARIO 1.2 the cam (firmware 0.3.1) is still working

roleoroleo, 2020-06-11:
Ok. So, when the cam can't access to internet, it stops working.
I don't know why, I will check it.

fabiosci, 2020-06-11:
thank you very much!
if it can be useful for furhter tests, i can update the other cam from 0.3.1 to 0.3.4 and monitor its behaviour in the same conditions

fabiosci, 2020-07-06:
another update on the following scenario:

> SCENARIO 1.2: Option "disable cloud" ON AND traffic NOT blocked from the router

after weeks, the cam stopped working.
Do you think that a possible workaround may be an automatic reboot option after n hours?

roleoroleo, 2020-08-20:
Sorry for the delay in the response.

I tried again your scenario and I think your idea is correct.
There is a memory leak when the connection is blocked.
The cam goes in low memory and crashes.

fabiosci, 2020-08-28:
just to let you know my other progresses in investigating this issue: it happens also when the cam is regularly accessing internet both from router (internet connection allowed from firewall) and from firmware (option "disable cloud" off).
this means that the connectivity is not responsible for the problem.
I also found a blog post where someone talks about your firmware and says that it works on yi home camera v1. my cam is v3 (based on the amazon link used to buy it...). i don't know if this can be the reason...

at the moment the cams are useless, hope you'll find the issue soon. If there are tests I can do to help, let me know.
Thanks for your support

elraro, 2020-08-31:
Im having exactly the same problem @fabiosci , but with https://github.com/roleoroleo/yi-hack-Allwinner firmware

roleoroleo, 2020-09-01:
I need more info.
Try to run this script:
https://github.com/roleoroleo/yi-hack-MStar/issues/215#issuecomment-682453320

I will check if there is a memory leak.

mhensema, 2020-09-01:
I'm experiencing the issue as well on the Allwinner firmware/device. I have cloud disabled, but I do see the device making DNS lookups for `api.eu.xiaoyi.com`. This seems to only happen at boot. Internet access was blocked for the device, but enabled it for now to see if it makes any difference.

fabiosci, 2020-09-07:
I run the script from [#215](https://github.com/roleoroleo/yi-hack-MStar/issues/215#issuecomment-682453320) with `nohup ./startup.sh &` on both cams. Both are configured with "disable cloud" on and internet connection blocked by router. I also see a DNS lookup for `api.eu.xiaoyi.com` and `api.github.com`.
One of them stopped working during the last night, probably when wifi was off.
Please ignore the date and time in the log file attached

[top-yicam1.log](https://github.com/roleoroleo/yi-hack-MStar/files/5182548/top-yicam1.log)

roleoroleo, 2020-09-07:
I can't see memory problems.
The mystery deepens.

fabiosci, 2020-09-07:
when also the second cam will stop working I'll post the other log to allow a comparison and find the differences.
I'll also share both configurations (which are different). this might help the investigation

roleoroleo, 2020-09-08:
If the led blinks the cam is disconnected from the wifi.
Is it possible that you have a wifi problem?

fabiosci, 2020-09-08:
i don't think so for at least 3 reasons:
1) only the yi cams have problems. no other devices disconnect
2) the cams are 2m far from the router with just a door (usually open) in between, definitively less than other devices
3) my router is a fritz box, which has a very good wifi signal

the only information i can add is that in the night i turn off the wifi using the router's schedule feature.
I don't think that turn off the wifi during the night is the problem, because the cam doesn't disconnect every day

mhensema, 2020-09-13:
I don't believe Wifi is the issue. The device is still connected. See screenshot. I checked the system log on my router (OpenWRT based), and see no deauth(s) or disconnect(s) from the camera.

After some time the camera does start working again (i.e. the webserver becomes available again) without a power cycle. I'll see if I can get some logging from httpd.

[image]

roleoroleo, 2020-09-13:
Could you check the uptime of the cam?

mhensema, 2020-09-13:
Uptime is at ~11d

[image]

fabiosci, 2020-09-13:
> I don't believe Wifi is the issue. The device is still connected. See screenshot. I checked the system log on my router (OpenWRT based), and see no deauth(s) or disconnect(s) from the camera.
>
> After [...]

it seems that we are experiencing different issues. in my case the device disconnects from the wifi network, the led start blinking and it does not reconnect automatically even after several days.
this is the 7th day without disconnection. At the moment I'm powering both cams with samsung smartphone charger and original usb cables. I'm starting thinking about a power supply issue (e.g. unstable 5v output from the original power supply device, thin cable, etc.)
What do you think about this hypothesis?

roleoroleo, 2020-09-13:
The only power problem I had is that if I use a low power supply the camera goes in boot loop.

fabiosci, 2020-09-14:
after 8 days (is almost the best performance so far) the first cam stopped working, the second one is still connected.
It seems that when it disconnects it's no able to establish the connection to wifi network again. It happens after some time that could be one day or more, such as in this case.
The cam appears disconnected from the router's page and, of course, it is unreachble also from other scanning apps (i.e. fing for android). I tried putty, but as expected I couldn't reach it.

As you said, it doesn't seem to be related to the power supply.

I will post the configurations of both cams as soon as possible, to highligh the differences

fabiosci, 2020-10-16:
I monitored the free memory and it is always 20000kB more or less also when it stop working.
I also disable RTSP, ONVIF and all other options and services just to easily find the problem, and it seems to be just a connection issue: the cam is unable to establish again the connection when the wifi turns on in the morning.
sometimes it's able to reconnect but sometimes not.
it seems to be not related to both enabled options and internet block.

my next test will be allow internet access, enable cloud and keep wifi off during the night. if i'm right sooner or later the cams will disconnect again

any suggestion to go deeper into the wifi matter?

roleoroleo, 2020-10-16:
I don't know how to help you.
I begin to think a kernel driver problem.
Probably you should add a network watchdog that reboot the cam when it's disconnected from the router.

fabiosci, 2020-10-16:
thanks anyway.
in your opinion could the kernel driver problem lead me to my issue also in case of internet allowed and cloud enabled?
i can add a watchdog. do you think that a solution like the one described [here](https://unix.stackexchange.com/questions/141095/automatically-reboot-if-no-wifi-connection-for-a-certain-time) would work?
in case of firmware upgrades, would the script and crontab setting be deleted?

roleoroleo, 2020-10-17:
> in your opinion could the kernel driver problem lead me to my issue also in case of internet allowed and cloud enabled?

The problem still exists, but the effect is probably less.

> i can add a watchdog. do you think that a solution like the one described [here](https://unix.stackexchange.com/questions/141095/automatically-reboot-if-no-wifi-connection-for-a-certain-time) would work?

Yes.

> in case of firmware upgrades, would the script and crontab setting be deleted?

If you add it to the sd card, it will not be deleted.

fabiosci, 2020-12-01:
I finally found some time to spend on this issue. I prepared the script and it works (i'm trying to reset only the wifi connection rather than reboot the cam) but it seems that crond is not running. the output of ps command doesn't show crond.

`/home/yi-hack/script # ps
PID   USER     TIME  COMMAND
    1 root      0:02 /init
    2 root      0:00 [kthreadd]
    3 root      0:01 [ksoftirqd/0]
    5 root      0:00 [kworker/0:0H]
    6 root      0:00 [kworker/u2:0]
    7 root      0:04 [rcu_preempt]
    8 root      0:00 [rcu_sched]
    9 root      0:00 [rcu_bh]
   10 root      0:00 [watchdog/0]
   11 root      0:00 [khelper]
   12 root      0:00 [writeback]
   13 root      0:00 [crypto]
   14 root      0:00 [bioset]
   15 root      0:00 [kblockd]
   16 root      0:00 [kworker/0:1]
   17 root      0:00 [kswapd0]
   18 root      0:00 [fsnotify_mark]
   31 root      0:00 [SCLDAZA_THREAD]
   32 root      0:00 [VIPDazaTask]
   36 root      0:00 [deferwq]
   39 root      0:00 /bin/ueventd
   40 root      0:01 [jffs2_gcd_mtd3]
   41 root      0:00 [jffs2_gcd_mtd2]
   82 root      0:00 [kworker/0:2]
   83 root      0:00 [mmcqd/0]
   94 root      0:00 [cryptodev_queue]
  130 root      0:00 [spi0]
  161 root      0:00 [cfg80211]
  177 root      0:01 [RTW_CMD_THREAD]
  180 root      0:00 ./log_server
  181 root      0:05 ./dispatch
  645 root     10:10 ./rmm
  648 root      0:01 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
  683 root      0:00 [kworker/u2:3]
  763 root      0:00 udhcpc -i wlan0 -b -s /home/app/script/default.script -x hostname:yicam-taverna
 1079 root      0:00 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
 1099 root      0:00 dropbear -R
 1104 root      0:00 ipc_multiplexer
 1119 root      0:01 dropbear -R
 1125 root      0:00 ntpd -p 192.168.188.2
 1165 root      0:00 -sh
 1315 root      0:00 ps
`

do I miss something?

roleoroleo, 2020-12-02:
I'm working on a new version where you can configure cron through the web interface.

fabiosci, 2020-12-02:
> I'm working on a new version where you can configure cron through the web interface.

thank you! the possibility to schedule a custom script it's a good news

roleoroleo, 2020-12-04:
Release in 0.3.9
Let me know if it works correctly.

fabiosci, 2020-12-04:
just upgraded.
first of all i checked if crond was running and ps gave me:
`1235 root      0:00 /usr/sbin/crond -c /var/spool/cron/crontabs/`

but crontab -l doesn't seems to work:
```
/home/yi-hack # crontab -l
crontab: can't change directory to '/crontabs': No such file or directory
```
the root file in /var/spool/cron/crontabs/ exists and its content is just
`0  *  *  *  *  /home/yi-hack/script/clean_records.sh 5`

I tried the new feature by adding in "Configurations" page the following:
`*/5 * * * * root /tmp/sd/watchdog/wd_wifi.sh`
then saved and rebooted. When the cam come back online, that job doesn't appear neither in Configuration page nor in root file, which is unchanged.

this is my wd_wifi.sh
```
#/bin/sh
now=$(date +"%Y_%m_%d")
LOGFILE=/tmp/sd/watchdog/log_$now.log

if ping -c4 -q fritz.box > /dev/null ; then
  echo '['$(date)'] Wifi OK' >> $LOGFILE
else
  echo '['$(date)'] CONNECTION KO. Reboot wifi' >> $LOGFILE
  sleep 1
  /sbin/ifconfig wlan0 down
  # Give interface time to reset before bringing back up
  sleep 10
  /sbin/ifconfig wlan0 up
  # Give WAN time to establish connection
  sleep 10
fi
```
and it is executable. If I manually execute it, it works as expected.

roleoroleo, 2020-12-04:
I tested your line and it works.
Did you clear the cache of the browser?
Check il the system.conf is correctly updated and check if cron file contains the line:
```
cat /home/yi-hack/etc/system.conf
cat /var/spool/cron/crontabs/root
```

fabiosci, 2020-12-04:
thanks, from another browser (so clean cache) it worked and both system.conf and root contains the new job (which is shown also in Configurations page after rebooting the cam).
the only issue now is that the script is not executed. I run a ping for 10 minutes and no packets has been lost (so the wifi connection hasn't been reset) and the log file hasn't been created

roleoroleo, 2020-12-04:
Try to remove "root" from the line.

fabiosci, 2020-12-04:
it works, thanks!
i will adapt the script, monitor the behaviour for some days and revert back to you

cchenga, 2021-01-27:
Hi @fabiosci, I got the same issue with yi-hack-Allwinner. Could you share your solution/workaround?

fabiosci, 2021-01-27:
Hi,
the root cause seems to be not related to the custom firmware, the only workaround i've found is to use a script to check if the connection is available or not (it's a simple ping to my router) and, if not, reset the wifi  interface.
The script is scheduled to run every morning when my router turns on the wifi network.
I haven't had time to go deeper, but experienced another issues: sometimes the wifi connection drops during the day. this could be fixed with the mentioned workaround.

Before closing the ticket I'd like to investigate on the random drops by reading the logs created by the scheduled script

fabiosci, 2021-04-12:
after several months, logs and tests now I can say that the connection is correctly established and maintained day after day.
the crontab feature in the UI allowed me to create and schedule a script to check the wifi connection and, if disconnected, reset the wifi adapter and, if necessary, reboot the cam. this solved my issue, even though it's a workaround.

in case it could help someone else, i post here my current solution:
- crontab configuration:
`*/5 07-23 * * * /tmp/sd/watchdog/watchdog.sh` --> to check the wifi and reboot the cam
`0 3 * * * find /tmp/sd/watchdog/logs/ -mtime +14 -delete` --> to delete older log files

I put the watchdog folder into /tmp/sd to keep it after a firmware update.

This is my watchdog.sh
```#/bin/sh
now=$(date +"%Y_%m_%d")
LOGFILE=/tmp/sd/watchdog/logs/wd_log_$now.log

if ping -c4 -q fritz.box > /dev/null ; then
  echo '['$(date)'] Wifi OK' >> $LOGFILE
else
  echo '['$(date)'] CONNECTION KO. Reboot wifi' >> $LOGFILE
  # Disable wifi interface
  /sbin/ifconfig wlan0 down
  # Give interface time to reset before bringing back up
  sleep 20
  /sbin/ifconfig wlan0 up
  # Give Wifi time to establish connection
  sleep 20
  # Check again the connection
  if ping -c4 -q fritz.box > /dev/null ; then
    echo '['$(date)'] Wifi connection successfully established after reset' >> $LOGFILE
  else
    echo '['$(date)'] Wifi connection cannot be established after reset. Reboot cam' >> $LOGFILE
    sync
    sync
    sync
    killall httpd
    killall mqttv4
    sleep 1
    reboot -f
  fi
fi
```

closed 2021-04-12 by fabiosci

torcuat, 2021-11-10:
> 
> 
> after several months, logs and tests now I can say that the connection is correctly established and maintained day after day. the crontab feature in the UI allowed me to create and schedule a script to check the wifi connection and, if disconnected, reset the wifi adapter and, if necessary, reboot the cam. this solved my issue, even though it's a workaround.
> 
> in case it could help someone else, i post here my current solution:
> 
>     * crontab configuration:
>       `*/5 07-23 * * * /tmp/sd/watchdog/watchdog.sh` --> to check the wifi and reboot the cam
>       `0 3 * * * find /tmp/sd/watchdog/logs/ -mtime +14 -delete` --> to delete older log files
> 
> 
> I put the watchdog folder into /tmp/sd to keep it after a firmware update.
> 
> This is my watchdog.sh
> 
> ```
> now=$(date +"%Y_%m_%d")
> LOGFILE=/tmp/sd/watchdog/logs/wd_log_$now.log
> 
> if ping -c4 -q fritz.box > /dev/null ; then
>   echo '['$(date)'] Wifi OK' >> $LOGFILE
> else
>   echo '['$(date)'] CONNECTION KO. Reboot wifi' >> $LOGFILE
>   # Disable wifi interface
>   /sbin/ifconfig wlan0 down
>   # Give interface time to reset before bringing back up
>   sleep 20
>   /sbin/ifconfig wlan0 up
>   # Give Wifi time to establish connection
>   sleep 20
>   # Check again the connection
>   if ping -c4 -q fritz.box > /dev/null ; then
>     echo '['$(date)'] Wifi connection successfully established after reset' >> $LOGFILE
>   else
>     echo '['$(date)'] Wifi connection cannot be established after reset. Reboot cam' >> $LOGFILE
>     sync
>     sync
>     sync
>     killall httpd
>     killall mqttv4
>     sleep 1
>     reboot -f
>   fi
> fi
> ```
> 
> Thanks roleoroleo for this firmware and your support!

Hi, thanks for this script. I am trying to do it but it doesn't execute. I've tried changing permissions with chmod via SSH, connected as root (without password) and it doesn't seem to be applying the changes:

`/tmp/sd/watchdog # chmod -c 777 watchdog.sh
mode of 'watchdog.sh' changed to 0777 (rwxrwxrwx)
/tmp/sd/watchdog # ls -la
total 96
drwx------    2 root     0            32768 Oct 29 11:35 .
drwx------    7 root     0            32768 Nov 10 13:26 ..
-rwx------    1 root     0              825 Nov 10 13:51 watchdog.sh`

Do you have any idea on why and what I should be doing? Thanks in advance.

roleoroleo, 2021-11-10:
You can't change permission on a fat32 filesystem.
But I think 700 is enough.

torcuat, 2021-11-10:
Thank you. The reason it didn't execute was that I wrote the .sh file in Notepad++ for Windows and it doesn't use the "Unix" file format by default. It can be solved in the Edit menu>EOL conversion.

martineiro23, 2022-09-09:
Hi,

I have the same problem with freezing of my 3 cameras as described above. I have to restart them to make them work again. Could someone explain to me step by step how to upload the watchdog.sh file to my camera?

Thank you

grochib, 2023-02-06:
> Uptime is at ~11d
> 
> [image]

> I don't believe Wifi is the issue. The device is still connected. See screenshot. I checked the system log on my router (OpenWRT based), and see no deauth(s) or disconnect(s) from the camera.
>
> After [...]

Hey @mhensema did you solve this? as far as I can see I have the same problem: the rtsp stream disappears in motioneye for 5 seconds, but the cam stays up and does not reboot. So there is no wifi issue here, as far as I can see. Did you find out what was behind this problem?

taurgis, 2024-03-31:
So I seem to have found a way to stabilise the connection by having 3 things running:
- Check WiFi by pinging, if it fails restart WiFi
- Check static IP, if it is still the one I configured in startup.sh - if not > reset IP
- Check RTSP process and restart it if it is down

This is still with the Cloud enabled and almost everything but RTSP enabled. I will let this run for a week or so and then disable cloud again.

```
now=$(date +"%Y_%m_%d")
LOGFILE=/tmp/sd/watchdog/logs/wd_log_$now.log

if ping -c4 -q google.com > /dev/null ; then
    echo '['$(date)'] Wifi OK' >> $LOGFILE
else
    echo '['$(date)'] CONNECTION KO. Reboot wifi' >> $LOGFILE
    # Disable wifi interface
    /sbin/ifconfig wlan0 down
    # Give interface time to reset before bringing back up
    sleep 20
    /sbin/ifconfig wlan0 up
    # Give Wifi time to establish connection
    sleep 20
    # Check again the connection
    if ping -c4 -q google.com > /dev/null ; then
        echo '['$(date)'] Wifi connection successfully established after reset' >> $LOGFILE
    else
        echo '['$(date)'] Wifi connection cannot be established after reset. Reboot cam manually' >> $LOGFILE
    fi
fi

if ps | grep [r]RTSPServer > /dev/null
then
    echo '['$(date)'] RTSP Running' >> $LOGFILE
else
    echo '['$(date)'] RTSP KO. Reboot RTSP' >> $LOGFILE
    /tmp/sd/watchdog/restart.sh &
fi

if ifconfig wlan0 | grep -oE "192.168.0.227" > /dev/null ; then
      echo '['$(date)'] IP OK' >> $LOGFILE
else
      echo '['$(date)'] IP NOT OK, RESETTING' >> $LOGFILE
      ifconfig wlan0 192.168.0.227 netmask 255.255.255.0 up
      route add default gw 192.168.0.1
      echo "nameserver 192.168.0.1" >> /tmp/resolv.conf
fi
```

restart.sh
```
now=$(date +"%Y_%m_%d")
LOGFILE=/tmp/sd/watchdog/logs/wd_log_$now.log

echo '['$(date)'] Restarting RTSP' >> $LOGFILE
killall wd_rtsp.sh
killall rRTSPServer
sleep 60
rRTSPServer -r low -a aa
```

[5 more replies]