roleoroleo/yi-hack-MStar #150: Timezone for UK

leeablett, 2020-04-18:
Hi there,

I live in the UK and when I set the timezone to "GMT0BST,M3.5.0/1,M10.5.0" it is either not setting or one hour out.

roleoroleo, 2020-04-18:
When you set it and reload configuration page, can you see the correct value?

leeablett, 2020-04-18:
No, it looks like the value is not saving....

leeablett, 2020-04-18:
how odd, tried IE and it worked but there seems to be an issue with Chrome on MS Windows 10.

leeablett, 2020-04-18:
Look like there may be a js error on chrome... DevTools failed to parse SourceMap: chrome-extension://hdokiejnpimakedhajhdlcegeplioahd/sourcemaps/onloadwff.js.map

leeablett, 2020-04-18:
Ignore that error as that was lastpass and now removed. Still does not show the value in chrome though.  Have set through IE which works fine.

roleoroleo, 2020-04-18:
Interesting.

closed 2020-05-05 by roleoroleo