roleoroleo/yi-hack-MStar #151: Availability to use the firmware with 47US

kabakakao, 2020-04-19:
Hi,
I really like the firmware and the feature for the yi-cams. I have a BFUS running with this firmware.

I have 2 additional 47US yi-cams running with  yi-hack-v4. But I am missing some features of yi-hack-MStar.
Is there a way to adopt the features to the yi-hack-v4?
I would really appreciate that

roleoroleo, 2020-04-19:
It depends on the feature.
Some feature are strictly related to the hardware.
But you have to ask to the mantainer of the project.

kabakakao, 2020-04-19:
I really like the Full Integration of rtsp. 
And the best Feature is the cleanup of old videofiles, when  the sd Card is nearly full

roleoroleo, 2020-04-19:
RTSP is not portable.
SD cleanup yes.

kabakakao, 2020-04-21:
Can you give me some hints, how to do that?
Also the Snapshot function and the mqtt feature are quite nice.
For me it seems, that the original repo isn't active anymore. So I will try it on myselves and create a PR afterwards.
Is it complicated to build the files? What, if I brick my device? Is it possible? Are there ways to recover the cam?

roleoroleo, 2020-04-22:
See here:
https://github.com/TheCrypt0/yi-hack-v4/wiki/Project-Setup-Hints

closed 2020-05-07 by roleoroleo