roleoroleo/yi-hack-MStar #152: 9FUS unable to get wifi persistent

sfl3005, 2020-04-20:
Hi roleo,

I start my 1080p camera with y203c 0.3 on sd card, I try to add in xiaomi application but after device is connected to wifi (LED is blue) mobile phone application times out.
Yi-hack admin page is reachable but after camera restart I still heard "welcome to xiaoyi camera waiting to connect" and wifi connection is not persistent.

if I look to SD card, there is several logs files.
I try to install an earlier y203c version but y203c is still present with same version. It seems like I am unable to flash an other time.

I have SSH and telnet access to camera after wizard boot.
I tried on two SD card one of 16G and one of 8G same issue.
sadly,  disabled cloud feature or not is same issue 
[image]
[image]
[image]

Please help me 
Kind regards
Sfl

roleoroleo, 2020-04-20:
Try to reset the fw with the reset button.

sfl3005, 2020-04-20:
Already done. Reset successful message heard but no change to the issue. Unable to add to xioami app.

sfl3005, 2020-04-20:
I downgrade the camera successfully
[image]

And upgrade it online throught web interface but same issue
Every time camera reboot runs xiaomi startup wizard

sfl3005, 2020-04-21:
I was used wrong mobile apps. xiaomi home vs YI HOME.
with the correct apps every thing works like a charm !

What a shame !
Thanks to have take time to read !

closed 2020-04-21 by sfl3005