roleoroleo/yi-hack-MStar #153: API

wiktorvip, 2020-04-21:
Is there any plan in future to add api, to be able manage camera remotely via cmd.

roleoroleo, 2020-04-21:
For example?

wiktorvip, 2020-04-21:
for example start/stop RTSP, or any other service. make some configuration like Timezone, so almost everything what is available via page, and some handler to get info about configuration in json format.

roleoroleo, 2020-04-21:
Timezone web configuration will be in the next release.
No plans for other features at the moment.
But contributions are welcome.

roleoroleo, 2020-04-22:
You can get also the config files:
http://192.168.1.243:8080/cgi-bin/get_configs.sh?conf=camera
http://192.168.1.243:8080/cgi-bin/get_configs.sh?conf=mqtt
http://192.168.1.243:8080/cgi-bin/get_configs.sh?conf=system
http://192.168.1.243:8080/cgi-bin/get_configs.sh?conf=proxychains

wiktorvip, 2020-04-22:
yes, this I already discovered.
I was playing little bit with changing parameter but did worked:
example:
`http://IP:8080/cgi-bin/set_configs.sh?conf=camera&led=no`

I had not time to troubleshooting more.

roleoroleo, 2020-04-22:
set_configs.sh needs a POST with a json file.

roleoroleo, 2020-04-22:
New wiki:
https://github.com/roleoroleo/yi-hack-MStar/wiki/Web-services-description

roleoroleo, 2020-04-22:
Do you want to try this?
[service.sh.txt](https://github.com/roleoroleo/yi-hack-MStar/files/4516784/service.sh.txt)

Examples:
http://IP:PORT/cgi-bin/service.sh?name=rtsp&action=start&param1=high
param1 is resolution
http://IP:PORT/cgi-bin/service.sh?name=onvif&action=start&param1=both&param2=yes
param1 is the profile and param2 is watermark
http://IP:PORT/cgi-bin/service.sh?name=mqtt&action=stop
http://IP:PORT/cgi-bin/service.sh?name=ftpd&action=stop&param1=busybox
http://IP:PORT/cgi-bin/service.sh?name=ftpd&action=start&param1=pure-ftpd

AirmakZ, 2020-04-27:
Hi @roleoroleo,
Incredible job with the customization of this hack firmware! Congratulations for the project.

Taking profit of this post I'd like to ask regarding this feature (Web Services).
All is working according wiki documentation, but I have a doubt.

Is there a way to read current status of Camera Setting?
I'm updating the value "switch_on" properly, but I could not find a way to read this parameter and confirm execution.

I tried a workaround using Configuration parameters, but unfortunately "get_configs.sh?conf=camera" is not updated in real time.

The purpose it to add a switch in Home Assistant like this:

```
  - platform: command_line
    switches:
      camara_garaje:
        command_on: "/usr/bin/curl -X POST http://IP:8080/cgi-bin/camera_settings.sh?switch_on=yes"
        command_off: "/usr/bin/curl -X POST http://IP:8080/cgi-bin/camera_settings.sh?switch_on=no"
        # WORKAROUND IF GET_CONFIGS IS REAL TIME UPDATED
        command_state: "/usr/bin/curl -X GET http://IP:8080/cgi-bin/get_configs.sh?conf=camera"
        value_template: '{{ value_json.SWITCH_ON == "yes" }}'
```

Is there currently a proper way to do it? Of could you include in your task list a Camera Settings Read function if it's able technically?

Thanks a lot!

roleoroleo, 2020-04-27:
The method "get_configs.sh?conf=camera" must be updated in real time.
This api write the file /home/yi-hack/etc/camera.conf immediately.
And send an ipc message to the cam to update the status.

AirmakZ, 2020-04-27:
mmmm I cannot see this behaviour.
```
SEND: http://user:pass@192.168.1.157:8080/cgi-bin/camera_settings.sh?switch_on=no
REPLY: { }
SEND: http://user:pass@192.168.1.157:8080/cgi-bin/get_configs.sh?conf=camera
REPLY: {
  "SWITCH_ON": "yes",
  "SAVE_VIDEO_ON_MOTION": "yes",
  "SENSITIVITY": "low",
  "LED": "yes",
  "ROTATE": "no",
  "IR": "yes",
  "NULL": "NULL"
}
```
Even playing with LED I cannot see the update in /home/yi-hack/etc/camera.conf. I connected by FTP to see file modification date and it's not update.
In the web interface I cannot see a value change neither.

Another test to made I to turn off/on the camera.
If I switch OFF the LED using camera_settings.sh, after restart the camera the LED still keeps OFF, but in config (file and get_configs) it's ON.

It's like if the first step you mention is not executed.... but why after restart the camera LED is not activated as camera.conf says?

roleoroleo, 2020-04-27:
Ok, I understand the problem.
Probably the documentation is not ok.
camera_settings.sh ONLY sends the ipc command to the cam: doesn't write the configuration to file.
You must use "set_configs.sh?conf=camera" sending a post message with a json content (see get_configs.sh).

AirmakZ, 2020-04-27:
yeah.. it seems.
If so there isn't a way to confirm if the camera_settings.sh action has been applied.
If I change config with another command, it's not a valid way I think.

Update the config as the first step same as you mention in the previous comment seems the proper way.

Is there a tech way to improve this point? Is fo I can open a new ticket for this enhancement?

roleoroleo, 2020-04-27:
I don't understand the aim.
Do you want to change a setting temporarily?
Or permanently?

camera_settings.sh changes a setting temporarily. It send an ipc command and there is no way to know if the system receive it. So, no feedback.

set_configs.sh changes the settings to a file but doesn't apply it.

As you can see in the file  camera_settings.js there are both web service calls.

roleoroleo, 2020-04-28:
I could change set_configs.sh to call automatically camera_settings.sh

AirmakZ, 2020-04-28:
Basically the control I'd like to do is permanent. I mean, I want to control camera ON and OFF with an automation (Alarm activation) from Home Assistant.
The camera will be always plugged, and if power supply is lost ideally is better to initilize it as last status.

As you propose, If after change a parameter with set_configs.sh the corresponding IPC command is sent internally, it can be a solution (I can read again camera_settings and confirm the new value)
Just I feel it's not same behaviour Web interface, where if you change a camera config and click save, action is only executed after restart.

Could it be opposite? I mean, after call camera_settings, update the configs...

In this condition seems we're loosing 'volatile' configuration function... but...  yesterday I noticed that after change LED status (with camera_settings.sh), it keeps status after DC off/on.... It's strange because config file is not updated... and It's not matching with the theory of your explanation.

roleoroleo, 2020-04-28:
When you send an ipc command, the cam save the state internally (in flash).
And when it restarts I don't need to resend the command.

Ok, I will change the behaviour in the next release.
Only 1 command to set file and apply change.
If I were good I would delete the camera.conf file and go to read directly in flash.

AirmakZ, 2020-04-28:
Understood now the camera_settings behaviour.

Thank you very much for considering this modification !!!

roleoroleo, 2020-04-29:
See here:
https://github.com/roleoroleo/yi-hack-MStar/issues/160#issuecomment-621295142

roleoroleo, 2020-05-24:
Changes added to the last firmware.

closed 2020-05-24 by roleoroleo