roleoroleo/yi-hack-MStar #154: Question: Camera supported

joekerna, 2020-04-21:
I have a quick question:
Am I right to believe that this hack will work on the following camera?
[Yi Home Camera 3](https://www.yitechnology.com/yi-home-camera-3-y25/)

roleoroleo, 2020-04-21:
To know if the xam is supported, I need a log of the boot.
See the wiki.

joekerna, 2020-04-21:
I don't have it yet. I want to make sure, I buy one that's supported.
The link says 
> Home Camera Y25

Does that _y25_ mean anything?

roleoroleo, 2020-04-21:
For my knowledge, it's not possible to know the exact model before buying.

joekerna, 2020-04-21:
So it's a matter of luck if it's supported?

roleoroleo, 2020-04-22:
Yi produces cams with the same name but based on different technologies.
The commercial name does not correspond to a specific model (intended as a part number) but to a series of models sometimes based on very different hardware.

closed 2020-05-07 by roleoroleo