roleoroleo/yi-hack-MStar #155: Camera Model 9FUSY211PB

alexmc1510, 2020-04-22:
Hello all,

As you mentioned, I have followed all steps to install y203 firmware but orange light does not stop to blink.
To problem is in the moment I received the camera I did not tried it with stock firmware and now I am not able to connect to the app. I don't know  the real stock firmware it had.

Do you have any suggestion?

Thanks

roleoroleo, 2020-04-22:
The hack contains all the original fw features.
So, if should be able to start the pairing with the app.
Something may have failed in the update procedure, try to apply again the hack changing version (i.e. if you used 0.3.1, try 0.2.9 or 0.3.0).

alexmc1510, 2020-04-22:
no luck....which version should I use? 203c or 25 and when I know that camera is flashed because orange led is blinking always. What I have noticed is that depending on the firmware, speed of led is different...

alexmc1510, 2020-04-22:
Some advances....with y25 firmware camera left a log folder. do you want to have a look at it?

roleoroleo, 2020-04-22:
Ok.
Post the log.

alexmc1510, 2020-04-22:
[ps1.txt](https://github.com/roleoroleo/yi-hack-MStar/files/4517685/ps1.txt)
[df.txt](https://github.com/roleoroleo/yi-hack-MStar/files/4517687/df.txt)
[file.txt](https://github.com/roleoroleo/yi-hack-MStar/files/4517689/file.txt)
[homever.txt](https://github.com/roleoroleo/yi-hack-MStar/files/4517690/homever.txt)
[log.txt](https://github.com/roleoroleo/yi-hack-MStar/files/4517691/log.txt)

Homever has been renamed just to avoid problems when uploading.
I think this log folder has been created because I click on Reset with SD card inside. At list is something....

TimDowker, 2020-04-22:
I have the same issue here.  9FUS model (YYS.2016) and firmware version 8.2.0.0A_201912270941

alexmc1510, 2020-04-22:
The problem is I have a pretty stone on my desktop....

alexmc1510, 2020-04-22:
My firmware....If homever file is ok is 4.5.0.0C_201910080934

roleoroleo, 2020-04-22:
> I have the same issue here. 9FUS model (YYS.2016) and firmware version 8.2.0.0A_201912270941

This hack doesn't work on your cam.
See this:
https://github.com/roleoroleo/yi-hack-Allwinner

@alexmc1510 
You model suffix is y203c, not y25.
Try again.

TimDowker, 2020-04-22:
@roleoroleo  Thank you!

alexmc1510, 2020-04-22:
Yes, my current file is 203c without success...

alexmc1510, 2020-04-22:
My cam does not say anything, does not speak

alexmc1510, 2020-04-23:
Any idea of what can I try?

roleoroleo, 2020-04-23:
Send me your email, I will send you my stock firmware.

roleoroleo, 2020-04-23:
But probably, the stock fw doesn't solve your problem.

alexmc1510, 2020-04-23:
Here you are,

alex.mc_1510@hotmail.com

I will be very greatful

roleoroleo, 2020-04-23:
Try the stock fw.

alexmc1510, 2020-04-23:
I have not received anything...
alex.mc_1510@hotmail.com

alexmc1510, 2020-04-23:
Did youu send it?

roleoroleo, 2020-04-23:
Yes, I sent it.
-------- Messaggio originale --------Da: alexmc1510 <notifications@github.com> Data: 23/04/20  17:53  (GMT+01:00) A: roleoroleo/yi-hack-MStar <yi-hack-MStar@noreply.github.com> Cc: roleo <roleo@iperbole.bologna.it>, Mention <mention@noreply.github.com> Oggetto: Re: [roleoroleo/yi-hack-MStar] Camera Model 9FUSY211PB (#155) 
Did youu send it?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/roleoroleo/yi-hack-MStar/issues/155#issuecomment-618479344",
"url": "https://github.com/roleoroleo/yi-hack-MStar/issues/155#issuecomment-618479344",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

bonuzzz, 2020-04-23:
@alexmc1510 what exactly camera do you have?  yi home camera 3?
I have 2 9fus and the work perfectly with hacked firmware and app. blinking yellow led means that camera is waiting qr code with wifi info. i use yi home app from google play, not mi home.

alexmc1510, 2020-04-23:
@roleoroleo I have sent you and email.

@bonuzzz  : I have yi home 1080p 9FUS. The matter is that I think the firmware is downloaded because orange led blinks slowly the first 30 seconds, after this, I seems that camera restarts and start blinking faster. The problem is I cannot hear anything from the camera so I don't know when I should put my phone with the barcode in front of the camera. I am using YI home APP.

Thanks in advance

bonuzzz, 2020-04-23:
@alexmc1510 you will not hear anything with hacked firmware. So,when yellow led is fast blinking, you should put your phone with qr code  in front of camera until led changes color to blue. And solid blue color means that camera successfully connected to wifi.

alexmc1510, 2020-04-23:
No luck....I have put my phone in front of the camera for about 2 min and led does not change....

bonuzzz, 2020-04-23:
it should. it depends on phone's position. try to move farther. and if it catch qr code it changes led color immediately. not need to stay qr code in one position.

orditeck, 2020-04-23:
https://github.com/roleoroleo/yi-hack-Allwinner worked well for my 9FUS (YYS.2016) with firmware version 8.2.0. Bought my cam on Amazon Canada a few days ago. Thanks @roleoroleo for pointing it out!

alexmc1510, 2020-04-23:
I promise tomorrow I will record a movie with all the steps I am following just to verify I am not stupid...
Thanks to all for your support.

alexmc1510, 2020-05-05:
Finally I returned the device and change by a new one. New one came with 8.2 firmware so I followed Yi-hack-Allwinner and now, the new camera is working!!
Thanks for your support and I think something happened during first hack that blocked the device.
Regards,

closed 2020-05-05 by alexmc1510