roleoroleo/yi-hack-MStar #160: set_configs from Web Services not updating camera.conf file content

AirmakZ, 2020-04-29:
Hi @roleoroleo !

I have a doubt regarding web service 'set_configs'.
While waiting you to try the implementation we discussed in the post API #153, I'm trying to win time from my side staring the implementation of the configurations write.

I build a REST command to use 'set_configs', as for example:
`curl -i -X POST -H "Content-Type: application/json" -d "{{\"SWITCH_ON\":\"yes\",\"SAVE_VIDEO_ON_MOTION\":\"yes\",\"SENSITIVITY\":\"low",\"LED\":\"no\",\"ROTATE\":\"no\",\"IR\":\"yes\",\"NULL\":\"NULL\"}" http://IP:8080/cgi-bin/set_configs.sh?conf=camera`
or 
`curl -i -X POST -H "Content-Type: application/json" -d "{\"LED\":\"no\"}" http://IP:8080/cgi-bin/set_configs.sh?conf=camera`

The response is `{"error":"false"}`, and I can see though FTP the **camera.conf file modification date updated**.
But, if open the file, the content inside is same... (I'm trying LED OFF just as dummy parameter)

I used another tool with UI to build the REST command (Insomnia Core) and I have the same result.

Am I doing something wrong? I'm missing something?
I'm using Yi Home 1080p 9FUS with v0.3.1

Thanks for your support !

roleoroleo, 2020-04-29:
Very strange that the modification date is changed.
As discussed in the other thread camera_settings doesn't change the file.
But if you want try the new camera_settings.sh script, it's ready:

```
#!/bin/sh

YI_HACK_PREFIX="/home/yi-hack"
CONF_FILE="$YI_HACK_PREFIX/etc/camera.conf"

CONF_LAST="CONF_LAST"

for I in 1 2 3 4 5 6
do
    CONF="$(echo $QUERY_STRING | cut -d'&' -f$I | cut -d'=' -f1)"
    VAL="$(echo $QUERY_STRING | cut -d'&' -f$I | cut -d'=' -f2)"

    if [ $CONF == $CONF_LAST ]; then
        continue
    fi
    CONF_LAST=$CONF
    CONF_UPPER="$(echo $CONF | tr '[a-z]' '[A-Z]')"

    sed -i "s/^\(${CONF_UPPER}\s*=\s*\).*$/\1${VAL}/" $CONF_FILE

    if [ "$CONF" == "switch_on" ] ; then
        if [ "$VAL" == "no" ] ; then
            ipc_cmd -t off
        else
            ipc_cmd -t on
        fi
    elif [ "$CONF" == "save_video_on_motion" ] ; then
        if [ "$VAL" == "no" ] ; then
            ipc_cmd -v always
        else
            ipc_cmd -v detect
        fi
    elif [ "$CONF" == "sensitivity" ] ; then
        ipc_cmd -s $VAL
    elif [ "$CONF" == "led" ] ; then
        if [ "$VAL" == "no" ] ; then
            ipc_cmd -l off
        else
            ipc_cmd -l on
        fi
    elif [ "$CONF" == "ir" ] ; then
        if [ "$VAL" == "no" ] ; then
            ipc_cmd -i off
        else
            ipc_cmd -i on
        fi
    elif [ "$CONF" == "rotate" ] ; then
        if [ "$VAL" == "no" ] ; then
            ipc_cmd -r off
        else
            ipc_cmd -r on
        fi
    fi
    sleep 1
done

printf "Content-type: application/json\r\n\r\n"

printf "{\n"
printf "}"
```

AirmakZ, 2020-04-29:
I'm not using **camera_settings** this time.
I'm playing with **set_configs.sh** (so I created new post)

AirmakZ, 2020-04-29:
By the way, the new script is working like a charm !!!

roleoroleo, 2020-04-29:
Ok.
I checked the sources and I realized now that the payload is not a json message.
Sorry but I didn't write this piece of code.
The payload is `LED=yes&IR=no`
Please use the new camera_settings.sh I posted above beacuse I will delete/change set_configs.sh in the next release.

AirmakZ, 2020-04-29:
Ah !!!! Undestood !!!! I focused in JSON and didn't try to add the content in the query. 

Sure, all my 3 cameras are already updated and Home Assistant automation is working perfectly.

Waiting for the next release !!! Good job guy!

closed 2020-04-29 by AirmakZ

xfceKris, 2020-06-10:
> Ah !!!! Undestood !!!! I focused in JSON and didn't try to add the content in the query.
> 
> Sure, all my 3 cameras are already updated and Home Assistant automation is working perfectly.
> 
> Waiting for the next release !!! Good job guy!

I know this issue is closed, but I need to know how you got automation working with home assistant? I know the issue I'm having is more on the home assistant configuration side, and not the firmware side, but I can't seem to get HA to even see the camera. I'd like for HA to be able to record video to an external drive when motion is detected if that's possible.

roleoroleo, 2020-06-10:
Add the cam as ONVIF.
Configure MQTT on the cam.
Add a binary sensor to HA.

```
binary_sensor:
  - platform: mqtt
    name: "Motion detection - kitchen"
    state_topic: "home-assistant/kitchen/motion_detection"
    payload_on: "ON"
    qos: 1
    off_delay: 10
```

You can't record a video directly in an external drive but you can get it using web-services.
See the wiki.