roleoroleo/yi-hack-MStar #161: Yi 1080p Dome BFUS 4.6.0 record without cloud not working

majkers, 2020-04-30:
Hi,
it seems to me that recording without cloud is not working. I have camera as in subject and this is what I have set:
- _Disable cloud_ set to **enabled**
- _Recording without Cloud_ set to **enabled**
- _Save video when a motion is detected_ set to **disabled**
Motion is detected correcly as I receive it via MQTT but on Motion Events section I don't see any new rocordings. Event if I reenable cloud and run Yi home app I don't see any recording history ie. I can't rewind backwards.

majkers, 2020-04-30:
Workarounds from https://github.com/roleoroleo/yi-hack-MStar/issues/93 did not work for me :(

roleoroleo, 2020-04-30:
My cam is configured exactly as you described and it works.
Check if mp4record service is running.

majkers, 2020-04-30:
Can You please tell my how?

roleoroleo, 2020-04-30:
Yes. mp4record is not running.
Try to run it manually:
```
# cd /home/yi-hack
# ./mp4record &
```

majkers, 2020-04-30:
I don't have it on any camera I have:

/home/yi-hack/bin # cd /home/yi-hack
/home/yi-hack # ./mp4record &
/home/yi-hack # -sh: ./mp4record: not found

majkers, 2020-04-30:
Ok I know what was wrong. Correct path is /home/yi-hack/app. Now it seems to record. But why it doesn't automatically?

roleoroleo, 2020-04-30:
It's a good question.
Probably it starts and then crashes.

roleoroleo, 2020-04-30:
I remember another user with a similar problem.
Try to change the sd card.

majkers, 2020-05-01:
I have to wait for the new one. How about adding some extra logs? Maybe they will help?

majkers, 2020-05-25:
OK I know what is the problem. Record without cloud works only when time serwer is enabled. It should be validated that both must be enabled. But I have one more problem. When I restart my camera, recording is not working once again. Then I have to delete all records ie. alle dictionariues and tmp file and restart it one more time and then it works ok.

roleoroleo, 2020-05-25:
You could try to start mp4record after time server.
Append at the end of system.sh these commands:
```
killall mp4record
cd /home/app
./mp4record &
```

closed 2020-08-20 by roleoroleo