roleoroleo/yi-hack-MStar #163: FW does not load on BFUSY YI HOME 1080P

LIR131, 2020-04-30:
Hi everybody,

I tried to install the yi-hack-MStar 0.3.1 following instructions.
Led does not blink yellow.
After power on the led is yellow for about 15 seconds, and after blinking blue, and finally stay blue.

Could someone give some help plz?

Note that my current fw is 8.2.0.0A_201912270941

michael11123, 2020-05-01:
you use 16G or 32G SD?

LIR131, 2020-05-01:
32G SD Samsung EVO Plus

roleoroleo, 2020-05-01:
Your cam is not supported.
See here: https://github.com/roleoroleo/yi-hack-Allwinner

LIR131, 2020-05-01:
Ok thanks.
All winners flash it.
So rtsp is not reachable.
Can't watch the stream in vlc. Do allwinner-yi-hack implement rtsp for my camera?

roleoroleo, 2020-07-17:
If the issue is not solved, please open a new issue in the allwinner repo.

closed 2020-07-17 by roleoroleo