roleoroleo/yi-hack-MStar #164: Yi 1080p Dome BFUS 4.6.0 setting timezone not working

majkers, 2020-05-01:
Hi
setting timezone in Yi 1080p Dome BFUS 4.6.0 is not working: Mys tring is:

`CET-1CEST,M3.5.0,M10.5.0/3`

and it is set ok. I have it when I reload page and in file /home/yi-hack/etc/TZ. But when I connect via SSH and fire date commnad I get:

`Fri Apr 17 22:28:49 CEST 2020` while it should be 1st of May 10:04... Even if I use this command:

`echo CET-1CEST,M3.5.0,M10.5.0/3  > /etc/TZ`

and reboot I still have the same date. Unsetting timezone makes no difference. Date stays the same

roleoroleo, 2020-05-01:
I think the problem is not the timezone but the time.
As you can see the timezone is correct: "CEST".
Check if ntpd is running properly.

majkers, 2020-05-01:
That was it. Thanks

closed 2020-05-01 by majkers