roleoroleo/yi-hack-MStar #165: YI 1080 9FCN error

michael11123, 2020-05-01:
hi everyone.
i just bought a yi 1080 9FCN. 
.i installed y203c to hack my Yi.
it's ok until " conneted wifi" and then a voice of chinese rang out.
so i finish process.
please support me.
thanks you so much.

michael11123, 2020-05-01:
[image]
Picture

michael11123, 2020-05-01:
https://youtu.be/b3MEIlOYarY

roleoroleo, 2020-05-01:
9FCN means that the cam is for the Chinese market.
Maybe other users with 6FCN can give you an answer.

michael11123, 2020-05-01:
I hope so :(

bonuzzz, 2020-05-01:
if wifi is connected so you should open camera web interface. btw my cams are not speaking at all with this custom firmware, so if web interface is not available, firmware probably has not been flashed successfully.

michael11123, 2020-05-01:
Camera talk only be used in china. And the fw has not been flashed. So i can access web interface.

menway, 2020-05-19:
> https://youtu.be/b3MEIlOYarY

Any luck with your camera? ^_^

What the Chinese voice said was: binding failed, wrong device type selected.

roleoroleo, 2020-10-23:
Closed due to inactivity.

closed 2020-10-23 by roleoroleo