roleoroleo/yi-hack-MStar #167: Setting timezone vs. yi app and rewind option

majkers, 2020-05-03:
Hi is possible for You to let someone set timezone via Your firmware and make rewind/playback option in yi app work properly? I have found this issues https://github.com/shadow-1/yi-hack-v3/issues/158 and https://github.com/TheCrypt0/yi-hack-v4/issues/68 but without solution. Is it possible?

roleoroleo, 2020-05-03:
No sorry. It's not possible.
If you often use the app you should not change the timezone.

closed 2020-05-03 by majkers

Igonzalezz, 2022-02-09:
I know this is an old tread but this can be achieve by using a mod version of the Yi Home..

majkers, 2022-02-09:
> mod version of the Yi Home

Any link?

Igonzalezz, 2022-02-09:
Can't find the original link but it's the APK that I has been using since I changed the TZ file in the /etc path. I got that APK from a Russian thread from 4pda.ru.. hope this can help you..  [Link](https://drive.google.com/file/d/1BoUXBe2nBkeIJtOLOPNz-fmSJQAzREQ_/view?usp=drivesdk)