roleoroleo/yi-hack-MStar #172: Baby crying detention

jahegu, 2020-05-08:
hello,

official software includes baby crying detention. Would it be possible to add this option on MQTT alerts?

BR,

roleoroleo, 2020-05-08:
Which model do you have?
In my model it's not available.

jahegu, 2020-05-08:
BFUS and 9FUS

roleoroleo, 2020-05-10:
Try the last version.

jahegu, 2020-05-10:
Thank you very much. I'll comment as soon as any of my Baby's cry.

BR,

roleoroleo, 2020-05-10:
I tested it with this:
https://www.youtube.com/watch?v=iVeKq9sriuA

PieVo, 2020-05-10:
Besides the baby-cry detection, it would eventually be possible to do basic audio level-triggering since we continuously read audio in tinyalsa.

jahegu, 2020-05-11:
I have checked several times. Native application trigger but I do not get any MQTT specific alarm. 
Movement messages arrives correctly.

roleoroleo, 2020-05-11:
https://github.com/roleoroleo/yi-hack-MStar/issues/175

jahegu, 2020-05-14:
I confirm that it worked on 0.3.3

Thank you very much for the enhacement!

kylehase, 2020-08-19:
I just picked up a Yi 1080p Dome BFUS and loaded 0.3.5 but only got baby_crying to publish once. I'm also not receiving crying notifications in the Yi app since that one. The mic is set to on in the app and baby crying detect is enabled in both the app and web UI. What steps can I take to troubleshoot this issue? I've tried disabling RTSP audio but that made no difference. Even went as far as to try downgrading to 0.3.4, which also didn't work.

kylehase, 2020-08-24:
It turned out to be both sensor reset duration as well as crying amplitude which caused the mqtt failure. Sensor reset duration seems to be >1min even when "alert frequency" is set to "high" in the Yi app. Also crying had to be surprisingly loud in order to trigger the alert. Unfortunately I believe these issues are within the Yi part of the firmware and likely not configurable beyond what is exposed in the Yi app.

For my use case, I'll try using an ffmpeg noise trigger from the RTSP stream instead.

roleoroleo, 2020-08-24:
I believe your analysis is correct.
The baby crying and human detection features are not working well.

closed 2020-11-21 by roleoroleo