roleoroleo/yi-hack-MStar #174: Is there a method to dump the camera firmware without serial if we can telnet to the camera?

avm99963, 2020-05-09:
Hi!

I have a camera which is not in the "supported devices" list in this repo, and I would love to run this custom firmware, which is an absolutely amazing work! However, I don't have a ttl-rs232/usb adapter, but I've been able to telnet to the camera by creating a sh script (based in the one found in the [xmflsct/yi-hack-1080p](https://github.com/xmflsct/yi-hack-1080p/blob/master/sd/test/factory_test.sh) repo) and placing it in `test/factory_test.sh` the SD card.

I'm sure it is a MStar based camera (at least that's what `/cat/cpuinfo` says):

```
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 12.04
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : MStar Infinity3 (Flattened Device Tree)
Revision        : 0000
Serial          : 0000000000000000
```

So I'm thinking about building this custom firmware myself for the camera and see how well it works, but for that I'd need to dump the contents of the camera. So, can anybody give me a little bit of advice on what tools I could use to dump the firmware? That would be very much appreciated!

In the wiki it says "To save these files without a previous hack, you need to extract them through a serial connection", so that gave me hopes that maybe there's a way without using a serial connection. In any case, if this is not feasible I'll get a ttl-rs232/usb adapter and try to solder some cables to the camera, although I'm so bad at soldering I will probably break something :P

Anyways, if I get it working I'll contribute the firmware file I dumped (I could send it to you @roleoroleo) so it can be used to compile future custom firmwares for that camera. Also, that way I could also contribute to the repo by adding functionality which is missing (although I see the firmware is quite complete, but I'd love to add more functionality if I see something's missing).

Thank you very much in advance!

roleoroleo, 2020-05-09:
Simply use dd to dump the image on the SD.
Check how many partitions there are with:

`cat /proc/mtd`

Then dump every partition with dd:

```
dd if=/dev/mtd/mtd0 of=/tmp/sd/mtd0.bin
dd if=/dev/mtd/mtd1 of=/tmp/sd/mtd1.bin
...
```

The path of mtd devices could be different.

avm99963, 2020-05-09:
Thanks, that worked! :)

I dumped all the partitions and used jefferson/mkfs.jffs2 to delete some binaries I added to the camera (which weren't in the original firmware) and delete the `back.bin` file.

I was able to compile all the programs and pack the firmware, but it seems like the camera doesn't accept the new firmware files. I've been troubleshooting it, here are some shell scripts I extracted from the original firmware and an 'output.txt' file where you can see the troubleshooting steps I've followed: https://gist.github.com/avm99963/35f910cca5e13410c5a8b7b4ade09262

It seems like the updater expects the home_h305m file to have the following structure:
* The first 24 bytes should be the version string (`verin`).
* The next 1024 bytes should be an encoding key (`enckey`).
* The rest is an encoded binary which is decoded by the updater using the encoding key. Then, the decoded binary should contain the following:
  * The first 33 bytes are the md5 hash for the 7-zip file, which is described below (`md5`).
  * The next 33 bytes are a password used later on (`pwd`).
  * The next 24 bytes are a version string (`verout`).
  * The rest is a 7-zip password protected file, whose md5 is `md5`.

The updater parses the file with the structure previously defined, and then performs the following verifications:
1. `ver_in == ver_out`. Otherwise, the updater exits.
2. `ver_in  != old_ver`, where `old_ver` is the version of the current old firmware. If it fails (the firmware being installed is the same already installed), the updater exits.
3. If the actual md5 of the 7-zip file does not match the `md5` value, the updater exits.

After this, the updater decompresses the 7-zip file using the `pwd` password and calls the `update.sh` script which actually performs the update. However, I just realized that the `sys_h305m` file is not mentioned anywhere, so maybe the update via the sd card is handled before running the `init.sh` script? If this was the case, I might have put the wrong suffix to the `home` and `sys` files, and that would be why it is not flashing the firmware. I thought I figured out the suffix was "h305m" by looking into the `init.sh` file, where it is referenced, but maybe I'm wrong.

Anyways, the camera is actually branded as an IEGeek 82, with codename IE82 (https://www.amazon.es/ieGeek-Vigilancia-Detecci%C3%B3n-Movimiento-Vigilabebes/dp/B07VJFXNTC). However, the cloud app being used is "Yi IOT" (which is no longer available), and the file structure and everything corresponds to a typical YI camera. I've tried renaming the firmware files to `sys_ie82` and `home_ie82` but that didn't work neither.

Thank you so much again for your help!

avm99963, 2020-05-09:
Just to cross-reference it, I just found [this issue](https://github.com/roleoroleo/yi-hack-MStar/issues/91#issuecomment-602728457) where the same updater behavior is discussed. So I was right about my guess that there was possibly another updater before the `init.sh` script!

But changing the names to `sys_h305` and `home_h305` didn't work neither, the camera just boots normally :/ The thing is I'm using a camera based on MStar and not Allwinner.

Tomorrow I'll try to replicate what you implemented for Allwinner cameras and see if it lets me update the firmware.

roleoroleo, 2020-05-10:
There are 2 different upgrade methods: from u-boot and form linux system.
The method I use in the hack is the 1st while the one used by yi is the 2nd.
With the 1st method you don't need a file with the strcuture that you correctly described above, you simply need a jffs2 image.
Pay attention that the file name is different (from 1st to second method).
Normally the second method has an "m" to the end of the name.
You have to connect the serial cable and check the name of the file requested by u-boot.
If doesn't work, try to change SD card. Many users have problems with sd cards.

aalbul, 2020-05-16:
@roleoroleo i have a question. If i manage to dump all the partitions by connecting to the camera using telnet, how can i use them to unbrick the camera later? (if it will have to).

P.S.: if this method will work, isn't it a good idea to describe this method in documentation? Don't get me wrong, just a suggestion :D

roleoroleo, 2020-05-17:
Simply copy the rootfs image file and home image file in the sd card with the correct name.
And reboot.

closed 2020-08-20 by roleoroleo