roleoroleo/yi-hack-MStar #175: MQTT problem in 0.3.2

roleoroleo, 2020-05-10:
There is s problem in 0.3.2.
The cam doesn't send mqtt messages because of an error in mqttv4 daemon.

joekerna, 2020-05-13:
I can confirm the issue for 0.3.3 and the 9FUS camera.

bonuzzz, 2020-05-13:
I don't know whether will help it or nit, but mqtt worked on 0.3.2 firmware and now on 0.3.3. I had motion detected messages

roleoroleo, 2020-05-13:
Ok, thank you.

hagenuck1, 2020-05-14:
For me 0.3.3 on y203 isn’t working, too. After setting the settings, save them and restart the camera in maintenance it has set the IP to the password and all other settings are empty. 
This was also set in the mqttv4.conf.

roleoroleo, 2020-05-14:
Please, clean the cache of your browser.

joekerna, 2020-05-14:
> > I can confirm the issue for 0.3.3 and the 9FUS camera.
> 
> Do you confirm the issue is solved or it's not solved?

I confirm that it is _not_ solved

jahegu, 2020-05-14:
I have the same problem if i use the same sd in several cameras. Format SD and try again

roleoroleo, 2020-05-14:
Do you have more than 1 device with the same client name?

joekerna, 2020-05-14:
> Do you have more than 1 device with the same client name?
> 

No, only that one device

joekerna, 2020-05-14:
> I have the same problem if i use the same sd in several cameras. Format SD and try again

I will try that. Is there nothing important on the SD?

hagenuck1, 2020-05-14:
> Please, clean the cache of your browser.

That fixed the strange errors.

But mitt is only connecting and not submitting any motions for me. Messages of motion are only the retained ones (also when setting retain to 0 in yicam this isn't updated in home assistant.)

Mosquito message:
1589490328: New connection from 192.168.XXX.XX on port XXXX.
1589490328: Client yi-camera already connected, closing old connection.
1589490328: New client connected from 192.168.XXX.XX as yi-camera (p2, c1, k120, u'mosquittousername').

jahegu, 2020-05-14:
I checked with MQTT Explorer directly connected to the server

roleoroleo, 2020-05-15:
Tested again in my config and it works correctly.
Try to kill it and execute manually. You should see some log.

joekerna, 2020-05-15:
> Tested again in my config and it works correctly.
> Try to kill it and execute manually. You should see some log.

How do I kill and execute it manually?

roleoroleo, 2020-05-15:
Enter with ssh.
```
killall mqttv4
/home/yi-hack/bin/mqttv4
```

hagenuck1, 2020-05-15:
I guess I found the issue. 
According to the manually sended command it detected a motion, but in the topic „motion_detection“, which was „motion“ before, so my binary sensor didn’t update. 

Will try what happens after a reboot and autostart of mqttv4, but guess the wrong topic was the issue, sorry, didn’t see that was changed.

joekerna, 2020-05-15:
> Enter with ssh.
> ```
> killall mqttv4
> /home/yi-hack/bin/mqttv4
> ```

`/home/yi-hack/bin # killall mqttv4
killall: mqttv4: no process killed
/home/yi-hack/bin # mqttv4
Starting mqttv4 v0.0.3
Trying to connect... Client yicam sending CONNECT
Client yicam received CONNACK (0)
. connected!
CALLBACK MOTION STOP
Segmentation fault
/home/yi-hack/bin #`

roleoroleo, 2020-05-15:
Try to repeat again the command.

joekerna, 2020-05-15:
> /home/yi-hack/bin # killall mqttv4
killall: mqttv4: no process killed
/home/yi-hack/bin # mqttv4 
Starting mqttv4 v0.0.3
Trying to connect... Client yicam sending CONNECT
Client yicam received CONNACK (0)
. connected!
Client yicam sending PINGREQ
Client yicam received PINGRESP
Client yicam sending PINGREQ
Client yicam received PINGRESP
Client yicam sending PINGREQ
Client yicam received PINGRESP
CALLBACK MOTION STOP
Segmentation fault

I've waited for a few minutes. There was definitely motion in front of the camera. The are also videos on the SD-Card.

roleoroleo, 2020-05-16:
Could you share your mqtt config?

joekerna, 2020-05-18:
```
enabled
server ip: 192.168.178.30
server port: 1883
Client ID: yicam
username: yicam
password: MY_MOSQUITTO_USER_PASSWORD
Topix prefix: yicam
Topic suffix: motion
Motion start message: 
Motion stop message:
Topic Suffix for video list message: 
Baby crying: ALL_EMPTY
Keepalive: 120
QoS: yicam
Retain: yicam
```

roleoroleo, 2020-05-18:
QoS cannot be a string: use 0, 1 or 2.
Retain cannot be a string: use 0 or 1.
Insert a string value in baby crying, if you don't use it don't worry.
Insert a string value in topic suffix for video list, if you don't use it don't worry.
Insert a string value in start and stop messages: for example "ON" and "OFF" or "1" and "0".

Try again with this new config.

joekerna, 2020-05-18:
Still nothing :(
[mqttv4.conf.txt](https://github.com/roleoroleo/yi-hack-MStar/files/4646019/mqttv4.conf.txt)

roleoroleo, 2020-05-18:
I will send a debug release, probably a problem in file list feature.

bonuzzz, 2020-05-18:
@roleoroleo could you make retain parameter configurable for each topic?

roleoroleo, 2020-05-18:
Ok.

roleoroleo, 2020-05-18:
@joekerna 
Try this:
[mqttv4.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4646534/mqttv4.tar.gz)

joekerna, 2020-05-19:
It's no longer crashing with the segmentation fault.
However it is not sending MQTT messages for every motion detected. I've been waving around the camera like crazy and the SD card has a lot of videos. I get very very few MQTT motion detections.

The MQTTV4 output looks like this:

> Starting mqttv4 v0.0.3
Trying to connect... Client yicam sending CONNECT
Client yicam received CONNACK (0)
. connected!
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 02 00 00 00 f5 00 f5 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 5f 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 20. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 f5 01 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 02 00 00 00 ed 00 ed 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE

roleoroleo, 2020-05-19:
Pay attention: when a motion start is detected, the cam doesn't detect a new motion start until it detects a motion stop.
And the cam detects a motion stop only if there is a still image for a lot of seconds.

This is the correct sequence:
1. Movement ---> motion start
2. Stop ~2 minutes ---> motion stop
3. Another movement ---> new motion start

joekerna, 2020-05-20:
ok.. I didn't know that. Then it seems to be working.
Thanks a lot!

Is there an example Home-assistant configuration for this MQTT motion detection? Do I need specific topics for example?

joekerna, 2020-06-01:
I'm looking at the MQTT messages in Home Assistant debug interface.
I'm only getting Motion Stop messages. Shouldn't there be a motion start message a while before the motion stop message?

roleoroleo, 2020-06-02:
Yes, there should be a motion start before the motion stop.
The correct sequence is start, stop and files.

joekerna, 2020-06-02:
> 
> 
> @joekerna
> Try this:
> [mqttv4.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4646534/mqttv4.tar.gz)

I'm still on this file. Can I provide any debug information?

bonuzzz, 2020-06-02:
@joekerna try to disable retain in web and check statuses with mqtt explorer for example

roleoroleo, 2020-06-02:
Update to 0.3.3.

joekerna, 2020-06-03:
I am on 0.3.3 and retain is disabled. Still no start message.
Can the QoS or Keepalive settings play a role?

I will reflash 0.3.3 and try MQTT Explorer later

roleoroleo, 2020-06-03:
Qos and keepalive shouldn't be a problem.
Please, try again (now with 0.3.3) to start the mqttv4 software manually using ssh.
```
/home/yi-hack/bin/mqttv4
```
In addition to unrecognized messages, we should also see motion stops and motion starts.

joekerna, 2020-06-03:
I'm already on 0.3.3. I tried reflashing but it didn't overwrite anything.
When I manually run `mqttv4` I get version `v0.0.3`. The same for the file you've uploaded in your previous post. The files differ in size though.
Which is the most recent version?

roleoroleo, 2020-06-03:
The system checks the checksum of the upgrade file.
You cannot upgrade on the same version. You have to downgrade to a previous version and then upgrade again.
The correct mqttv4 binary is the one contained in 0.3.3.

roleoroleo, 2020-07-16:
@joekerna 
Is this issue solved?

joekerna, 2020-08-11:
yes,.. thanks!

closed 2020-08-13 by roleoroleo