roleoroleo/yi-hack-MStar #176: Merge wuestion

majkers, 2020-05-11:
Looks like someone is trying to merge some of Your code to yi hack https://github.com/TheCrypt0/yi-hack-v4/pull/230 Do you think it might work?

roleoroleo, 2020-05-11:
It depends on the merge.
There are modules that can be ported in other platform without problem.
For example ptz and onvif.

majkers, 2020-05-12:
In that case I would be much calmer if you'd do that merge...pleaaaase :)

majkers, 2020-07-24:
Sorry for reopening but...You said that:

> There are modules that can be ported in other platform without problem.
For example ptz and onvif.

Can You please make a fork of  TheCrypt0 repo and port modules that can be ported on cameras with Hi3518e Chipset? I habe 4 yi cameras. One ooutdoor, one home, two dome. One dome plus home with MStar Infinity Chipset and other with Hi3518e Chipset. It drives me nuts that I can do so many cool staff with Your software and can't with Hi3518e Chipset :( Please. I am ready to donate You for this...

roleoroleo, 2020-07-25:
It's not a donation problem.
I have no time for this.
Which features do you need?

majkers, 2020-07-28:
It is hard to choose but PTZ and go to preset and FTPupload are the best

roleoroleo, 2020-12-19:
See this project: https://github.com/alienatedsec/yi-hack-v4

majkers, 2020-12-19:
Thanks for a tip. I keep my fingers crossed :)

closed 2021-02-11 by roleoroleo