roleoroleo/yi-hack-MStar #178: RTSP Broken when RTSP Audio != None

Simpsonpt, 2020-05-12:
Serial: 4FUS 
Base: 4.2.0.0H_201909041620
Firmware Version: 0.3.2 or 0.3.3 

The RTSP stream stops working when we choose any RTSP Audio, the Snapshot feature gets a black screen and VLC can't connect, the RTSP service can't correctly start (confirmed via portscanner). When forced via wd_rtsp.sh the service begins (port 554 open), but the stream is not working anyway. The log file don't help `Starting RTSP watchdog...` .

MoiZie, 2020-05-12:
i can confirm this bug

bonuzzz, 2020-05-12:
I have audio stream on 0.3.3 in vlc, but after when I turned off audio and turned on again. I use highres stream only.

roleoroleo, 2020-05-12:
This is an experimental feature.
Please check if the stream is stable when audio is off.

bonuzzz, 2020-05-12:
I have to restart vlc and camera several times sometimes to get video working.

roleoroleo, 2020-05-13:
I am doing some experiments with 8 KHz instead of 16 KHz.
It seems more stable.

roleoroleo, 2020-05-13:
Please, give me some feedback on the quality without audio.

ypselon, 2020-05-13:
With wersion 0.3.3 with audio / video high, more than 4h without problems.

Simpsonpt, 2020-05-13:
As stated in the bug with audio the RTSP stream is broken, without audio RTSP looks good.

ypselon, 2020-05-13:
Hum.. i have 4 Cameras 6FUS v4.5.0 and 4 Cameras BFUS v4.5.0. without any problem with video audio stream maybe a problem with version 4.2.0 ?

bonuzzz, 2020-05-13:
with audio rtsp works for about 1.5 hours and breaks after that.  without any messages in dmesg at this time

ypselon, 2020-05-13:
[image]

torcuat, 2020-05-14:
I confirm the bug
on a 9FUSY256JI
Base version: 4.2.0.0H_201909041620

If RTSP audio is activated, the rtsp stream is broken. Desactivating services (FTP, SSH, etc) doesn't change anything.

bonuzzz, 2020-05-14:
if you reboot camera second time then rtsp stream will work... when rtsp stream doesn't work, snapshot also has black screen.

roleoroleo, 2020-05-14:
If you disable audio, does RTSP stream work well?

torcuat, 2020-05-14:
If I disable audio and reboot to apply changes the RTSP stream works as it did in previous versions when there was no audio option.

In my case, it never works with audio activated, no matter the reboots.

bonuzzz, 2020-05-14:
I thought black snapshot is indication of not working rtsp, but no.. (config now is both streams without audio)

roleoroleo, 2020-05-14:
Black snapshot means that rmm doesn't start correctly.

roleoroleo, 2020-05-14:
Beta version for black screen:
[y25_0.3.3.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4630242/y25_0.3.3.tar.gz)
Only suffix y25, fw version 4.2.0*

PieVo, 2020-05-17:
I think the sleep 2 after starting rmm is critical... On my camera the feed starts when I dump some more data from the audio_pipe.

roleoroleo, 2020-05-18:
Ok, I will check it.

roleoroleo, 2020-05-18:
Please go here:
https://github.com/roleoroleo/yi-hack-MStar/issues/183

Simpsonpt, 2020-05-19:
Tried the Beta version but the problem persists precisely in the same way when audio is activated the stream gets broken. But thank you for trying to address this problem.

Simpsonpt, 2020-06-25:
Closing this issues since the **yi-hack-MStar 0.3.4** release fixed this problem. Thanks for the great work @roleoroleo everyone that help debugging the problem.

closed 2020-06-25 by Simpsonpt