roleoroleo/yi-hack-MStar #179: RTSP stream stopped working after a while

bonuzzz, 2020-05-13:
I have this in dmesg:
[JPE, _JpeCtxGetBitInfo] eState=JPE_OUTBUF_FULL_STATE
[JPE, _JpeCtxGetBitInfo] ERROR!! tEncOutBuf.addr:0x23908280, tEncOutBuf.size:300                                                                             80
[JPE, _JpeCtxGetBitInfo] eState=JPE_OUTBUF_FULL_STATE
[JPE, _JpeCtxGetBitInfo] ERROR!! tEncOutBuf.addr:0x23908280, tEncOutBuf.size:300                                                                             80
[JPE, _JpeCtxGetBitInfo] eState=JPE_OUTBUF_FULL_STATE
[JPE, _JpeCtxGetBitInfo] ERROR!! tEncOutBuf.addr:0x23908280, tEncOutBuf.size:300                                                                             80
[JPE, _JpeCtxGetBitInfo] eState=JPE_OUTBUF_FULL_STATE

roleoroleo, 2020-05-13:
What's your config?
high? low? audio?

bonuzzz, 2020-05-13:
I used high res stream with audio. Testing now without audio.

majkers, 2020-05-13:
I have the same. Yi dome h201c with base version 4.6.0.0A_201908271549. RTSP i HomeAssistant is working for a while and stops. High res enalbed only without audio. In version 0.3.1 was working flawlessly.

majkers, 2020-05-14:
And again after a while it stopped working. Very unstable. Without audio though

roleoroleo, 2020-05-14:
Ok, I will check the problem and I will send a beta version to test.
I don't know why h201c is different from y203c.

roleoroleo, 2020-05-14:
This binary will restore the behavior of version 0.3.1:
[rRTSPServer.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4628978/rRTSPServer.tar.gz)

majkers, 2020-05-14:
I don't know either but I have h201c home camera and connected to HomeAssistant and I don't see any problems with it. How can I apply this file? Copy only? No rights change?

roleoroleo, 2020-05-14:
Rename the older rRTSPServer in /home/yi-hack/bin
Copy the file.
Change permission: `chown 0755 rRTSPServer`

It is the first time that it happens that h201c, y203c and y25 work differently.

majkers, 2020-05-14:
OK. Maybe there is something with me network ors sth. Let me staty with original and test few more days. I will let you know.

twinstef, 2020-05-16:
Hello,
I have the same instability problem with 0.3.3 & 0.3.2.
And an other bad one : rtsp lowRes not working at all.

By putting the 0.3.1 rRTSPServer inside /home/yi-hack/bin a can have rtsp Lowstream  again but instability is still here.

If i downgrade all firmware to 0.3.1 everything is OK (Hires & Lowres).

In 0.3.1 High & Low are on distincts ports (8554 for Low). But since 0.3.3(or .2?) they share the same port. I wonder it this is not the source of instability?

My Cam: Yi Dome Camera 1080p - BFUSY1EEEW - and i use h201c

bonuzzz, 2020-05-17:
rrtspserver binary crashed even without any connections after about 5 minutes

roleoroleo, 2020-05-18:
Please go here:
https://github.com/roleoroleo/yi-hack-MStar/issues/183

closed 2020-05-18 by bonuzzz